espressif/esp-idf #14692: [ESP32-C3] Cannot select UART1 for log output

doragasu, 2024-10-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1 (e088c3766ba440e72268b458a68f27b6e7d63986)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

For ESP32-C3, I cannot select UART 1 as output for console in menuconfig. If I select Custom Uart in ESP System Settings, I can change the GPIO pins for RX and TX, but the option to change the UART device number is disabled, as shown in screenshot:

[image]

Option `UART peripheral to use for console output` should be enabled an I should be able to select `1`.

### What is the actual behavior?

Option `UART peripheral to use for console output` is disable and thus I cannot select UART 1.

### Steps to reproduce.

1. Run `idf.py menuconfig`
2. Go to `Component config -> ESP System Settings`
3. In option `Channel for console output` select `Custom UART`.
4. Optionally change `Channel for console secondary output`, the result will be the same.
5. Option `UART peripheral to use for console output` is not available, cannot select UART1

ESP-Marius, 2024-10-10:
Support for this has already been added in 46d33e46efdf368f3f77d0879804a3f3283474bb, so it is available on newer release branches

doragasu, 2024-10-11:
Oh, thanks! This is not merged into 5.1.1, right? So I have to upgrade to 5.3?

ESP-Marius, 2024-10-14:
Yup, that is correct.

closed 2024-10-14 by ESP-Marius