espressif/esp-idf #14697: ble_multi_conn_cent crashes after peripherals connect

maxikrie, 2024-10-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

ESP32-C6FH4 (QFN32) (revision v0.0)

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6-DevKitM-1 V1.0

### Power Supply used.

USB

### What is the expected behavior?

As the described in the tutorial, multiple peripherals (in this case emulated) should connect to one central. I am using a ESP32-devkit for the peripheral and a ESP32-C6 devkit for the central.

### What is the actual behavior?

As expected, three emulated peripherals connect to the central. However, after a short period the central crashes.

### Steps to reproduce.

1. Attach ESP32-C6 with ble_multi_conn_cent example
2. Attach ESP32 with ble_multi_conn_prph example
3. Observe how several peripherals successfully connect
4. Central crashes

### Debug Logs.

```plain
I (23466) ESP_MULTI_CONN_CENT: Create connection. -> peer addr c7:27:e7:81:ad:21
I (23536) ESP_MULTI_CONN_CENT: Connection established. Handle:0, Total:1
I (24096) ESP_MULTI_CONN_CENT: Create connection. -> peer addr db:93:0d:bb:dc:94
I (24416) ESP_MULTI_CONN_CENT: Connection established. Handle:1, Total:2
I (24956) ESP_MULTI_CONN_CENT: Create connection. -> peer addr e9:43:84:25:7e:4f
I (25016) ESP_MULTI_CONN_CENT: Connection established. Handle:2, Total:3
E (32266) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (32266) task_wdt:  - IDLE (CPU 0)
E (32266) task_wdt: Tasks currently running:
E (32266) task_wdt: CPU 0: ble_ll_task
E (32266) task_wdt: Print CPU 0 (current core) backtrace
Print CPU 0 (current core) registers
Core  0 register dump:
MEPC    : 0x4204bcc2  RA      : 0x4204bc8c  SP      : 0x4083e6f0  GP      : 0x40812884  
--- 0x4204bcc2: r_ble_ll_conn_next_event at ??:?
0x4204bc8c: r_ble_ll_conn_next_event at ??:?

TP      : 0x4083e7f0  T0      : 0x400208f0  T1      : 0x00000001  T2      : 0x00000000  
--- 0x400208f0: systimer_hal_set_alarm_target in ROM

S0/FP   : 0x40823488  S1      : 0x00000136  A0      : 0x0000001a  A1      : 0x00000001  
A2      : 0x00000002  A3      : 0x00000001  A4      : 0x00000002  A5      : 0x4082e688  
A6      : 0x01997439  A7      : 0x00000001  S2      : 0x00000001  S3      : 0x0000000f  
S4      : 0x00000000  S5      : 0x00000023  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x00000008  T4      : 0x00000028  T5      : 0x01997514  T6      : 0x019973de  
MSTATUS : 0x00000000  MTVEC   : 0x00000023  MCAUSE  : 0x00000000  MTVAL   : 0x0000000f  
MHARTID : 0x00000002
```

### More Information.

I want to emphasize that I am using a ESP32 instead of ESP32-C6 for the peripheral, while using a ESP32-C6 for the central. I am not sure if this is an issue.

maxikrie, 2024-10-10:
I have investigated this a bit further and it seems that the task in question `ble_ll_task` is part of the closed controller binary libble_app.a.
`grep: ./esp-idf/components/bt/controller/lib_esp32c6/esp32c6-bt-lib/libble_app.a: binary file matches`

I can therefore unfortunately debug it further. Please advise!

zhaoweiliang2021, 2024-10-11:
@maxikrie  Thanks for raising this question.
[esp32c6_wdt_issue_patch.zip](https://github.com/user-attachments/files/17339093/esp32c6_wdt_issue_patch.zip)
The cause has been found and a fix is ​​provided. You can further test it.
Please unzip the compressed package and get the libble_app.a file, replace the corresponding file in the directory components/bt/controller/lib_esp32c6/esp32c6-bt-lib, and then test.
Local testing currently passes, but the fix requires further testing. Please wait for the official version to be released.
If you have any other questions during this process, please give us feedback in time, thanks.

closed 2024-10-11 by maxikrie