espressif/esp-idf #14698: PHY data partition, restoring default fails

jcecconi, 2024-10-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

File: components/esp_phy/phy_init.c

Description:
The function esp_phy_get_init_data fails to write the default values without reporting any error. It prints the message: "failed to validate PHY data partition, restoring default data into flash...", but the default values are not actually written to flash. This issue occurs because, prior to calling esp_partition_write, the following line should be executed:

err = esp_partition_erase_range(partition, 0, partition->size);

Without this line, the ESP32 will print the message "restoring default data" every time it boots, but it will fail to properly write the data to the partition.

mhdong, 2025-02-12:
hi @jcecconi Thank you very much for your question. Could you please provide the steps to reproduce the issue?

Alvin1Zhang, 2025-03-07:
@jcecconi Thanks for reporting.  Could you please provide the steps to reproduce the issue? Thanks.

closed 2025-04-21 by espressif-bot