espressif/esp-idf #14699: [esp32-p4] question about sdmmc

chegewara, 2024-10-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,
i think i know the answer, but i would like to make sure.
Is it possible to use wifi/ble and sd card in sdmmc mode at the same time on ev board?

On internet ive found some very early datasheet which stands that P4 has one sdmmc host which can supports mode:
> 4-bit (supports two SD/SDIO/MMC 4.41 cards, and one SD card operating at 1.8 V in 4-bit mode)

Now i have code where wifi/ble is working nicely. Then i am trying to add sd card and here is result:
```
I (437) host_init: ESP Hosted : Host chip_ip[18]
I (478) H_API: ESP-Hosted starting. Hosted_Tasks: prio:23, stack: 5120 RPC_task_stack: 5120
sdio_mempool_create free:168724 min-free:168724 lfb-def:126976 lfb-8bit:126976

I (483) gpio: GPIO[18]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (493) gpio: GPIO[19]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (502) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (511) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (520) gpio: GPIO[16]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (530) gpio: GPIO[17]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (539) H_API: ** add_esp_wifi_remote_channels **
I (544) transport: Add ESP-Hosted channel IF[1]: S[0] Tx[0x4000d7d6] Rx[0x4001bb68]
--- 0x4000d7d6: transport_drv_sta_tx at /project/caplamp-configurator/managed_components/espressif__esp_hosted/host/drivers/transport/transport_drv.c:208
0x4001bb68: esp_wifi_remote_channel_rx at /project/caplamp-configurator/managed_components/espressif__esp_wifi_remote/esp_wifi_remote_net.c:19

I (553) transport: Add ESP-Hosted channel IF[2]: S[0] Tx[0x4000d71e] Rx[0x4001bb68]
...

I (798) sdmmc_periph: sdmmc_host_init: SDMMC host already initialized, skipping init flow
I (807) gpio: GPIO[43]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (816) gpio: GPIO[44]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (826) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (835) gpio: GPIO[40]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (844) gpio: GPIO[41]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (854) gpio: GPIO[42]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (964) gpio: GPIO[42]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
File: TEST.TXT => 1048576
```
so far so good, but then this is happening:
```
I (2162) sdio_wrapper: SDIO master: Data-Lines: 4-bit Freq(KHz)[40000 KHz]
I (2162) sdio_wrapper: GPIOs: CLK[18] CMD[19] D0[14] D1[15] D2[16] D3[17] Slave_Reset[54]
I (2162) H_SDIO_DRV: Starting SDIO process rx task
I (2167) sdio_wrapper: Queues: Tx[20] Rx[20] SDIO-Rx-Mode[1]
E (2210) sdmmc_sd: sdmmc_init_sd_scr: send_scr (1) returned 0xffffffff
E (2210) sdio_wrapper: sdmmc_card_init failed
I (2211) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2220) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2229) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2239) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2248) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2257) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2267) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2276) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2285) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2294) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2304) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2313) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2322) gpio: GPIO[43]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2332) gpio: GPIO[44]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2341) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2350) gpio: GPIO[40]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2360) gpio: GPIO[41]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2369) gpio: GPIO[42]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2379) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2388) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2397) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2406) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
E (2416) H_SDIO_DRV: sdio card init failed

FreeRTOS: FreeRTOS Task "sdio_read" should not return, Aborting now!
Core  1 register dump:
--- Stack dump detected
MEPC    : 0x4ff00c58  RA      : 0x4ff07418  SP      : 0x4ff1e470  GP      : 0x4ff10c80  
--- 0x4ff00c58: panic_abort at /opt/esp/idf/components/esp_system/panic.c:468
0x4ff07418: vPortTLSPointersDelCb at /opt/esp/idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:700

TP      : 0x4ff1e4a0  T0      : 0x4fc0a778  T1      : 0x20000000  T2      : 0xffffffff  
S0/FP   : 0x00000000  S1      : 0x00000000  A0      : 0x4ff112b0  A1      : 0x400b30ef  
A2      : 0x00000023  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x4ff19000  
A6      : 0x4000b3be  A7      : 0x0000000a  S2      : 0x00000000  S3      : 0x00000000  
--- 0x4000b3be: usb_serial_jtag_write at /opt/esp/idf/components/esp_driver_usb_serial_jtag/src/usb_serial_jtag_vfs.c:184

S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x00000000  T4      : 0x4005bbbc  T5      : 0x4005bb74  T6      : 0x4005ad38  
--- 0x4005bbbc: sdmmc_host_io_int_wait at /opt/esp/idf/components/esp_driver_sdmmc/src/sdmmc_host.c:981
0x4005bb74: sdmmc_host_io_int_enable at /opt/esp/idf/components/esp_driver_sdmmc/src/sdmmc_host.c:975
0x4005ad38: sdmmc_host_deinit_slot at /opt/esp/idf/components/esp_driver_sdmmc/src/sdmmc_host.c:813

MSTATUS : 0x00011880  MTVEC   : 0x4ff00003  MCAUSE  : 0x00000002  MTVAL   : 0x00000000  
--- 0x4ff00003: _vector_table at ??:?

MHARTID : 0x00000001  

--- Backtrace:

panic_abort (details=details@entry=0x4ff112b0 <msg> "") at /opt/esp/idf/components/esp_system/panic.c:468
468         asm("unimp");   // should be an invalid operation on RISC-V targets
#0  panic_abort (details=details@entry=0x4ff112b0 <msg> "") at /opt/esp/idf/components/esp_system/panic.c:468
#1  0x4ff07418 in vPortTaskWrapper (pxCode=<optimized out>, pvParameters=<optimized out>) at /opt/esp/idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:260
```

Thanks

igrr, 2024-10-11:
@chegewara Could you please specify the commit of IDF which you are using, as well as the code you are using to initialize both Wi-Fi and SD card?

cc @mantriyogesh

chegewara, 2024-10-11:
Hi @igrr 
i thought its not possible to use both at the same time, but just in case i asked this.
- I am using current master `ESP-IDF v5.4-dev-3489-g6e5a178b31`
- hosted
```
  espressif/esp_hosted:
    component_hash: 2bc6a1d52c7253c584d34dc2f180f138e99db4d7341fcc8552cea49020442c96
```
- esp_remote_wifi
```
  espressif/esp_wifi_remote:
    component_hash: dd44abf72eab81106bb33f95728cd8cfd2fa0647dcaec0c99ecbb47f929d8a2e
```

I am not using wifi in this app, only ble and usb-netif to serve files over USB NCM.

Thanks

igrr, 2024-10-11:
On ESP32-P4, SD cards can be used along with esp_hosted/esp_wifi_remote since commit d1571c144b667ccee348ac4cf75f2f02e91dbcb3. However since this is a relatively new feature, there could be something that isn't fully working.

Would it be possible for you to share a project which reproduces this issue?

chegewara, 2024-10-11:
Sure, i can prepare some code, just i have to clean it up.

mantriyogesh, 2024-10-11:
@chegewara , 
I suspect that you are using `esp-extconn` along with `esp-hosted`.
Just to rule out, Please attach the sdkconfig as well.

If in case missed, While [adding dependencies](https://github.com/espressif/esp-hosted/blob/feature/esp_as_mcu_host/docs/esp32_p4_function_ev_board.md#31-adding-components) for esp-hosted, please you remove `esp-extconn` as shown below:

[image]

chegewara, 2024-10-11:
> @chegewara , I suspect that you are using `esp-extconn` along with `esp-hosted`. Just to rule out, Please attach the sdkconfig as well.
>
> If in case missed, While [adding [...]

Nope, im not using it. Ive been trying to test extconn, but i figured out it is completely different interface.

```
dependencies:
  chegewara/usb-netif:
    component_hash: 11a03df325f1f470de4bcb65a356758c7accf4c30d2f2beed85702461836f65c
    dependencies:
    - name: espressif/esp_tinyusb
      registry_url: https://components.espressif.com
      require: private
      version: ^1.4.0
    - name: idf
      require: private
      version: ^5.0
    source:
      registry_url: https://components.espressif.com/
      type: service
    targets:
    - esp32s2
    - esp32s3
    - esp32p4
    version: 0.1.0
  espressif/eppp_link:
    component_hash: c2fd9c57ac14a68a62c9d9cdeec39a2b265abc16779196ce2c9f5000c9a4dd8b
    dependencies:
    - name: espressif/esp_serial_slave_link
      registry_url: https://components.espressif.com/
      require: private
      version: ^1.1.0
    - name: idf
      require: private
      version: '>=5.2'
    source:
      registry_url: https://components.espressif.com
      type: service
    version: 0.2.0
  espressif/esp_hosted:
    component_hash: 35cdc6173d3106f43974d05b61da7aa9c190b17f03a2c5dfb2b6664a4d0e9b0a
    dependencies:
    - name: idf
      require: private
      version: '>=5.3'
    source:
      registry_url: https://components.espressif.com/
      type: service
    version: 0.0.20
  espressif/esp_serial_slave_link:
    component_hash: 8c534b7d4986ca0e4c41db71f559492b795ab862f1d15359d7993034696b7bcc
    dependencies:
    - name: idf
      require: private
      version: '>=5.0'
    source:
      registry_url: https://components.espressif.com/
      type: service
    version: 1.1.0
  espressif/esp_tinyusb:
    component_hash: 2be80f1c9fcdc1ccf945e001856b88fee5a82c321187c3f4351519928e5de168
    dependencies:
    - name: idf
      require: private
      version: '>=5.0'
    - name: espressif/tinyusb
      registry_url: https://components.espressif.com/
      require: public
      version: '>=0.14.2'
    source:
      registry_url: https://components.espressif.com
      type: service
    version: 1.4.5
  espressif/esp_wifi_remote:
    component_hash: dd44abf72eab81106bb33f95728cd8cfd2fa0647dcaec0c99ecbb47f929d8a2e
    dependencies:
    - name: espressif/eppp_link
      registry_url: https://components.espressif.com
      require: private
      version: '>=0.1'
    - name: espressif/esp_hosted
      registry_url: https://components.espressif.com
      require: private
      version: '>=0.0.6'
    - name: idf
      registry_url: https://components.espressif.com
      require: private
      version: '>=5.3'
    source:
      registry_url: https://components.espressif.com/
      type: service
    version: 0.3.0
  espressif/tinyusb:
    component_hash: 214989d502fc168241a4a4f83b097d8ac44a93cd6f1787b4ac10069a8b3bebd3
    dependencies:
    - name: idf
      require: private
      version: '>=5.0'
    source:
      registry_url: https://components.espressif.com/
      type: service
    targets:
    - esp32s2
    - esp32s3
    - esp32p4
    version: 0.15.0~10
  idf:
    source:
      type: idf
    version: 5.4.0
direct_dependencies:
- chegewara/usb-netif
- espressif/esp_hosted
- espressif/esp_wifi_remote
- idf
manifest_hash: 399eb35fe6bbdcbc278e8197aa2af7f691c32ee15286aa714265f7b19ec9d40c
target: esp32p4
version: 2.0.0
```

I will post example code in a moment.

chegewara, 2024-10-11:
I know this is not place to report it, but i thought i will just mention.
There is issue with this exact version of hosted compnent
```
  espressif/esp_hosted:
    component_hash: 35cdc6173d3106f43974d05b61da7aa9c190b17f03a2c5dfb2b6664a4d0e9b0a
    dependencies:
    - name: idf
      require: private
      version: '>=5.3'
    source:
      registry_url: https://components.espressif.com/
      type: service
    version: 0.0.20
```

It is forcing flag `CONFIG_ESP_WIFI_REMOTE_ENABLED` which doesnt let to enable `CONFIG_ESP_ENABLE_BT`

with `0.0.19` i can use nimble.
 
Thanks

chegewara, 2024-10-11:
@mantriyogesh 
Here is test app 
https://github.com/esp32-p4-excercises/sdmmc-crash-test

Thanks

SohKamYung-Espressif, 2024-10-11:
Regarding the SD Card Init code in  https://github.com/espressif/esp-idf/issues/14699#issuecomment-2406740797

Change this section
```c
    sdmmc_host_t host = SDMMC_HOST_DEFAULT();
    host.max_freq_khz = SDMMC_FREQ_HIGHSPEED;
```
to this

```c
    sdmmc_host_t host = SDMMC_HOST_DEFAULT();
    host.slot = SDMMC_HOST_SLOT_0;
    host.max_freq_khz = SDMMC_FREQ_HIGHSPEED;
```

The `SDMMC_HOST_DEFAULT()` macro uses `SDMMC_HOST_SLOT_1`, which is assigned to SDIO to the C6.

To access the SD Card slot on the P4 DevKit board, you may have to change it to `SDMMC_HOST_SLOT_0`.

chegewara, 2024-10-11:
On P4 ev board there is only 1 usable slot, slot 1.
When i change slot to 0 i get this
```
E (787) sdmmc_periph: sdmmc_host_init_slot(619): doesn't support routing from GPIO matrix, driver uses dedicated IOs
E (798) vfs_fat_sdmmc: slot init failed (0x102).
```
I also cant change esp_hosted slot to 0, which probably means i cant use hosted + sdmmc sd card with current devkit.

> From sdkconfig, ESP_HOSTED_ENABLED is 'y'.

yes, because component dependency is set to esp_host =0.0.19 i yaml file.

I dont know why you cant build this code, all required components are included (esp-nimble-cpp) or are from components registry including this one:
https://github.com/esp32-open-source/usb-components/tree/master/usb-netif

Thanks for investigating this

igrr, 2024-10-11:
There is a bit of a conflict with slots numbering, let me try to explain this.

ESP32-P4 SoC has two SDMMC slots:
- Slot 0 can be used over IO MUX only, and is routed to a fixed set of pins: https://github.com/espressif/esp-idf/blob/6e5a178b3120dced7fa5c29c655cc22ea182df3d/components/soc/esp32p4/include/soc/sdmmc_pins.h#L9-L14
- Slot 1 pins can be freely assigned to any GPIOs using GPIO Matrix.

`SDMMC_SLOT_CONFIG_DEFAULT()` macro sets the same GPIOs as slot 0. However, `SDMMC_HOST_DEFAULT()` sets the slot number to slot 1.

On the ESP32-P4 EV board,
- SD card is connected to Slot 0 pins
- SDIO is connected to other pins

So when we run an SD example app, everything still works, because slot 1 can be routed to the SD card (attached to slot 0 pins) via GPIO matrix. However when we also need to use slot 1 for SDIO, this no longer works.

To initialize the SD card to use slot 0 and slot 0 pins, currently we have to do something like this:

```c++
    sdmmc_host_t host = SDMMC_HOST_DEFAULT();
    host.slot = SDMMC_HOST_SLOT_0;
    sdmmc_slot_config_t slot_config = {
        .width = 4,
        .cd = SDMMC_SLOT_NO_CD,
        .wp = SDMMC_SLOT_NO_WP,
    };
```