espressif/esp-idf #14700: HFP_HF audio connects then automatically disconnect

spacetronics, 2024-10-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

**SETUP**
ESP32-WROVER-E
INMP441
ESP-IDF v5.3
Eclipse IDE
Redmi Note 7

So i'm trying to run the hfp_hf example to stream my audio to my smartphone using bluetooth. I've already set my audio path configurations to PCM. I've set my INMP441 pins as:
SD: GPIO_OUTPUT_PCM_DOUT (26)
SCK: GPIO_OUTPUT_PCM_CLK_OUT (5)
WS: GPIO_OUTPUT_PCM_FSYNC (25)
L/R: Not connected

The bluetooth connection between my ESP32 and smartphone was successful. But when i try to connect audio, the first audio state was connected, then connected_msbc, and then it disconnected. Since i've set my audio path to PCM, i have no idea why the logs are stating connected_msbc. I tried #14231 by 
> set the value = 1 in handler_callback(value)
and the connected_msbc logs are gone, but the disconnecting issue still persist. 

The debug logs are as follow
**[idf.py monitor]**
`Executing action: monitor
Serial port COM6
Connecting.....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting......
Detecting chip type... ESP32
Running idf_monitor in directory C:\Users\mbran\workspace\hfp_hf
Executing "C:\Espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe C:\Espressif\frameworks\esp-idf-v5.3\tools/idf_monitor.py -p COM6 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 0 C:\Users\mbran\workspace\hfp_hf\build\hfp_hf.elf --force-color -m 'C:\Espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe' 'C:\Espressif\frameworks\esp-idf-v5.3\tools\idf.py'"...
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM6 instead...
--- esp-idf-monitor 1.4.0 on \\.\COM6 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7176
load:0x40078000,len:15564
ho 0 tail 12 room 4
load:0x40080400,len:4
0x40080400: _init at ??:?

load:0x40080404,len:3904
entry 0x40080640
I (31) boot: ESP-IDF v5.3 2nd stage bootloader
I (31) boot: compile time Oct  8 2024 21:47:17
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v3.1
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=21e94h (138900) map
I (152) esp_image: segment 1: paddr=00031ebc vaddr=3ffbdb60 size=04c88h ( 19592) load
I (159) esp_image: segment 2: paddr=00036b4c vaddr=40080000 size=094cch ( 38092) load
[... 99 lines cut ...]
W (17014) BT_APPL: bta_hf_client_send_at: No service, skipping 11 command
I (17014) BT_RFCOMM: PORT_WriteData() max_len:9
I (17064) BT_RFCOMM: PORT_ReadData() handle:2 max_len:512
I (17074) BT_RFCOMM: PORT_WriteData() max_len:10
I (17124) BT_RFCOMM: PORT_ReadData() handle:2 max_len:512
W (23254) BT_HCI: hci cmd send: sniff: hdl 0x80, intv(400 800)
W (23324) BT_HCI: hcif mode change: hdl 0x80, mode 2, intv 800, status 0x0
I (23334) BT_HF: ESP_BT_GAP_MODE_CHG_EVT mode:2`

**[connect audio]**
`connect audio
I (25434) BT_RFCOMM: PORT_WriteData() max_len:7
I (25434) BT_HF: APP HFP event: AUDIO_STATE_EVT
I (25434) BT_HF: --audio state connecting
hfp_hf> W (26644) BT_HCI: hcif mode change: hdl 0x80, mode 0, intv 0, status 0x0
I (26644) BT_HF: ESP_BT_GAP_MODE_CHG_EVT mode:0
I (26664) BT_RFCOMM: PORT_ReadData() handle:2 max_len:512
I (26674) BT_RFCOMM: PORT_WriteData() max_len:9
I (26674) BT_RFCOMM: PORT_ReadData() handle:2 max_len:512
I (26684) BT_RFCOMM: PORT_ReadData() handle:2 max_len:512
I (26694) BT_HF: APP HFP event: AUDIO_STATE_EVT
I (26694) BT_HF: --audio state connected_msbc
W (26714) BT_HCI: hcif disc complete: hdl 0x180, rsn 0x13
I (26724) BT_HF: APP HFP event: AUDIO_STATE_EVT
I (26724) BT_HF: --audio state disconnected`

jtwaleson, 2024-10-13:
Working on something similar, please note that INMP441 requires L/R to be connected to either ground or VDD. With nothing connected, the value is floating and output can randomly go to either the left or right channel.

spacetronics, 2024-10-14:
> Working on something similar, please note that INMP441 requires L/R to be connected to either ground or VDD. With nothing connected, the value is floating and output can randomly go to either the left or right channel.

i connected the L/R to GND. still not working

jtwaleson, 2024-10-14:
> i connected the L/R to GND. still not working

Sorry, should have been more explicit. This wasn't to solve the issue, just something that could cause additional problems.

spacetronics, 2024-10-14:
oh ok, thanks for the reminder!

esp-qing, 2024-10-14:
Hi, @spacetronics 
> Since i've set my audio path to PCM, i have no idea why the logs are stating connected_msbc. 
> set the value = 1 in handler_callback(value)
> and the connected_msbc logs are gone, but the disconnecting issue still persist.

This issue has been fixed, and the commit id of v5.3 is 3af62b6e4868f385e2f33a90af1f46eb604d8ac2. You can update the branch so that there is no need to change the idf code.

> So i'm trying to run the hfp_hf example to stream my audio to my smartphone using bluetooth. I've already set my audio path configurations to PCM. I've set my INMP441 pins as:
SD: GPIO_OUTPUT_PCM_DOUT (26)
SCK: GPIO_OUTPUT_PCM_CLK_OUT (5)
WS: GPIO_OUTPUT_PCM_FSYNC (25)
L/R: Not connected
> The bluetooth connection between my ESP32 and smartphone was successful. But when i try to connect audio, the first audio state was connected, then connected_msbc, and then it disconnected. 

There is no problem with the connection between ESP32 and smartphone. I think It may be a connection issue with the audio device  (I have tried connecting other audio devices before, and there is no problem with the voice output.).  I read the data sheet of INMP441. In addition to the connection of the signal pins and the L/R  pin, the CHIPEN pin needs to be set high. (when set low, disabled; when set high, enabled). Please try it and see if there is still a problem?

spacetronics, 2024-10-14:
> This issue has been fixed, and the commit id of v5.3 is https://github.com/espressif/esp-idf/commit/3af62b6e4868f385e2f33a90af1f46eb604d8ac2. You can update the branch so that there is no need to change the idf code.

so i don't know how to update the branch using eclipse IDE so i just edit it manually, but the connected_msbc log still exist.

> the CHIPEN pin needs to be set high. (when set low, disabled; when set high, enabled). Please try it and see if there is still a problem?

my INMP441 only have VDD but i think its the same as CHIPEN pin (CMIIW) and i connect it to 3V3 pin in esp32. the issue sadly still persist.

jtwaleson, 2024-10-14:
> > the CHIPEN pin needs to be set high. (when set low, disabled; when set high, enabled). Please try it and see if there is still a problem?
> 
> I tested my microphone with [this](https://github.com/atomic14/esp32-i2s-mic-test) and it already worked. My INMP441 board only have VDD pin but i think its the same as CHIPEN pin (CMIIW) and i connect it to 3V3 pin in esp32. the issue sadly still persist.

If you use the common round 6 pin PCB board with the INMP441: that has CHIPEN connected to VDD internally.

spacetronics, 2024-10-15:
> If you use the common round 6 pin PCB board with the INMP441: that has CHIPEN connected to VDD internally.

Yep, i use that board. I just looked at the schematic in the datasheet and you're correct.

spacetronics, 2024-10-19:
so i followed this [https://www.instructables.com/ESP32-Mic-Testing-With-INMP441-and-DumbDisplay/](mic testing) (it uses bluetooth) and it worked so idk why the esp idf one always disconnnected. do i need a specific application in my phone so that my mic can be heard from the phone? i use "Bluetooth Audio Connect Widget"

jtwaleson, 2024-10-27:
Having done a lot of research I can tell you that currently INMP441 will not work with HFP_HF in PCM mode. INMP441 uses Philips / 1 bit-shift / i2s the timing format, instead of the "normal" / left-aligned / "MSB" timing format. See here: https://github.com/espressif/esp-idf/issues/13738#issuecomment-2423758387

Until esp-idf allows you to configure the timing offset in menuconfig, the microphone will not function correctly and produce a lot of noise with a barely perceptible voice signal. See also this thread https://github.com/espressif/esp-idf/issues/13738

Now why it's always disconnecting I don't know. In my case it did "work" as documented but only produced static.

esp-qing, 2024-10-28:
> i updated the branch and the connected_msbc log is fixed so thank you. @esp-qing
> 
> However, i followed this [mic testing](https://www.instructables.com/ESP32-Mic-Testing-With-INMP441-and-DumbDisplay/) tutorial (it uses bluetooth) and it worked so idk why the esp idf one always disconnnected. do i need a specific application in my phone so that my mic can be heard from the phone? i use "Bluetooth Audio Connect Widget"

Generally, there is no need to install app on the phone.  
From your debug logs, the error code is 0x13. In the Bluetooth Core Specification, the user on the remote device either terminated the connection or stopped broadcasting packets. Because the bluetooth is disconnected, the audio is disconnected. There are many reasons for the Bluetooth disconnection. The easiest way is to try a different phone?
(Regarding the INMP441, you can refer to [jtwaleson](https://github.com/espressif/esp-idf/issues/14700#issuecomment-2439970917)'s comments.)

[image]

[image]

esp-qing, 2024-12-25:
> I see, sadly i need to use INMP441 because of the space constraint for my project. I haven't tried a different phone, but since i need to use INMP441, would the outcome be the same? 

Hi, The default PCM output waveform of hfp example is stereo mode(introduced in README) and similar to the MSB Alignment Standard in the esp32 I2S document. This waveform cannot be changed temporarily.
>  And Android use HFP v1.5, which means there's no wide band speech support. So i can't use mSBC audio path.

Only CVSD codec can be used in PCM datapath, so there is no need to use mSBC.
> But i have a question, the mic testing tutorial that i mentioned is using bluetooth serial, what is the difference between them and HFP? And what is the downsides of using bluetooth serial over HFP? Thank you.

Personal opinion, one obvious difference is that the mic testing tutorial that you mentioned can freely configure the waveform by software, but the default PCM waveform in HFP is fixed. If you need Philips Standard Mode, it can be met through software configuration, but it may not be implemented in hfp-hf using PCM datapath.

closed 2025-02-10 by spacetronics