espressif/esp-idf #14702: [ESP-IDF Applications on Host] ESP_ERROR_CHECK_WITHOUT_ABORT references _esp_error_check_failed_without_abort() which is not implemented

FrancoisB-HEX, 2024-10-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1-395-g707d097b01

### Operating System used.

Linux

### How did you build your project?

CLion IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Compiling an ESP-IDF project to run on Host (Linux) for Simulation with sources that make use of `ESP_ERROR_CHECK_WITHOUT_ABORT(x)` should succeed

### What is the actual behavior?

Compiling an ESP-IDF project to run on Host (Linux) for Simulation with sources that make use of `ESP_ERROR_CHECK_WITHOUT_ABORT(x)` fails linking since `_esp_error_check_failed_without_abort()` is not implemented in _esp_err_check_linux.c_

### Steps to reproduce.

1. Setup an example ESP-IDF project and make use of the `ESP_ERROR_CHECK_WITHOUT_ABORT(x)` macro
2. Configure the `--preview set-target linux` ESP-IDF on Host Simulation
3. Build the project
4. Build error:
```
undefined reference to `_esp_error_check_failed_without_abort'
```
...

### More Information.

I know ESP-IDF on Host is still in preview so I just want to point this out since _esp_err_check_linux.c_ does exist and implement the related function `_esp_error_check_failed()`

ESP-Marius, 2024-10-11:
Hey,

This has been fixed on master in 023d112cbfb74e919ee5192f3436ba8c65ca2b3c, so it should work in later version of IDF.

Since the linux target is still in preview we dont backport these kind of fixes/features,  so if you really need this macro in v5.3 you'd have to cherry-pick it into your project.

FrancoisB-HEX, 2024-10-11:
Ok thanks for the info. It was simple enough to add this function to my _components/esp_common/src/esp_err_check_linux.c_ file which is just `_esp_error_check_failed()` without `abort();`:
```
void _esp_error_check_failed_without_abort(esp_err_t rc, const char *file, int line, const char *function, const char *expression)
{
    printf("ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x%x", rc);
#ifdef CONFIG_ESP_ERR_TO_NAME_LOOKUP
    printf(" (%s)", esp_err_to_name(rc));
#endif //CONFIG_ESP_ERR_TO_NAME_LOOKUP
    printf(" at %p\n", __builtin_return_address(0));
    printf("file: \"%s\" line %d\nfunc: %s\nexpression: %s\n", file, line, function, expression);
}
```

Closed by https://github.com/espressif/esp-idf/commit/023d112cbfb74e919ee5192f3436ba8c65ca2b3c

closed 2024-10-11 by FrancoisB-HEX