espressif/esp-idf #14708: 在使用APSTA模式下例程中终端输出信息又问题

cannnnnnnnnnnn, 2024-10-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

在V5.3.1分支中，测试station和softap_sta在WiFi连接配置中修改正确WiFi名称并添加错误的WiFi密码查看回调函数的输出信息，会出现wifi:state的输出信息，这个输出信息我并不清楚是什么原因输出也不清楚如何关闭，应为之还在测试例程中就没有这样的信息输出只有回调函数中的信息输出，实际上这些信息是不因该输出的，不清楚是硬件的原因还是软件的原因

测试输出信息如下：

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x178c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xcb8
load:0x403cb700,len:0x2db0
entry 0x403c8914
I (27) boot: ESP-IDF v5.3.1-dirty 2nd stage bootloader
I (27) boot: compile time Oct 11 2024 09:44:30
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (44) boot.esp32s3: SPI Flash Size : 8MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c080020 size=1fa58h (129624) map
I (123) esp_image: segment 1: paddr=0002fa80 vaddr=3fc99400 size=00598h (  1432) load
I (124) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=77c00h (490496) map
I (217) esp_image: segment 3: paddr=000a7c28 vaddr=3fc99998 size=043c0h ( 17344) load
I (221) esp_image: segment 4: paddr=000abff0 vaddr=40374000 size=15388h ( 86920) load
I (250) boot: Loaded app from partition at offset 0x10000
I (250) boot: Disabling RNG early entropy source...
I (262) cpu_start: Multicore app
I (272) cpu_start: Pro cpu start user code
I (272) cpu_start: cpu freq: 160000000 Hz
I (272) app_init: Application information:
I (275) app_init: Project name:     wifi_station
I (280) app_init: App version:      1
I (284) app_init: Compile time:     Oct 11 2024 09:46:33
I (290) app_init: ELF file SHA256:  06bbcd7c4...
I (296) app_init: ESP-IDF:          v5.3.1-dirty
I (301) efuse_init: Min chip rev:     v0.0
I (306) efuse_init: Max chip rev:     v0.99
I (311) efuse_init: Chip rev:         v0.1
I (316) heap_init: Initializing. RAM available for dynamic allocation:
I (323) heap_init: At 3FCA18D0 len 00047E40 (287 KiB): RAM
I (329) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (335) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (341) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (348) spi_flash: detected chip: gd
I (352) spi_flash: flash io: dio
W (356) spi_flash: Detected size(16384k) larger than the size in the binary image header(8192k). Using the size in the binary image header.
I (369) sleep: Configure to isolate all GPIO pins in sleep state
I (376) sleep: Enable automatic switching of GPIO sleep configuration
I (383) main_task: Started on CPU0
I (403) main_task: Calling app_main()
I (423) wifi station: ESP_WIFI_MODE_STA
I (433) pp: pp rom version: e7ae62f
I (433) net80211: net80211 rom version: e7ae62f
I (443) wifi:wifi driver task: 3fcabbac, prio:23, stack:6656, core=0
I (453) wifi:wifi firmware version: ccaebfa
I (453) wifi:wifi certification version: v7.0
I (453) wifi:config NVS flash: enabled
I (453) wifi:config nano formating: disabled
I (453) wifi:Init data frame dynamic rx buffer num: 32
I (463) wifi:Init static rx mgmt buffer num: 5
I (463) wifi:Init management short buffer num: 32
I (473) wifi:Init dynamic tx buffer num: 32
I (473) wifi:Init static tx FG buffer num: 2
I (473) wifi:Init static rx buffer size: 1600
I (483) wifi:Init static rx buffer num: 10
I (483) wifi:Init dynamic rx buffer num: 32
I (493) wifi_init: rx ba win: 6
I (493) wifi_init: accept mbox: 6
I (493) wifi_init: tcpip mbox: 32
I (503) wifi_init: udp mbox: 6
I (503) wifi_init: tcp mbox: 6
I (513) wifi_init: tcp tx win: 5760
I (513) wifi_init: tcp rx win: 5760
I (513) wifi_init: tcp mss: 1440
I (523) wifi_init: WiFi IRAM OP enabled
I (523) wifi_init: WiFi RX IRAM OP enabled
I (533) phy_init: phy_version 680,a6008b2,Jun  4 2024,16:41:10
W (573) phy_init: saving new calibration data because of checksum failure, mode(0)
I (593) wifi:mode : sta (68:b6:b3:21:d4:d0)
I (593) wifi:enable tsf
I (593) wifi station: wifi_init_sta finished.
I (603) wifi:new:<10,0>, old:<1,0>, ap:<255,255>, sta:<10,0>, prof:1, snd_ch_cfg:0x0
I (603) wifi:state: init -> auth (0xb0)
I (643) wifi:state: auth -> assoc (0x0)
I (693) wifi:state: assoc -> run (0x10)
I (3793) wifi:state: run -> init (0xf00)
I (3803) wifi:new:<10,0>, old:<10,0>, ap:<255,255>, sta:<10,0>, prof:1, snd_ch_cfg:0x0
I (3803) wifi station: retry to connect to the AP
I (3803) wifi station: connect to the AP fail
I (6213) wifi station: retry to connect to the AP
I (6213) wifi station: connect to the AP fail
I (6213) wifi:new:<10,0>, old:<10,0>, ap:<255,255>, sta:<10,0>, prof:1, snd_ch_cfg:0x0
I (6223) wifi:state: init -> auth (0xb0)
I (6263) wifi:state: auth -> assoc (0x0)
I (6293) wifi:state: assoc -> run (0x10)
I (9393) wifi:state: run -> init (0xf00)
I (9403) wifi:new:<10,0>, old:<10,0>, ap:<255,255>, sta:<10,0>, prof:1, snd_ch_cfg:0x0
I (9403) wifi station: retry to connect to the AP
I (9403) wifi station: connect to the AP fail
I (11813) wifi station: retry to connect to the AP
I (11813) wifi station: connect to the AP fail
I (11813) wifi:new:<10,0>, old:<10,0>, ap:<255,255>, sta:<10,0>, prof:1, snd_ch_cfg:0x0
I (11823) wifi:state: init -> auth (0xb0)
I (11863) wifi:state: auth -> assoc (0x0)
I (11893) wifi:state: assoc -> run (0x10)
I (14993) wifi:state: run -> init (0xf00)
I (15003) wifi:new:<10,0>, old:<10,0>, ap:<255,255>, sta:<10,0>, prof:1, snd_ch_cfg:0x0
I (15003) wifi station: retry to connect to the AP
I (15003) wifi station: connect to the AP fail
I (17413) wifi station: connect to the AP fail
I (17413) wifi station: Failed to connect to SSID:Xiaomi_76E5, password:123456
I (17413) main_task: Returned from app_main()

cannnnnnnnnnnn, 2024-10-11:
在实际的测试中，这些wifi:state信息会影响回调函数的输出

MaxwellAlan, 2024-10-11:
Hi @cannnnnnnnnnnn 

wifi:state 这些 log 是WiFi协议栈的内部连接过程状态变化的信息，由于是密码错误，所以log中会出现 
>I (14993) wifi:state: run -> init (0xf00)

如果想要关闭该log，可以通过调用 [esp_log_level_set("wifi", ESP_LOG_WARN); ](https://github.com/espressif/esp-idf/blob/release/v5.0/examples/wifi/iperf/main/cmd_wifi.c)来关闭Info log

cannnnnnnnnnnn, 2024-10-11:
HI @MaxwellAlan

感谢您的解答，在实际的测试中确实没有信息输出了
但我自己移植的代码中发现，设备重连的时间间隔不一致，而在输出wifi:state的信息中发现，在设备重连几次才会有一次回调这个是为什么呢？

以下是输出的log信息
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x178c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xcb8
load:0x403cb700,len:0x2db0
entry 0x403c8914
I (27) boot: ESP-IDF v5.3.1-dirty 2nd stage bootloader
I (27) boot: compile time Oct 11 2024 10:55:40
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (44) boot.esp32s3: SPI Flash Size : 16MB
I (49) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=25920h (153888) map
I (127) esp_image: segment 1: paddr=00035948 vaddr=3fc99600 size=04968h ( 18792) load
I (132) esp_image: segment 2: paddr=0003a2b8 vaddr=40374000 size=05d60h ( 23904) load
I (139) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=8b1dch (569820) map
I (244) esp_image: segment 4: paddr=000cb204 vaddr=40379d60 size=0f7a0h ( 63392) load
I (267) boot: Loaded app from partition at offset 0x10000
I (267) boot: Disabling RNG early entropy source...
I (279) cpu_start: Multicore app
I (288) cpu_start: Pro cpu start user code
I (289) cpu_start: cpu freq: 160000000 Hz
I (289) app_init: Application information:
I (291) app_init: Project name:     ESP_Synthesis
I (297) app_init: App version:      8498b1a-dirty
I (302) app_init: Compile time:     Oct 11 2024 10:55:14
I (308) app_init: ELF file SHA256:  a31d2b2a4...
I (314) app_init: ESP-IDF:          v5.3.1-dirty
I (319) efuse_init: Min chip rev:     v0.0
I (324) efuse_init: Max chip rev:     v0.99 
I (329) efuse_init: Chip rev:         v0.1
I (333) heap_init: Initializing. RAM available for dynamic allocation:
I (341) heap_init: At 3FCA1B50 len 00047BC0 (286 KiB): RAM
I (347) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (353) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (359) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (366) spi_flash: detected chip: gd
I (369) spi_flash: flash io: dio
I (374) sleep: Configure to isolate all GPIO pins in sleep state
I (380) sleep: Enable automatic switching of GPIO sleep configuration
I (388) main_task: Started on CPU0
I (418) main_task: Calling app_main()
I (438) pp: pp rom version: e7ae62f
I (438) net80211: net80211 rom version: e7ae62f
I (458) wifi:wifi driver task: 3fcabb10, prio:23, stack:6656, core=0
I (468) wifi:wifi firmware version: ccaebfa
I (468) wifi:wifi certification version: v7.0
I (468) wifi:config NVS flash: enabled
I (468) wifi:config nano formating: disabled
I (468) wifi:Init data frame dynamic rx buffer num: 32
I (478) wifi:Init static rx mgmt buffer num: 5
[... 198 lines cut ...]
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);

    /*Initialize WiFi */
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_APSTA));

    // 检查AP模式下是否初始化成功
    ESP_ERROR_CHECK(wifi_init_softap());

    // 检查STA模式下是否初始化成功
    ESP_ERROR_CHECK(wifi_init_sta());

    /* Start WiFi */
    ESP_ERROR_CHECK(esp_wifi_start());

    ESP_ERROR_CHECK(wifi_connect("Xiaomi_76E5","123456"));

    start_http_server();
}
```

cannnnnnnnnnnn, 2024-10-11:
```c
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x178c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xcb8
load:0x403cb700,len:0x2db0
entry 0x403c8914
I (27) boot: ESP-IDF v5.3.1-dirty 2nd stage bootloader
I (27) boot: compile time Oct 11 2024 10:55:40
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (44) boot.esp32s3: SPI Flash Size : 16MB
I (49) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=25920h (153888) map
I (127) esp_image: segment 1: paddr=00035948 vaddr=3fc99600 size=04968h ( 18792) load
I (132) esp_image: segment 2: paddr=0003a2b8 vaddr=40374000 size=05d60h ( 23904) load
I (139) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=8b1cch (569804) map
I (244) esp_image: segment 4: paddr=000cb1f4 vaddr=40379d60 size=0f7a0h ( 63392) load
I (267) boot: Loaded app from partition at offset 0x10000
I (267) boot: Disabling RNG early entropy source...
I (279) cpu_start: Multicore app
I (288) cpu_start: Pro cpu start user code
I (289) cpu_start: cpu freq: 160000000 Hz
I (289) app_init: Application information:
I (291) app_init: Project name:     ESP_Synthesis
I (297) app_init: App version:      8498b1a-dirty
I (302) app_init: Compile time:     Oct 11 2024 10:55:14
I (308) app_init: ELF file SHA256:  bf086050a...
I (314) app_init: ESP-IDF:          v5.3.1-dirty
I (319) efuse_init: Min chip rev:     v0.0
I (324) efuse_init: Max chip rev:     v0.99
I (329) efuse_init: Chip rev:         v0.1
I (333) heap_init: Initializing. RAM available for dynamic allocation:
I (341) heap_init: At 3FCA1B50 len 00047BC0 (286 KiB): RAM
I (347) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (353) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (359) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (366) spi_flash: detected chip: gd
I (369) spi_flash: flash io: dio
I (374) sleep: Configure to isolate all GPIO pins in sleep state
I (380) sleep: Enable automatic switching of GPIO sleep configuration
I (388) main_task: Started on CPU0
I (418) main_task: Calling app_main()
I (438) pp: pp rom version: e7ae62f
I (438) net80211: net80211 rom version: e7ae62f
I (458) wifi:wifi driver task: 3fcabb10, prio:23, stack:6656, core=0
I (468) wifi:wifi firmware version: ccaebfa
I (468) wifi:wifi certification version: v7.0
I (468) wifi:config NVS flash: enabled
I (468) wifi:config nano formating: disabled
I (468) wifi:Init data frame dynamic rx buffer num: 32
I (478) wifi:Init static rx mgmt buffer num: 5
I (478) wifi:Init management short buffer num: 32
I (488) wifi:Init dynamic tx buffer num: 32
I (488) wifi:Init static tx FG buffer num: 2
I (488) wifi:Init static rx buffer size: 1600
I (498) wifi:Init static rx buffer num: 10
I (498) wifi:Init dynamic rx buffer num: 32
[... 117 lines cut ...]
I (98828) wifi:new:<10,2>, old:<10,0>, ap:<10,2>, sta:<10,0>, prof:1, snd_ch_cfg:0x0
I (98828) wifi:state: init -> auth (0xb0)
I (98868) wifi:state: auth -> assoc (0x0)
I (98898) wifi:state: assoc -> run (0x10)
I (105018) wifi:state: run -> init (0xfc0)
I (105028) wifi:new:<10,0>, old:<10,2>, ap:<10,2>, sta:<10,0>, prof:1, snd_ch_cfg:0x0
I (105028) wifi:new:<10,2>, old:<10,0>, ap:<10,2>, sta:<10,0>, prof:1, snd_ch_cfg:0x0
I (105028) wifi:state: init -> auth (0xb0)
I (105068) wifi:state: auth -> assoc (0x0)
I (105098) wifi:state: assoc -> run (0x10)
I (111208) wifi:state: run -> init (0xfc0)
I (111228) wifi:new:<10,0>, old:<10,2>, ap:<10,2>, sta:<10,0>, prof:1, snd_ch_cfg:0x0
I (111228) wifi:new:<10,2>, old:<10,0>, ap:<10,2>, sta:<10,0>, prof:1, snd_ch_cfg:0x0
I (111228) wifi:state: init -> auth (0xb0)
I (111268) wifi:state: auth -> assoc (0x0)
I (111298) wifi:state: assoc -> run (0x10)
I (117408) wifi:state: run -> init (0xfc0)
I (117428) wifi:new:<10,0>, old:<10,2>, ap:<10,2>, sta:<10,0>, prof:1, snd_ch_cfg:0x0
I (117428) wifi station: retry to connect to the AP
I (117428) wifi station: connect to the AP fail
I (120258) wifi station: connect to the AP fail
I (120258) wifi station: Failed to connect to SSID:Xiaomi_76E5 password:123456
I (120258) http_server: Starting HTTP Server on port: '80'
I (120268) main_task: Returned from app_main()
```

MaxwellAlan, 2024-10-11:
Hi @cannnnnnnnnnnn 

>    wifi_config_t wifi_sta_config = {
        .sta = {
            .ssid = {0},
            .password = {0},
            .scan_method = WIFI_ALL_CHANNEL_SCAN,
            .failure_retry_cnt = MAXIMUM_RETRY,
            /* Authmode threshold resets to WPA2 as default if password matches WPA2 standards (password len => 8).
             * If you want to connect the device to deprecated WEP/WPA networks, Please set the threshold value
             * to WIFI_AUTH_WEP/WIFI_AUTH_WPA_PSK and set the password with length and format matching to
             * WIFI_AUTH_WEP/WIFI_AUTH_WPA_PSK standards.
             */
            .threshold.authmode = WIFI_AUTH_WPA2_PSK,
            .sae_pwe_h2e = WPA3_SAE_PWE_BOTH,
        }};

STA 默认的参考代码中有个  .failure_retry_cnt = MAXIMUM_RETRY , 在应用层发起wifi连接时，底层会自动在连接失败时提供重连机制，如果不需要的话可以把这个值设置成0.

cannnnnnnnnnnn, 2024-10-11:
Hi @MaxwellAlan 

感谢您的解答，在我的程序里这段我定义的数量是五，但是在第二段log中可以看到wifi:state的信息，设备重连多次后才会有一次回调任务，这个是为什么呢？

```c
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x178c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xcb8
load:0x403cb700,len:0x2db0
entry 0x403c8914
I (27) boot: ESP-IDF v5.3.1-dirty 2nd stage bootloader
I (27) boot: compile time Oct 11 2024 10:55:40
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (44) boot.esp32s3: SPI Flash Size : 16MB
I (49) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=25920h (153888) map
I (127) esp_image: segment 1: paddr=00035948 vaddr=3fc99600 size=04968h ( 18792) load
I (132) esp_image: segment 2: paddr=0003a2b8 vaddr=40374000 size=05d60h ( 23904) load
I (139) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=8b1cch (569804) map
I (244) esp_image: segment 4: paddr=000cb1f4 vaddr=40379d60 size=0f7a0h ( 63392) load
I (267) boot: Loaded app from partition at offset 0x10000
I (267) boot: Disabling RNG early entropy source...
I (279) cpu_start: Multicore app
I (288) cpu_start: Pro cpu start user code
I (289) cpu_start: cpu freq: 160000000 Hz
I (289) app_init: Application information:
I (291) app_init: Project name:     ESP_Synthesis
I (297) app_init: App version:      8498b1a-dirty
I (302) app_init: Compile time:     Oct 11 2024 10:55:14
I (308) app_init: ELF file SHA256:  bf086050a...
I (314) app_init: ESP-IDF:          v5.3.1-dirty
I (319) efuse_init: Min chip rev:     v0.0
I (324) efuse_init: Max chip rev:     v0.99
I (329) efuse_init: Chip rev:         v0.1
I (333) heap_init: Initializing. RAM available for dynamic allocation:
I (341) heap_init: At 3FCA1B50 len 00047BC0 (286 KiB): RAM
I (347) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (353) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (359) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (366) spi_flash: detected chip: gd
I (369) spi_flash: flash io: dio
I (374) sleep: Configure to isolate all GPIO pins in sleep state
I (380) sleep: Enable automatic switching of GPIO sleep configuration
I (388) main_task: Started on CPU0
I (418) main_task: Calling app_main()
I (438) pp: pp rom version: e7ae62f
I (438) net80211: net80211 rom version: e7ae62f
I (458) wifi:wifi driver task: 3fcabb10, prio:23, stack:6656, core=0
I (468) wifi:wifi firmware version: ccaebfa
I (468) wifi:wifi certification version: v7.0
I (468) wifi:config NVS flash: enabled
I (468) wifi:config nano formating: disabled
I (468) wifi:Init data frame dynamic rx buffer num: 32
I (478) wifi:Init static rx mgmt buffer num: 5
I (478) wifi:Init management short buffer num: 32
I (488) wifi:Init dynamic tx buffer num: 32
[... 121 lines cut ...]
I (98828) wifi:new:<10,2>, old:<10,0>, ap:<10,2>, sta:<10,0>, prof:1, snd_ch_cfg:0x0
I (98828) wifi:state: init -> auth (0xb0)
I (98868) wifi:state: auth -> assoc (0x0)
I (98898) wifi:state: assoc -> run (0x10)
I (105018) wifi:state: run -> init (0xfc0)
I (105028) wifi:new:<10,0>, old:<10,2>, ap:<10,2>, sta:<10,0>, prof:1, snd_ch_cfg:0x0
I (105028) wifi:new:<10,2>, old:<10,0>, ap:<10,2>, sta:<10,0>, prof:1, snd_ch_cfg:0x0
I (105028) wifi:state: init -> auth (0xb0)
I (105068) wifi:state: auth -> assoc (0x0)
I (105098) wifi:state: assoc -> run (0x10)
I (111208) wifi:state: run -> init (0xfc0)
I (111228) wifi:new:<10,0>, old:<10,2>, ap:<10,2>, sta:<10,0>, prof:1, snd_ch_cfg:0x0
I (111228) wifi:new:<10,2>, old:<10,0>, ap:<10,2>, sta:<10,0>, prof:1, snd_ch_cfg:0x0
I (111228) wifi:state: init -> auth (0xb0)
I (111268) wifi:state: auth -> assoc (0x0)
I (111298) wifi:state: assoc -> run (0x10)
I (117408) wifi:state: run -> init (0xfc0)
I (117428) wifi:new:<10,0>, old:<10,2>, ap:<10,2>, sta:<10,0>, prof:1, snd_ch_cfg:0x0
I (117428) wifi station: retry to connect to the AP
I (117428) wifi station: connect to the AP fail
I (120258) wifi station: connect to the AP fail
I (120258) wifi station: Failed to connect to SSID:Xiaomi_76E5 password:123456
I (120258) http_server: Starting HTTP Server on port: '80'
I (120268) main_task: Returned from app_main()
```

MaxwellAlan, 2024-10-11:
Hi @cannnnnnnnnnnn 

 .failure_retry_cnt 这个变量是底层自动重连的设置，在底层多次自动重连过程中，不会告知应用层连接失败的事件，如果不需要底层重连的机制，可以直接把  .failure_retry_cnt = 0; 配成0后让应用层根据事件来选择重连或者其他操作

cannnnnnnnnnnn, 2024-10-11:
Hi @MaxwellAlan 

好的，感谢您的解答

closed 2024-10-11 by cannnnnnnnnnnn