espressif/esp-idf #14711: ESP-TLS: esp_tls_conn_new_sync cannot open non-TLS connection

aenima1891, 2024-10-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm using ESP-IDF v5.3.1
In esp_tsl.h there is the esp_tls_conn_new_sync function

`/**
 * @brief      Create a new blocking TLS/SSL connection
 *
 * This function establishes a TLS/SSL connection with the specified host in blocking manner.
 *
 * @param[in]  hostname  Hostname of the host.
 * @param[in]  hostlen   Length of hostname.
 * @param[in]  port      Port number of the host.
 * @param[in]  cfg       TLS configuration as esp_tls_cfg_t. If you wish to open
 * non-TLS connection, keep this NULL. For TLS connection,
 *  a pass pointer to esp_tls_cfg_t. At a minimum, this
 *   structure should be zero-initialized.
 * @param[in]  tls       Pointer to esp-tls as esp-tls handle.
 *
 * @return
 *   -1      If connection establishment fails.
 *   1      If connection establishment is successful.
 *   0      If connection state is in progress.
 */
int esp_tls_conn_new_sync(const char *hostname, int hostlen, int port, const esp_tls_cfg_t *cfg, esp_tls_t *tls);
`

if I pass a cfg=NULL param, the function return -1.
In fact, the function implementation is

int esp_tls_conn_new_sync(const char *hostname, int hostlen, int port, const esp_tls_cfg_t *cfg, esp_tls_t *tls)
{
    if (!cfg || !tls || !hostname || hostlen < 0) {
        return -1;
    }
    uint64_t start_time_us;
    start_time_us = esp_tls_get_platform_time();
    ...

There is a mismatch between the header doc and the func implementation.

AdityaHPatwardhan, 2024-11-12:
Hi @aenima1891 Did you try providing a zero initialised cfg parameter instead of just passing `cfg = NULL`

AdityaHPatwardhan, 2024-12-16:
Hi @aenima1891 I am closing the issue due to lack of response.
Please feel free to re-open if you are still facing the issue.
Thanks,
Aditya

closed 2024-12-16 by AdityaHPatwardhan