espressif/esp-idf #14712: nimBLE compilation error when CONFIG_BT_NIMBLE_MAX_BONDS == 0

KaeLL, 2024-10-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Setting `CONFIG_BT_NIMBLE_MAX_BONDS` to 0 (a valid value) results in
```
/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config.c: In function 'ble_store_config_delete_local_irk':
/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config.c:732:10: error: implicit declaration of function 'ble_store_config_delete_obj'; did you mean 'ble_store_config_delete_cccd'? [-Wimplicit-function-declaration]
  732 |     rc = ble_store_config_delete_obj(ble_store_config_local_irks,
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |          ble_store_config_delete_cccd
/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config.c: In function 'ble_store_config_delete_rpa_rec':
/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config.c:878:11: error: implicit declaration of function 'ble_store_config_find_rpa_rec'; did you mean 'ble_store_config_read_rpa_rec'? [-Wimplicit-function-declaration]
  878 |     idx = ble_store_config_find_rpa_rec(key_rpa_rec);
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |           ble_store_config_read_rpa_rec
```

Also, almost entirely unrelated:
https://github.com/espressif/esp-idf/blob/6e5a178b3120dced7fa5c29c655cc22ea182df3d/components/bt/host/nimble/Kconfig.in#L172

rahult-github, 2024-10-14:
Hi @KaeLL , 

Thanks for reporting. The attached patch should help fix the issue . You can give it a try.  The patch goes in $IDF_PATH/components/bt/host/nimble/nimble path 

[fix_compilation_issue.txt](https://github.com/user-attachments/files/17361817/fix_compilation_issue.txt)

> Also, almost entirely unrelated:

Noted. will fix this too.

rahult-github, 2025-01-30:
Changes merged via commit https://github.com/espressif/esp-idf/commit/f8ccb29d1210d35f23257d20cac1f040a0662b86

closed 2025-01-30 by rahult-github