espressif/esp-idf #14716: OTA时，使用电信4g的手机给esp32c3开热点，无法通过http来下载固件，会报超时错误

iamugly, 2024-10-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Espressif SoC revision.

ESP32-C3 chip revision: v0.3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C3-DevKitc-02v1.1

### Power Supply used.

External 5V

### What is the expected behavior?

我使用OTA的例程，将esp连接到我手机的热点，当我用手机的电信5g时，ota的过程是正常的。但当我将手机的流量改为电信4g，重新运行ota，这时就无法下载固件，根据log看连接上http以后，下载数据会一直报超时。我尝试过修改esp_http_client_config_t结构体里的timeout_ms参数，但依然会报超时，改大以后只是等待超时的时间更久了。

### What is the actual behavior?

无法下载下来固件

### Steps to reproduce.

1. 手机用电信4g流量，开热点
2. esp连接上热点
3. 开始ota
4. http连接上以后下载固件包下载不下来
...

### Debug Logs.

```plain
以下是开始OTA时的log
simple_ota_example: Starting logger OTA task
I (64689) simple_ota_example: Attempting to download update from http://smartenergy.csisolar.com/device/firmware/2024/10/12/1844985603830824961/be62fc8cfaf1456ea2f735bf6d8f5731.bin
D (64709) HTTP_CLIENT: Begin connect to: http://smartenergy.csisolar.com:80
D (64719) esp-tls: host:smartenergy.csisolar.com: strlen 24
D (64919) esp-tls: [sock=55] Resolved IPv4 address: 117.85.70.238
D (64919) esp-tls: Enable TCP keep alive. idle: 5, interval: 5, count: 3
D (64919) esp-tls: [sock=55] Connecting to server. HOST: smartenergy.csisolar.com, Port: 80
simple_ota_example: HTTP_EVENT_ON_CONNECTED
D (65119) HTTP_CLIENT: Write header[3]: GET /device/firmware/2024/10/12/1844985603830824961/be62fc8cfaf1456ea2f735bf6d8f5731.bin HTTP/1.1
User-Agent: ESP32 HTTP Client/1.0
Host: smartenergy.csisolar.com
Content-Length: 0

D (65129) simple_ota_example: HTTP_EVENT_HEADER_SENT
 HTTP_CLIENT: on_message_begin
D (65649) HTTP_CLIENT: HEADER=Server:Tengine
D (65649) simple_ota_example: HTTP_EVENT_ON_HEADER, key=Server, value=Tengine
D (65649) HTTP_CLIENT: HEADER=Content-Type:application/octet-stream
D (65659) simple_ota_example: HTTP_EVENT_ON_HEADER, key=Content-Type, value=application/octet-stream
D (65659) HTTP_CLIENT: HEADER=Content-Length:1139424
D (65669) simple_ota_example: HTTP_EVENT_ON_HEADER, key=Content-Length, value=1139424
D (65679) HTTP_CLIENT: HEADER=Connection:keep-alive
D (65679) simple_ota_example: HTTP_EVENT_ON_HEADER, key=Connection, value=keep-alive
D (65689) HTTP_CLIENT: HEADER=Date:Sat, 12 Oct 2024 06:56:39 GMT
D (65689) simple_ota_example: HTTP_EVENT_ON_HEADER, key=Date, value=Sat, 12 Oct 2024 06:56:39 GMT
D (65699) HTTP_CLIENT: HEADER=x-oss-request-id:670A1DA757D514373724D5F6
D (65709) simple_ota_example: HTTP_EVENT_ON_HEADER, key=x-oss-request-id, value=670A1DA757D514373724D5F6
D (65719) HTTP_CLIENT: HEADER=x-oss-cdn-auth:success
D (65719) simple_ota_example: HTTP_EVENT_ON_HEADER, key=x-oss-cdn-auth, value=success
D (65729) HTTP_CLIENT: HEADER=Accept-Ranges:bytes
D (65729) simple_ota_example: HTTP_EVENT_ON_HEADER, key=Accept-Ranges, value=bytes
D (65739) HTTP_CLIENT: HEADER=x-oss-object-type:Normal
D (65749) simple_ota_example: HTTP_EVENT_ON_HEADER, key=x-oss-object-type, value=Normal
D (65749) HTTP_CLIENT: HEADER=x-oss-storage-class:Standard
D (65759) simple_ota_example: HTTP_EVENT_ON_HEADER, key=x-oss-storage-class, value=Standard
D (65769) HTTP_CLIENT: HEADER=Content-MD5:4/dGBVwigJKwF6FqRfwQBQ==
D (65769) simple_ota_example: HTTP_EVENT_ON_HEADER, key=Content-MD5, value=4/dGBVwigJKwF6FqRfwQBQ==
D (65779) HTTP_CLIENT: HEADER=x-oss-server-time:28
D (65789) simple_ota_example: HTTP_EVENT_ON_HEADER, key=x-oss-server-time, value=28
D (65789) HTTP_CLIENT: HEADER=Via:cache14.l2cn1833[153,154,304-0,H], cache21.l2cn1833[155,0], cache6.cn7596[252,263,200-0,H], cache8.cn7596[266,0]
D (65809) simple_ota_example: HTTP_EVENT_ON_HEADER, key=Via, value=cache14.l2cn1833[153,154,304-0,H], cache21.l2cn1833[155,0], cache6.cn7596[252,263,200-0,H], cache8.cn7596[266,0]
HTTP_CLIENT: HEADER=Vary:Origin
D (65829) simple_ota_example: HTTP_EVENT_ON_HEADER, key=Vary, value=Origin
D (65839) HTTP_CLIENT: HEADER=ETag:"E3F746055C228092B017A16A45FC1005"
D (65849) simple_ota_example: HTTP_EVENT_ON_HEADER, key=ETag, value="E3F746055C228092B017A16A45FC1005"
D (65859) HTTP_CLIENT: HEADER=Last-Modified:Sat, 12 Oct 2024 06:17:00 GMT
D (65859) simple_ota_example: HTTP_EVENT_ON_HEADER, key=Last-Modified, value=Sat, 12 Oct 2024 06:17:00 GMT
D (65869) HTTP_CLIENT: HEADER=x-oss-hash-crc64ecma:956103214857714626
D (65879) simple_ota_example: HTTP_EVENT_ON_HEADER, key=x-oss-hash-crc64ecma, value=956103214857714626
D (65889) HTTP_CLIENT: HEADER=Age:0
D (65889) simple_ota_example: HTTP_EVENT_ON_HEADER, key=Age, value=0
D (65909) HTTP_CLIENT: HEADER=Ali-Swift-Global-Savetime:1728716199
D (65909) simple_ota_example: HTTP_EVENT_ON_HEADER, key=Ali-Swift-Global-Savetime, value=1728716199
D (65919) HTTP_CLIENT: HEADER=X-Cache:HIT TCP_REFRESH_HIT dirn:2:868419196
D (65929) simple_ota_example: HTTP_EVENT_ON_HEADER, key=X-Cache, value=HIT TCP_REFRESH_HIT dirn:2:868419196
D (65939) HTTP_CLIENT: HEADER=X-Swift-SaveTime:Sat, 12 Oct 2024 06:56:39 GMT
D (65939) simple_ota_example: HTTP_EVENT_ON_HEADER, key=X-Swift-SaveTime, value=Sat, 12 Oct 2024 06:56:39 GMT
D (65949) HTTP_CLIENT: HEADER=X-Swift-CacheTime:237
D (65959) simple_ota_example: HTTP_EVENT_ON_HEADER, key=X-Swift-CacheTime, value=237
D (65959) HTTP_CLIENT: HEADER=Timing-Allow-Origin:*
D (65969) simple_ota_example: HTTP_EVENT_ON_HEADER, key=Timing-Allow-Origin, value=*
D (65979) HTTP_CLIENT: HEADER=EagleId:7555469c17287161992297927e
D (65979) simple_ota_example: HTTP_EVENT_ON_HEADER, key=EagleId, value=7555469c17287161992297927e
D (65989) HTTP_CLIENT: http_on_headers_complete, status=200, offset=864, nread=864
D (65999) HTTP_CLIENT: http_on_body 496
I (65999) HTTP_CLIENT: Body received in fetch header state, 0x3fcca46c, 496
D (66009) simple_ota_example: HTTP_EVENT_ON_DATA, len=496
D (66009) HTTP_CLIENT: content_length = ld
I (66019) esp_https_ota: Starting OTA...
I (66019) esp_https_ota: Writing to partition subtype 17 at offset 0x190000
D (66029) HTTP_CLIENT: Data processed ld != Data specified in content length ld
D (66029) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=ld
D (66049) HTTP_CLIENT: need_read=528, byte_to_read=528, rlen=528, ridx=496
D (66049) HTTP_CLIENT: http_on_body 528
D (66049) simple_ota_example: HTTP_EVENT_ON_DATA, len=528
D (66129) esp_https_ota: Written image length 2048
D (66129) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=ld
D (66129) HTTP_CLIENT: need_read=4096, byte_to_read=4096, rlen=4096, ridx=0
D (66129) HTTP_CLIENT: http_on_body 4096
D (66139) simple_ota_example: HTTP_EVENT_ON_DATA, len=4096
D (66219) esp_https_ota: Written image length 6144
D (66219) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=ld
D (66219) HTTP_CLIENT: need_read=4096, byte_to_read=4096, rlen=1648, ridx=0
D (66229) HTTP_CLIENT: http_on_body 1648
D (66229) simple_ota_example: HTTP_EVENT_ON_DATA, len=1648
D (66229) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=ld

这里等待五秒后就会报select超时
D (81249) transport_base: poll_read: select - Timeout before any socket was ready!
D (81249) HTTP_CLIENT: need_read=2448, byte_to_read=2448, rlen=0, ridx=1648
D (81249) HTTP_CLIENT: Connection timed out before data was ready!
D (81259) esp_https_ota: Written image length 7792
D (81259) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=ld
```

Alvin1Zhang, 2024-10-14:
Thanks for reporting.

nileshkale123, 2024-10-22:
Hello @iamugly ,

Thank you for reaching out with the details about the OTA timeout issue you're facing on the 4G network.

We understand that OTA works on the 5G network but encounters timeouts on 4G, even after adjusting the timeout settings. This could potentially be related to network-specific factors such as latency, packet loss, or bandwidth limitations, but to investigate this further, we would need more information.

To assist you better, could you please provide the following:

 - A small test example (or a patch file) that we can use to reproduce the issue on our side. If the issue can be recreated using the standard OTA example in ESP-IDF, it would be helpful to have any specific modifications you made.
 - The sdkconfig file you are using for your project to help us understand the configuration in detail.

We will review this information and work on reproducing and diagnosing the issue. Additionally, if you have any logs or further details on the error, they could provide valuable insights.

Looking forward to your response.

iamugly, 2024-10-25:
> Hello @iamugly ,
>
> Thank you for reaching out with the details about the OTA timeout issue you're facing on the 4G network.
>
> We understand that OTA works on the 5G network but encounters timeouts on [...]

Hello @nileshkale123 ，
Thanks for your reply.

After my attempts, I found that this is not a problem caused by the difference between 4g and 5g, but only the telecom 4g operator will have this problem. When I used a 4g phone from another carrier to turn on the esp hotspot, the ota function was also normal.

I have the same problem with "simple_ota_example",Here is my sdkconfig file.
[sdkconfig.txt](https://github.com/user-attachments/files/17517703/sdkconfig.txt)

I also used wireshark to capture the upgrade package, which may be helpful for you to analyze the problem
[simple_ota.zip](https://github.com/user-attachments/files/17517795/simple_ota.zip)

nileshkale123, 2024-10-25:
Hello @iamugly ,

Could you try applying the following patch and let me know your observations? If the issue persists, please share the detailed debug logs.
[quick_patch.txt](https://github.com/user-attachments/files/17518662/quick_patch.txt)

iamugly, 2024-10-28:
> Hello @iamugly ,
> 
> Could you try applying the following patch and let me know your observations? If the issue persists, please share the detailed debug logs. [quick_patch.txt](https://github.com/user-attachments/files/17518662/quick_patch.txt)

Hi @nileshkale123 
I used your patch, but the problem persists. The following is debug logs
[ota_log.zip](https://github.com/user-attachments/files/17540244/ota_log.zip)

nileshkale123, 2024-11-29:
Hello @iamugly ,

Sorry for the late reply. 

Could you please try one more time by increasing client timeout [at](https://github.com/espressif/esp-idf/blob/master/examples/system/ota/simple_ota_example/main/simple_ota_example.c#L92). You can set timeout in  [this](https://github.com/espressif/esp-idf/blob/master/examples/system/ota/advanced_https_ota/main/advanced_https_ota_example.c#L130) way.

iamugly, 2024-11-29:
> Hello @iamugly ,
>
> Sorry for the late reply.
>
> Could you please try one more time by increasing client timeout [...]

[image]

I changed the timeout but still couldn't download

[image]

nileshkale123, 2024-11-29:
Hello @iamugly ,

I successfully downloaded your hosted OTA image without any issues on my side with simple_ota_example.

However, when I used your provided sdkconfig file, I encountered a different issue than you mentioned during the OTA process. Could you please create a minimal application in IDF by making the necessary modifications to the simple_ota_example to reproduce the issue? It would be greatly helpful if you could provide this reproducible setup along with all the required files.

nileshkale123, 2024-12-05:
Hello @iamugly,

Please provide us with the minimal application to reproduce the issue and debug it further as instructed in the above reply.
Thanks

nileshkale123, 2024-12-10:
Hello @iamugly,

I am closing this issue for now. Please feel free to reopen if needed.

Thank you.

closed 2024-12-10 by nileshkale123