espressif/esp-idf #14717: Plz add pixiwps libary plzz

Rahad1069, 2024-10-12:


Rahad1069, 2024-10-12:
We want to perform pixie dust attack for cracking wps pin by esp32/esp8266. So that's why, we need the pixiewps libary

kapilkedawat, 2024-10-14:
Hi @Rahad1069, this is not in our roadplan and will not be part of ESP-IDF.

Sherry616, 2024-10-14:
Thanks for reporting, we will close this issue. 
Thanks for using our Espressif product!

closed 2024-10-14 by Sherry616

Naster17, 2025-03-10:
> ### Is your feature request related to a problem?
> 
> _No response_
> 
> ### Describe the solution you'd like.
> 
> _No response_
> 
> ### Describe alternatives you've considered.
> 
> _No response_
> 
> ### Additional context.
> 
> _No response_

Oh no cringe

Naster17, 2025-03-10:
> We want to perform pixie dust attack for cracking wps pin by esp32/esp8266. So that's why, we need the pixiewps libary

Oh no cringe