espressif/esp-idf #14719: BLE_MESH_SUPPORT_BLE_ADV sends incorrect data since v5.2.2 for both Bluedroid and NimBLE on ESP32

buganini, 2024-10-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Last working version: v5.2.1
Target: ESP32
Bluetooth Stack: Both Bluedroid and Nimble

With v5.2.1 I can see both adv_data and scan_rsp_data in Android nRF Connect app.
With v5.2.2 I can see only scan_rsp_data in Android nRF Connect app.

forx157, 2024-10-14:
Hi, @buganini 

Are you saying that the issue only occurs when using ESP32 with IDFv5.2.2? That other chips with different IDF versions do not have any problems?

buganini, 2024-10-14:
I've only tested on ESP32/ESP32-C3 with different code base, but roughly yes, I think to problem is only on ESP32 with v5.2.2+

buganini, 2024-10-14:
BTW, on ESP32-C3, I have both legacy/ext adv working fine. ESP32 can only do legacy adv, the issue may be related to switching multiple adv.

closed 2024-10-14 by buganini