espressif/esp-idf #14720: WiFi Mesh unstable when parent offline

michaelsimp, 2024-10-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.0

### Espressif SoC revision.

Chip is ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

I expect the ESP32 to continue to run the application without crashing when the WIFI Mesh parent disappears.
If the MESH_ROOT was powered off, I expect a MESH_NODE to assume the role of MESH_ROOT
If the WIFI Router is powered off, when I restore it, I expect the mesh network to establish itself

### What is the actual behavior?

1. Sometimes these tests work perfectly.  The Mesh network goes down and the nodes start scanning. If I restore the WiFi router,  the Mesh network is reestablished
2. Sometimes the Mesh network goes down, and can't recover. It doesn't crash but it doesn't scan properly and reestablish the Mesh network.
3. Very regularly, if I power off the WiFi router the MESH_ROOT intermittently crashes OR if I power off the MESH_ROOT a MESH_NODE intermittently crash

### Steps to reproduce.

1. Power on system comprising 2 x ESP32-S3 dev boards and a Wifi router
2. Connect a serial terminal (I am using PUTTY) to each serial port for monitoring
3. Let the Mesh network get established and verify MESH_ROOT and MESH_NODE connected.
4. Power off the WiFi Router
5. In the example (logs below) the MESH_ROOT crashed

### Debug Logs.

```plain
I (00:58:03.336) aWifiMesh: <MESH_EVENT_MESH_STARTED>ID:77:77:77:77:77:76
I (136546) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x3, need_scan_router:0x0, look_for_nwk_count:1
I (00:58:03.336) aWifiMesh: This node MAC:48:ca:43:9b:53:d8
I (00:58:03.354) aWifiMesh: WiFi Mesh started successfully, heap:141084, root not fixed
WIN> I (140766) mesh: [S6]VONETS, 00:17:13:20:bd:74, channel:8, rssi:-12
I (140776) mesh: find router:[ssid_len:6]VONETS, rssi:-12, 00:17:13:20:bd:74(encrypted), new channel:8, old channel:0
I (140786) mesh: [FIND][ch:0]AP:11, otherID:0, MAP:1, idle:1, candidate:0, root:0[00:17:13:20:bd:74]router found
I (140796) mesh: [FIND:1]find a network, channel:8, cfg<channel:0, router:VONETS, 00:00:00:00:00:00>

I (00:58:07.590) aWifiMesh: <MESH_EVENT_FIND_NETWORK>new channel:8, router BSSID:00:00:00:00:00:00
W (140796) wifi:<MESH AP>adjust channel:1, secondary channel offset:1(40U)
W (140816) wifi:<MESH AP>adjust channel:8, secondary channel offset:1(40U)
I (141126) mesh: [SCAN][ch:8]AP:1, other(ID:0, RD:0), MAP:0, idle:0, candidate:1, root:0, topMAP:0[c:0,i:0][00:17:13:20:bd:74]router found<>
[... 76 lines cut ...]
I (1324184) mesh: [SCAN][ch:8]AP:2, other(ID:0, RD:0), MAP:1, idle:0, candidate:1, root:1, topMAP:0[c:0,i:0][00:17:13:20:bd:74]router found<>
I (1324184) mesh: 7391[selection]try rssi_threshold:-78, backoff times:0, max:5<-78,-82,-85>
I (1324204) mesh: [DONE]connect to parent:ESPM_3372B8, channel:8, rssi:-14, 30:30:f9:33:72:b9[layer:1, assoc:0], my_vote_num:0/voter_num:0, rc[00:00:00:00:00:00/-120/0]
I (1324214) mesh: set router bssid:00:17:13:20:bd:74
I (1324834) mesh: <MESH_NWK_MIE_CHANGE><><><><ROOT ADDR><><><>
I (1324834) mesh: <MESH_NWK_ROOT_ADDR>from assoc, layer:2, root_addr:30:30:f9:33:72:b9, root_cap:1
I (1324844) mesh: <MESH_NWK_ROOT_ADDR>idle, layer:2, root_addr:30:30:f9:33:72:b9, conflict_roots.num:0<>
I (1324854) mesh: [scan]new scanning time:600ms, beacon interval:300ms
I (00:22:03.744) aWifiMesh: <MESH_EVENT_ROOT_ADDRESS>root address:30:30:f9:33:72:b9
I (1324854) mesh: 2012<arm>parent monitor, my layer:2(cap:6)(node), interval:4526ms, retries:1<normal connected>
I (00:22:03.771) aWifiMesh: <MESH_EVENT_PARENT_CONNECTED>layer:2-->2, parent:30:30:f9:33:72:b9<layer2>, ID:77:77:77:77:77:76
I (00:22:03.785) mesh_netif: It was a wifi station removing stuff
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4212753c  PS      : 0x00060830  A0      : 0x82127613  A1      : 0x3fcc15d0
A2      : 0xffffffff  A3      : 0x00000000  A4      : 0x00000278  A5      : 0x00000001
A6      : 0x3fcc09d0  A7      : 0x00000278  A8      : 0x3c1505e4  A9      : 0x3fcd778c
A10     : 0x3fcc09d0  A11     : 0x00000000  A12     : 0x00000101  A13     : 0x3c1505e4
A14     : 0x00000007  A15     : 0x3fcaa7f4  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000284  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff

Backtrace: 0x42127539:0x3fcc15d0 0x42127610:0x3fcc1620 0x4037e0aa:0x3fcc1640

I sometimes get MTX task stack overflows too when I try this, same as https://github.com/espressif/esp-idf/issues/13882

zhangyanjiaoesp, 2024-10-22:
@michaelsimp 
I have tested using the `ip_internal_network` example, but I didn't reproduce your problem. Can you provide the `.elf` file when the crash issue happen? Or can you provide the core dump decode file?

michaelsimp, 2024-10-24:
Hi
      
      Thanks for the response.
      
      I have a deadline for a project demo this week but I will go back
      and reinstall from scratch and rebuild and test and send you the
      files requested.
      
      One thing I note is that I originally installed IDF into vscode
      when 5.2.2 was the current build.  I have since upgraded to ver
      5.3.0 and selected this and did a clean build on my  project.  
      
      Now I have gone back to "show examples" to recreate the
      ip_internal_network project from scratch, it asks which version of
      the IDF I want to use but only lists ver 5.2.2.  Only once I
      create the project I can change the version of IDF to 5.3.0
      
      Using winmerge I compared the directory structure and files from
      my target folder containing ip_internal_network
      from ver 3.2.2, with the
      ...\esp\v5.3\esp-idf\examples\mesh\ip_internal_network folder and
      the only change was in the partition.csv which now must be
      aligned.
      
      I also note there is now a version 5.3.1 marked as stable, and
      5.3.0 has gone?  I also found migration notes for going from 5.3
      to 5.4 although I can't find this in the installer. Is 5.4 the
      master (development branch).
      
      
      
      Which version would you recommend I use?  I have found everything
      in IDF to be stable and bug free except for the WiFi Mesh.
    
      Could you advise how I update IDF properly so it will enable me to
      select latest version of IDF when I create project from "Show
      Examples".
      Last time I followed the instructions from Visual Studio at:
https://marketplace.visualstudio.com/items?itemName=espressif.esp-idf-extension#:~:text=In%20Visual%20Studio%20Code%2C%20select,not%20supported%20inside%20configured%20paths
      
      Regards
      Michael
      
    
    On 22/10/2024 10:57 pm, ZYJ wrote:
    
    
      
      @michaelsimp
        I have tested using the ip_internal_network
        example, but I didn't reproduce your problem. Can you provide
        the .elf file when the crash
        issue happen? Or can you provide the core dump decode file?
      —
      [
{
"target": "https://github.com/espressif/esp-idf/issues/14720#issuecomment-2428832704",
"url": "https://github.com/espressif/esp-idf/issues/14720#issuecomment-2428832704",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]
    
    
  
[... 247 lines cut ...]
DNHYzhIIWcdb0EuOs6BfJj9f27ZPfhGNdBJiUCSH/bU0ARYfiWmymMKtfF45ARIXsoEsjGYp
q3Ppa5J15WSNmAhKGvdJmtkstWut/5xCNFlmuPJ9hhNSxtC0DdQXmmr6G7eYytypxCNZHZxY
Cmyd0ocRnz238apzbRq/2rZcxrLtMeJsqOM0VqNmNvJlpUQT5EkIMiEQsSwIRV9B0BehSWnN
/gzNbBaeBslvypOJoa0TTfpE0mcjfpI7OXjjp6tENBlbKZU81S2MIAi3euSFyZRwhA5zh7ND
fTsxRZ1KpJul2NL32UQj+xPkhaNve/igLVtGVSanxZflE1O+z2krW9/MbesPP8XSFZXN2K+B
aIwPlEY4xKnTdhMN38YIJGI2KQUCyS2bRQYClt0gGsUS0mZSVDLVcjJYUmp9H4lGsyIVobP9
SsKdk7qJxqpvIJos1qmYaKLcT92yaEykkrt3VRQxa+uUe5mtJNFkmezJHt6p/ZQQv3uBstVz
qC8v2iJbJ/0imdxWZxaNi7+hdqJRcTOQll3fHIjGdmLkatE4kaqTsdGVQq0mmuRkgO/RVUdt
8PdfL36B3spPRVxM7XjrIXn5xXYcFnI4cybT1tB2ppOT5NMz2QkZb80uCJzngQP9u5LzM7be
cuo7O4MFZ2yEtXD6wsaojaczoomwOpVSpzJrPCDAm4neGt9ediIap7kxOWwNmJkWsvqddjDg
QjTxll1yvqvOYJXQ9YeBCzl3hUEcO1GJZnBwIKiZJpNjzSQJ5NgfWdkq+ffdtGi4HyE5Bg58
M3f8eJiu4Xdj1KNI7chT2FKJp0ian0M9abFdwosXiumug9L2gV1P0+FL1Zuquq9IPkZO2x81
+J74XRF+vMxK555qBb+np0Ox5qjt8kXGj3vjYvKxa4dEkxCzcE1BOKVK5ll9bcJABqn/xHa8
LR/Bq8fp2kMsHq/kn1H1LeOwItMSSo7phflxtGjafLzNSGZgYDBElGeudCYa7gjulGgcibFV
xdqmDLngtsxUb5WiFBa23Rf2wjzbYWrckF7C/y9ANJG1A6IRtMbJlC+sZc1UANE0g7uh17Zt
m9gQ+NaJWzMgmorVyf5SZcUd1tUciKYuZIu1a7xnU6yJJkozotm2bVu4ZeLbpcSScbFomNCi
f4b9vXr1ewrfo2li8OizAAIgmgJgoaiKQBbR8O2TdesUNRhd1OP//Y3feDd97cXfBtpAAAgA
gRCB9576Zdq+/R7JmhEdwRrRiBaMiKFo1bzznf+MhoeHadYs/vIb0AYCQKBfETh+/DgdO3aM
7rvvL6slGk5G3DmcHolzy6dfIce4gUDvIyBuh0JLRThdSv0z4S/JiZPRonGxaiJyiYgFRNP7
yoURAgGOgEo0EdnIx9maI5j9Pjp6UeiBMW2VlG8lYlGdxFHZqKnIp4MPEAACvYBAfHAUWinq
R3L4hoQT35tJK0VEZCIam1Uj/q7eJu4FYDEGIAAE3BDgxMJKZ5FM+FsW0RQhGzeRUAoIAIFe
RMBENnycxpvBJhCyLBa7JYP9Uy8qFcbUbwjo2yUTAiLZpCQT2jLhn/8fbOZ0e5NBKGYAAAAA
SUVORK5CYII=

--------------UEln60op0cSX066h04nQooA4--

michaelsimp, 2024-10-24:
Hi
I did a quick set of tests today.

I created the ip_internal_network project from examples and configured as follows:
Set IDF version to 5.3.0
Set target device to ESP32-S3 with jtag integrated debugger
Set partition table Factory partition to 0x400000
Set device flash size to 16MB - matches my ESP32-S3
Set the Router SSID to "VONETS" and Password to "pass9999"
Set Panic Handler to "Print registers and halt"

Clean and Build project.
Load into 2 ESP32-S3 with serial terminals connected for monitoring.
One becomes MESH_ROOT and the other connects as MESH_NODE
Took turns at powering off the MESH_ROOT and watching the other become MESH_ROOT and then powering it back on and it connects as a MESH_NODE. This seemed to work ok today.

But what I did find easy to reproduce was:
Power off MESH_ROOT and power back on BEFORE other MESH_NODE becomes MESH_ROOT.
The original MESH_ROOT I power cycled, becomes MESH_ROOT again, but the MESH_NODE remains disconnected.

See attached files:
MESH_ROOT powered off at line 171
MESH_NODE loses connection around line 33 and never recovers

Also see .elf and .bin files in attachment.  I don't know what or where the "Core dump decode file is", but these tests don't show a CPU crash.  I cant run under the debugger due to the power cycle tests.

michaelsimp, 2024-10-24:
I did some more tests which can easily cause CPU crashes.

Power on both nodes, one becomes MESH_ROOT and one MESH_NODE.
Power off WIFI router.
MESH_ROOT crashes. See file attached MESH_ROOT "Router power off.txt" (crash at end)

Second test. Power up only one node, becomes MESH_ROOT
Power off Router
This time MESH_ROOT does not crash
Power on Router
MESH_ROOT does not crash
Power on a second node which connects to the first MESH_ROOT - see line 492
MESH_ROOT crashes.
See file attached "MESH_ROOT crash on MESH_NODE connect.txt"
[MESH_ Testing 2.zip](https://github.com/user-attachments/files/17513316/MESH_.Testing.2.zip)

michaelsimp, 2024-10-24:
[MESH-Testing.zip](https://github.com/user-attachments/files/17513348/MESH-Testing.zip)
This is the attachment for the first tests, 2 entries up. It did not upload properly last before

brianignacio5, 2024-10-29:
Hi @michaelsimp 

The esp-idf vscode extension allows you to save settings in multiple places: User (Global settings for vscode), Workspace and Workspace folder (your project's .vscode/settings.json). The `ESP-IDF: Show Examples` command shows you the current esp-idf path used in the current vscode window. You can change where to save settings with the `ESP-IDF: Select where to save configuration settings` command. It sounds confusing but it does allow to use multiple projects each with different esp-idf versions (even at the same time! Using vscode workspace) More information in [here](https://docs.espressif.com/projects/vscode-esp-idf-extension/en/latest/additionalfeatures/multiple-projects.html)

It seems the example you are trying to use have some components with specific behavior in each esp-idf version. So building a v5.2.2 example using esp-idf v5.3 might produce some compilation problems. How about creating an example using esp-idf v5.3 ?

1. Open a vscode window.
2. Select esp-idf v5.3 from status bar (recommended) or the `ESP-IDF: Configure ESP-IDF extension`.
3. Run the `ESP-IDF: Doctor command`. Check that esp-idf is indeed using v5.3
4. Run the `ESP-IDF: Show examples`. The esp-idf path shown should be v5.3 now
5. Create your project from esp-idf example and try to build.

We will try to update the Show examples command to show all available esp-idf versions from esp-idf vscode extension to make this easier.

zhangyanjiaoesp, 2024-10-29:
@michaelsimp The backtrace of the crash issue is here:

```
xtensa-esp32s3-elf-addr2line -piaf 0x4208c922:0x3fca7e60 0x4201c951:0x3fca7ee0 0x4201f96f:0x3fca7f20 0x4202345e:0x3fca7f50 0x420167bd:0x3fca7f70 -e ip_internal_network.elf

0x4208c922: parse_msg at C:\Users\micha\Documents\Cybertek\Software\ip_internal_network\build/C:/Users/micha/esp/v5.3/esp-idf/components/lwip/apps/dhcpserver/dhcpserver.c:993
 (inlined by) handle_dhcp at C:\Users\micha\Documents\Cybertek\Software\ip_internal_network\build/C:/Users/micha/esp/v5.3/esp-idf/components/lwip/apps/dhcpserver/dhcpserver.c:1190
 (inlined by) handle_dhcp at C:\Users\micha\Documents\Cybertek\Software\ip_internal_network\build/C:/Users/micha/esp/v5.3/esp-idf/components/lwip/apps/dhcpserver/dhcpserver.c:1106
0x4201c951: udp_input at C:\Users\micha\Documents\Cybertek\Software\ip_internal_network\build/C:/Users/micha/esp/v5.3/esp-idf/components/lwip/lwip/src/core/udp.c:404
0x4201f96f: ip4_input at C:\Users\micha\Documents\Cybertek\Software\ip_internal_network\build/C:/Users/micha/esp/v5.3/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c:746
0x4202345e: ethernet_input at C:\Users\micha\Documents\Cybertek\Software\ip_internal_network\build/C:/Users/micha/esp/v5.3/esp-idf/components/lwip/lwip/src/netif/ethernet.c:186
0x420167bd: tcpip_thread_handle_msg at C:\Users\micha\Documents\Cybertek\Software\ip_internal_network\build/C:/Users/micha/esp/v5.3/esp-idf/components/lwip/lwip/src/api/tcpip.c:174
 (inlined by) tcpip_thread at C:\Users\micha\Documents\Cybertek\Software\ip_internal_network\build/C:/Users/micha/esp/v5.3/esp-idf/components/lwip/lwip/src/api/tcpip.c:148
```
I think this issue is caused by the mismatch between your IDF version and the version in the example. Please update your version according to Brain's suggestion and test it again.

michaelsimp, 2024-10-30:
I installed IDF version 5.3.1
I had a problem at step 2 in your instructions to entries up states: "Select esp-idf v5.3 from status bar (recommended) or the ESP-IDF: Configure ESP-IDF extension."
When I try is reports, "Open a folder first."
So I open a folder of a the project I made in version IDF 3.0.  Then on the status bar I could select Version 5.3.1
When I run Run the ESP-IDF: Doctor command, I get the following errors:
- Extension configuration report has been copied to the clipboard with errors.
- Cannot open file ../report.txt. Detail: FIles above 50MB cannot be synchnrozied with extensions.
I checked my report,txt and found it was over 181MB
I tried continuing:
When I select Show examples it only shows 5.3.1 which is good.
I created ip_internal_network, but when completed the status bar reports ESP-IDF v5.2.2 again.
So I tried deleting the large report.txt and trying again.
Same problem, it created a report.txt of 181MB again

brianignacio5, 2024-10-30:
Delete this file:

`%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`

and try to run `ESP-IDF: Doctor command` again. Seems that your extension log have been logging a lot and vscode limit.

About the ESP-IDF v5.2.2 again, it is because the newly created project does not set settings when created. You can select the v5.3.0 from status bar again.

Again sorry for this issue, will work to make it easier to use in the next release of esp-idf extension.

michaelsimp, 2024-10-30:
I need to make some real progress on this so I have completely uninstalled esp-idf and manually deleted all the ESP and espressif folders including all 3 IDF versions.
I have reinstalled ESP-IDF and only IDF version 5.3.1 to remove all doubt.
I will rebuild and test and report
Thanks

michaelsimp, 2024-10-30:
Hi again
Its hard to tell, but seems as if it might be a little more robust, especially with the router power off and on test.
[Attachments.zip](https://github.com/user-attachments/files/17565119/Attachments.zip)
But it still crashes, see files attached including my .elf

"Fail 1.txt" is taken from the Mesh_Root
Line 1458 Mesh_Node disconnected
Line 1495 Mesh_Node reconnects
Line 1496 crash

"Fails 2.txt" is taken from the Mesh_Node
Line 1789 Mesh_Root disconnected
Line 1822 Reconnect
Line 1832 Crash divide by zero

zhangyanjiaoesp, 2024-10-30:
It's weird, I have tested it multiple times as you said (the following two cases) and it can connect normally without any crashing issues.

> Power on both nodes, one becomes MESH_ROOT and one MESH_NODE. Power off WIFI router. MESH_ROOT crashes. See file attached MESH_ROOT "Router power off.txt" (crash at end)
>
> Second test. Power up only [...]

I'm using the Github IDF, and I will try to test with the vscode extension

michaelsimp, 2024-10-30:
FYI I use vscode for coding and building and
      sometimes JTAG debugging
      
      Most of the time in testing I am am monitoring serial com port
      using Putty terminals on com ports
      
      Thanks
    
    
    On 30/10/2024 4:56 pm, ZYJ wrote:
    
    
      
      It's weird, I have tested it multiple times as you
        said (the following two cases) and it can connect normally
        without any crashing issues.
      
        Power on both nodes, one becomes MESH_ROOT and one
          MESH_NODE. Power off WIFI router. MESH_ROOT crashes. See file
          attached MESH_ROOT "Router power off.txt" (crash at end)
        Second test. Power up only one node, becomes
          MESH_ROOT Power off Router This time MESH_ROOT does not crash
          Power on Router MESH_ROOT does not crash Power on a second
          node which connects to the first MESH_ROOT - see line 492
          MESH_ROOT crashes.
      
      I'm using the Github IDF, and I will try to test
        with the vscode extension
      —
      [
{
"target": "https://github.com/espressif/esp-idf/issues/14720#issuecomment-2445782877",
"url": "https://github.com/espressif/esp-idf/issues/14720#issuecomment-2445782877",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

zhangyanjiaoesp, 2024-10-30:
@michaelsimp 

Are you using the completely unmodified example code during your testing?

michaelsimp, 2024-10-30:
Yes.  I did not change anything except:
 - set target device to ESP32-S3 - Internal JTAG debug - What device are you testing with?  Could this be a factor?
 - change partition table, factory partition size to 0x400000
 - Using vscode GUI menuconfig:
     - change device flash size to 15MB 
     - set WiFi router SSID to VONETS and password to "pass9999"
 See my source files attached where you can see most are untouched with the original install date 30/10/24 02:13pm NZ time.
Could someone please answer the questions in my message above.

michaelsimp, 2024-10-30:
FYI I use vscode for coding and building and sometimes JTAG debugging
Most of the time in testing I am am monitoring serial com port using Putty terminals on com ports

In addition to the crashing, sometimes a MESH_NODE will not reconnect to the MESH network.  As a work around for this, I want to stop and restart the wifi mesh network?  No matter what I try, my nodes intermittently crash on restart?  Is this perhaps related?

I am hoping you could please answer a few questions to help me.

What are the recommended steps to stop and then restart the WiFi Mesh network?  I currently have :
```
    ESP_ERROR_CHECK(esp_mesh_stop());
    ESP_ERROR_CHECK(esp_mesh_deinit());
```
but this causes a lot of error logging which stops if I add ...
`    ESP_ERROR_CHECK(mesh_netifs_destroy());`

My restart is as follow:
```
void wifiMeshStart() {
    ESP_LOGW(TAG, "Wifi Mesh switch on");
    /*  mesh initialization */
    ESP_ERROR_CHECK(esp_mesh_init());
    ESP_ERROR_CHECK(esp_mesh_set_max_layer(CONFIG_MESH_MAX_LAYER));
    ESP_ERROR_CHECK(esp_mesh_set_vote_percentage(1));
    ESP_ERROR_CHECK(esp_mesh_set_ap_assoc_expire(10));
    /* set blocking time of esp_mesh_send() to 30s, to prevent the esp_mesh_send() from permanently for some reason */
    ESP_ERROR_CHECK(esp_mesh_send_block_time(30000));
    mesh_cfg_t cfg = MESH_INIT_CONFIG_DEFAULT();
#if !MESH_IE_ENCRYPTED
    cfg.crypto_funcs = NULL;
#endif
    /* mesh ID */
    memcpy((uint8_t *) &cfg.mesh_id, MESH_ID, MAC_SIZE);
    /* router */
    cfg.channel = CONFIG_MESH_CHANNEL;

    cfg.router.ssid_len = strlen(meshProvisionData.ssid);
    memcpy((uint8_t *) &cfg.router.ssid, meshProvisionData.ssid, cfg.router.ssid_len);
    memcpy((uint8_t *) &cfg.router.password, meshProvisionData.password, strlen(meshProvisionData.password));
    
    ESP_ERROR_CHECK(esp_mesh_set_ap_authmode((wifi_auth_mode_t) CONFIG_MESH_AP_AUTHMODE));
    cfg.mesh_ap.max_connection = CONFIG_MESH_AP_CONNECTIONS;
    cfg.mesh_ap.nonmesh_max_connection = CONFIG_MESH_NON_MESH_AP_CONNECTIONS;
    memcpy((uint8_t *) &cfg.mesh_ap.password, CONFIG_MESH_AP_PASSWD, strlen(CONFIG_MESH_AP_PASSWD));
    ESP_ERROR_CHECK(esp_mesh_set_config(&cfg));
    /* mesh start */
    ESP_ERROR_CHECK(esp_mesh_start());
    ESP_LOGI(TAG, "WiFi Mesh started successfully");
}
```

I notice with my custom application:
I have 1 MESH_ROOT and 5 MESH_NODEs spread around the office.  Mesh_Node seem to connect to parents not based on the RSSI as documented.  

The IDF documentation states "To prevent nodes from forming a weak upstream connection, ESP-WIFI-MESH implements an RSSI threshold mechanism for beacon frames."   Is this configurable and if so where?  I cant find it in the API or in MenuConfig.  What is the default RSSI threshhold value?

The IDF documentation states in Preferred Parent Node "The preferred parent node is determined based on the following criteria: Which layer the parent node candidate is situated on. The number of downstream connections (child nodes) the parent node candidate currently has".  
Does this mean RSSI is not part of the parent selection process?

Is it recommend to use self-organized networking or for serious applications should I manually build the mesh network? I will only have a max of 10 mesh nodes altogether but they do a reasonable amount of MQTT5 communications to the cloud.

zhangyanjiaoesp, 2024-10-31:
@michaelsimp  The following are the answers for your questions:
1. > What are the recommended steps to stop and then restart the WiFi Mesh network? I currently have :
    > 
    > ```
    >     ESP_ERROR_CHECK(esp_mesh_stop());
    >     ESP_ERROR_CHECK(esp_mesh_deinit());
    > ```
    Call `esp_mesh_stop()` is enough.

2. > but this causes a lot of error logging which stops if I add ...
    > ` ESP_ERROR_CHECK(mesh_netifs_destroy());`
    Where did you add the `mesh_netifs_destory()` function? What does the error log look like? Can you provide an example？

3. Where did you call the `wifiMeshStart()` function?
4. > I have 1 MESH_ROOT and 5 MESH_NODEs spread around the office. Mesh_Node seem to connect to parents not based on
    > the RSSI as documented.

     RSSI is not the only criterion for selecting the parent node, the layer and connections also need to be considered.
5. >The IDF documentation states "To prevent nodes from forming a weak upstream connection, ESP-WIFI-MESH implements an RSSI threshold mechanism for beacon frames." Is this configurable and if so where? I cant find it in the API or in MenuConfig. What is the default RSSI threshhold value?
    
    You can call this API:
https://github.com/espressif/esp-idf/blob/9106c43accd9f5e75379f62f12597677213f5023/components/esp_wifi/include/esp_mesh_internal.h#L216
6. >The IDF documentation states in Preferred Parent Node "The preferred parent node is determined based on the following criteria: Which layer the parent node candidate is situated on. The number of downstream connections (child nodes) the parent node candidate currently has". Does this mean RSSI is not part of the parent selection process?

    Same to the fourth point, selecting parent need to consider RSSI, layer and connections, the doc need to be updated.
8. > Is it recommend to use self-organized networking or for serious applications should I manually build the mesh network? I will only have a max of 10 mesh nodes altogether but they do a reasonable amount of MQTT5 communications to the cloud.

    You can use self-organized network.

zhangyanjiaoesp, 2024-10-31:
@michaelsimp I can reproduce the crash using the vscode, I will check the difference between VSCode and standard IDF

michaelsimp, 2024-10-31:
Hi Zhangyanjiaoesp
This excellent news for me.  Hopefully it is just something simple you will find soon and be able to offer me a fix.
Thank you

michaelsimp, 2024-10-31:
Hi Zhangyanjiaoesp

Thanks so much for taking the time to answer all my questions.  

Please note THESE tests are with MY application (NOT with example program ip_internal_network) running on a network of 6 nodes - all ESP32-S3.  My application has a CLI console integrated so I can trigger actions and see the responses on the COM port.

Q1 I will go back to just calling `esp_mesh_stop()` and see what happens.
     To restart should I just be able to call `ESP_ERROR_CHECK(esp_mesh_start());`

Q2 Triggered from the CLI Console I was calling:
```
    ESP_ERROR_CHECK(esp_mesh_stop());
    ESP_ERROR_CHECK(esp_mesh_deinit());
    ESP_ERROR_CHECK(mesh_netifs_destroy());
```

Q3 `wifiMeshStart()` is also called from my CLI console

The CLI Console is started from my Mainline as is my WiFi Mesh application (built on top of the ip_internal_network source).
Triggering the Mesh Stop and Start would be called from the CLI Console thread.  I am assuming this is ok and does not need any mutex protection.  Please advise how I should call it if this is a problem.

Q4     I understand this

Q5 Thanks

Q6 Thanks for clarifying this but I am not finding this to be the case.  I distributed some MESH_Nodes across the office with the aim of creating a multi-hop network between the far extremes. But it does not form as expected or at all well for healthy RSSI.  I have nodes which are close to my Mesh_Root or 2nd layer Mesh_Nodes which are not at parent capacity numbers.  When my other Mesh_Nodes do connect to these parents they provide an RSSI on the child to the parent of -35dBm.  But they most frequently want connect to nodes a much longer distance away getting a RSSI of < -70dBm.

I read somewhere the default RSSI threshold is -120dBm, but I am finding nodes with RSSIs < -70dBm often lose their MQTT connection to the broker.  I have an office environment and I have located the nodes approx 10 to 20 meters apart with a max of one wall between but they are not all line of site.  I very much doubt I could even get a connection at RSSI less than -100dBm.  I am thinking it may be a signal to noise ratio issue so I have scanned the office for WiFi channel usage and selected channel 1 on my WIFI Router as nothing else is using this channel and no other channels overlap.  I know this is not an easy question to answer with precision and I appreciate the many influences, but realistic what is the ballpark min RSSI range at which I can expect a node to work reliably at what sort of distance range.

Q7  Because of my Q6 response above, I have started evaluating the example project "manual_networking" to make my MESH_NODES manually scan and select MESH_NODES with the healthiest RSSI.  It sounds like you are saying the IDF framework should already be doing this?
So I am now wondering if the vscode crashing issue is also causing this to not work properly and your fix might fix both.
Should I put manual scanning and parent selection changes to one side and wait for the outcome of the vscode crashing?
I guess I would prefer to use the self-organized network as much as possible, if it works as you describe.
Please advise / confirm manual scanning and parent selection should not be required and I might need you to look at the node parent selection for healthy RSSI next.

Best regards

zhangyanjiaoesp, 2024-10-31:
@michaelsimp 
1. According to the backtrace of the crash issue, it seems related to DHCP, it won't affect the mesh networking.
2. I'm sorry I didn't quite understand your question regarding the selection of the parent node. Can you draw a picture to explain it? For example, where are nodes A, B, and C located? What level? How many child nodes are connected below? What is the RSSI of A, B, and C scanning each other? Do you expect A to connect to B but actually connect A to C?

michaelsimp, 2024-10-31:
See attached:
[MeshMaps.zip](https://github.com/user-attachments/files/17584596/MeshMaps.zip)

"Target Network .png" shows the walls as black lines and nodes as circles.  The blue lines are approx what I was expecting to see.

It forms very randomly but with bad choices like the file "Actual example Network.png" with links and dBm in red.

My project is configured for up to 50 nodes and  3 children per node as I wanted to force some layers.

michaelsimp, 2024-10-31:
"According to the backtrace of the crash issue, it seems related to DHCP, it won't affect the mesh networking."

That may be the case with the crash issue you found, but the root cause of the vscode IDF environment may cause more than one issue.  I guess you will know better when you get to the bottom of the vscode IDF environment issue

Do the diagrams I sent help you understand my issue better now?

zhangyanjiaoesp, 2024-10-31:
Yes, I now understand your question. Once the ROOT node is formed, the chances for other idle nodes to connect to the root node are the same; as long as they can scan the root node within the RSSI threshold range, they can connect and become second-layer nodes. Therefore, it is reasonable for C and D to connect to A and become second-layer nodes. What is the RSSI that B, E, and F receive from A? Since each node can connect to 3 child nodes, if they are within the RSSI threshold range and root A is not yet fully connected, at least one of B, E, or F should be able to connect to A.

I think you can call `esp_mesh_set_rssi_threshold()` to limit the RSSI threshold for optional parents, which would allow nearby nodes to connect as much as possible. However, nodes D and E are too far from node A. If you set the same RSSI threshold for all nodes, the connection results might still not meet your expectations, unless you configure different RSSI thresholds for each node. Alternatively, you could only call the `esp_mesh_set_parent()` function to specify the parent of each node.

[image]

michaelsimp, 2024-10-31:
My main concern is F connecting to C when it should go to E or A or B, and also E Connecting to D instead of A or B.

Raising the threshold should prevent this but will prevent connection is some nodes go off, causing more problems.

I think I should have my nodes scan manually for a list of eligible parents and select the one with the best RSSI.  Do nodes only become eligible to become parents once they have a connection to a parent?

Can I make a node scan after it has connected to a parent to look for a better connection if its RSSI is lower than a useful connection of say -65dBm.  If I only do this occasionally, is there a down side?

Is the RSSI an average value, or from the latest message received from the parent or something else?

zhangyanjiaoesp, 2024-11-01:
> Do nodes only become eligible to become parents once they have a connection to a parent?

Yes, only the [idle nodes](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/esp-wifi-mesh.html#node-types) are eligible to become parent nodes. 
> Can I make a node scan after it has connected to a parent to look for a better connection if its RSSI is lower than a useful connection of say -65dBm. If I only do this occasionally, is there a down side?

Of cause you can switch the parent node as you want. In the self-organized mesh networking, the node can switch parent autonomously, see the [doc](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/esp-wifi-mesh.html#parent-node-switching).

> Is the RSSI an average value, or from the latest message received from the parent or something else?
The RSSI is from the latest beacon from the parent.

zhangyanjiaoesp, 2024-11-01:
@michaelsimp For the crash issue, can you please cherry-pick [this branch](https://github.com/espressif/esp-idf/commit/fc92df3ac3205a4985e0a3165b287dc7f2234890) and test again?

michaelsimp, 2024-11-01:
Hi
Today I merged some of the code from the example project "manual_networking" into my project.
I start everything up self organized.
Once connected, I have a task which checks if MESH_NODEs and MESH_LEAFs have a parent connection < 60dBm.  If so they run:
```
    ESP_ERROR_CHECK(esp_mesh_set_self_organized(0, 0));
    esp_wifi_scan_stop();
    wifi_scan_config_t scan_config = { 0 }; // mesh softAP is hidden
    scan_config.show_hidden = 1;
    scan_config.scan_type = WIFI_SCAN_TYPE_PASSIVE;
    esp_wifi_scan_start(&scan_config, 0);
```
On MESH_EVENT_SCAN_DONE I added mesh_scan_done_handler(scan_done->number);

I modified `mesh_scan_done_handler()` from "manual_networking" so that it will not stop on the first eligible parent.  It keeps scanning the entire list and takes the one with the highest RSSI.

The results are really good.  Please see attached my mesh mapping software output:

- Left map is typical of self organized.
- Right map is the same setup but with the new rescan to select better parents.

[mapping.zip](https://github.com/user-attachments/files/17596324/mapping.zip)

Thanks for helping by answering my questions.

michaelsimp, 2024-11-01:
Its the end of my week but I will try to test the mesh_netif.c over the weekend

michaelsimp, 2024-11-02:
Hi Zhangyanjiaoesp
No its still crashes.
Quite reproducible if your get two nodes connected and then power off the root for a couple of seconds and back on.
See attached zip with my mesh_netif.c source file and the bin and elf files for you to check.
Also Crash1.txt with the MESH_NODE crashing line 182 IntegerDivideByZero after the test above.

michaelsimp, 2024-11-03:
Perhaps related, I regularly get "Send with err code 16392 ESP_ERR_MESH_ARGUMENT" from mesh_netif: 
The source code has 3 instances which could print this error message. I added some diagnostics to prove it comes from the function: `static esp_err_t mesh_netif_transmit_from_node_sta(void *h, void *buffer, size_t len)`

The example below shows my code MQTT publishing a test message once / second cleanly and then I get a bunch of "Send with err code 16392 ESP_ERR_MESH_ARGUMENT" messages, and then it goes clean again.

What does this mean?
```
...
I (11:41:13.293) aMQTT: Console Publish: Topic: SiteID/00003030F93349BC/Fuelpoint/6/PumpStatus Payload: Status=12&TransType=1&State=3&TransactionSeqNo=123456&Hose=2&Product=6&Money=12345.67&Volume=987654.32&UnitPrice=12.34&LimitTyp=1&LimitAmount=100.00
I (11:41:14.302) aMQTT: Console Publish: Topic: SiteID/00003030F93349BC/Fuelpoint/7/PumpStatus Payload: Status=13&TransType=1&State=3&TransactionSeqNo=123456&Hose=2&Product=7&Money=12345.67&Volume=987654.32&UnitPrice=12.34&LimitTyp=1&LimitAmount=100.00
I (11:41:15.303) aMQTT: Console Publish: Topic: SiteID/00003030F93349BC/Fuelpoint/8/PumpStatus Payload: Status=14&TransType=1&State=3&TransactionSeqNo=123456&Hose=2&Product=8&Money=12345.67&Volume=987654.32&UnitPrice=12.34&LimitTyp=1&LimitAmount=100.00
I (11:41:16.313) aMQTT: Console Publish: Topic: SiteID/00003030F93349BC/Fuelpoint/9/PumpStatus Payload: Status=15&TransType=1&State=3&TransactionSeqNo=123456&Hose=2&Product=9&Money=12345.67&Volume=987654.32&UnitPrice=12.34&LimitTyp=1&LimitAmount=100.00
I (11:41:17.313) aMQTT: Console Publish: Topic: SiteID/00003030F93349BC/Fuelpoint/10/PumpStatus Payload: Status=16&TransType=1&State=3&TransactionSeqNo=123456&Hose=2&Product=10&Money=12345.67&Volume=987654.32&UnitPrice=12.34&LimitTyp=1&LimitAmount=100.00
I (11:41:18.313) aMQTT: Console Publish: Topic: SiteID/00003030F93349BC/Fuelpoint/1/PumpStatus Payload: Status=0&TransType=1&State=3&TransactionSeqNo=123456&Hose=2&Product=1&Money=12345.67&Volume=987654.32&UnitPrice=12.34&LimitTyp=1&LimitAmount=100.00
I (259339) mesh: 5130<assoc>parent layer:1, channel:1, rssi:-44, assoc:4, rssi threshold<-78,-82,-85>
E (11:41:21.973) mesh_netif: C Send with err code 16392 ESP_ERR_MESH_ARGUMENT
E (11:41:21.999) mesh_netif: C Send with err code 16392 ESP_ERR_MESH_ARGUMENT
I (11:41:22.323) aMQTT: Console Publish: Topic: SiteID/00003030F93349BC/Fuelpoint/5/PumpStatus Payload: Status=4&TransType=1&State=3&TransactionSeqNo=123456&Hose=2&Product=5&Money=12345.67&Volume=987654.32&UnitPrice=12.34&LimitTyp=1&LimitAmount=100.00
E (11:41:22.356) mesh_netif: C Send with err code 16392 ESP_ERR_MESH_ARGUMENT
E (11:41:22.723) mesh_netif: C Send with err code 16392 ESP_ERR_MESH_ARGUMENT
E (11:41:22.755) mesh_netif: C Send with err code 16392 ESP_ERR_MESH_ARGUMENT
E (11:41:23.223) mesh_netif: C Send with err code 16392 ESP_ERR_MESH_ARGUMENT
I (11:41:23.333) aMQTT: Console Publish: Topic: SiteID/00003030F93349BC/Fuelpoint/6/PumpStatus Payload: Status=5&TransType=1&State=3&TransactionSeqNo=123456&Hose=2&Product=6&Money=12345.67&Volume=987654.32&UnitPrice=12.34&LimitTyp=1&LimitAmount=100.00
E (11:41:23.362) mesh_netif: C Send with err code 16392 ESP_ERR_MESH_ARGUMENT
E (11:41:23.723) mesh_netif: C Send with err code 16392 ESP_ERR_MESH_ARGUMENT
E (11:41:23.765) mesh_netif: C Send with err code 16392 ESP_ERR_MESH_ARGUMENT
I (11:41:24.333) aMQTT: Console Publish: Topic: SiteID/00003030F93349BC/Fuelpoint/7/PumpStatus Payload: Status=6&TransType=1&State=3&TransactionSeqNo=123456&Hose=2&Product=7&Money=12345.67&Volume=987654.32&UnitPrice=12.34&LimitTyp=1&LimitAmount=100.00
E (11:41:24.723) mesh_netif: C Send with err code 16392 ESP_ERR_MESH_ARGUMENT
E (11:41:25.723) mesh_netif: C Send with err code 16392 ESP_ERR_MESH_ARGUMENT
E (11:41:26.223) mesh_netif: C Send with err code 16392 ESP_ERR_MESH_ARGUMENT
E (11:41:26.723) mesh_netif: C Send with err code 16392 ESP_ERR_MESH_ARGUMENT
E (11:41:27.723) mesh_netif: C Send with err code 16392 ESP_ERR_MESH_ARGUMENT
E (11:41:27.917) mesh_netif: C Send with err code 16392 ESP_ERR_MESH_ARGUMENT
E (11:41:27.973) mesh_netif: C Send with err code 16392 ESP_ERR_MESH_ARGUMENT
I (11:41:19.319) aMQTT: Console Publish: Topic: SiteID/00003030F93349BC/Fuelpoint/2/PumpStatus Payload: Status=1&TransType=1&State=3&TransactionSeqNo=123456&Hose=2&Product=2&Money=12345.67&Volume=987654.32&UnitPrice=12.34&LimitTyp=1&LimitAmount=100.00
START MQTT CONSOLE REDIRECT
I (11:41:34.675) aMQTT: Console Publish: Topic: SiteID/00003030F93349BC/Fuelpoint/8/PumpStatus Payload: Status=7&TransType=1&State=3&TransactionSeqNo=123456&Hose=2&Product=8&Money=12345.67&Volume=987654.32&UnitPrice=12.34&LimitTyp=1&LimitAmount=100.00
I (11:41:34.692) mqtt5_client: MQTT_MSG_TYPE_SUBACK return code is 1
I (11:41:34.709) aMQTT: MQTT_EVENT_SUBSCRIBED, msg_id=345  inFlightCounter: 0
I (11:41:20.323) aMQTT: Console Publish: Topic: SiteID/00003030F93349BC/Fuelpoint/3/PumpStatus Payload: Status=2&TransType=1&State=3&TransactionSeqNo=123456&Hose=2&Product=3&Money=12345.67&Volume=987654.32&UnitPrice=12.34&LimitTyp=1&LimitAmount=100.00
I (11:41:35.693) aMQTT: Console Publish: Topic: SiteID/00003030F93349BC/Fuelpoint/9/PumpStatus Payload: Status=8&TransType=1&State=3&TransactionSeqNo=123456&Hose=2&Product=9&Money=12345.67&Volume=987654.32&UnitPrice=12.34&LimitTyp=1&LimitAmount=100.00
[... cut ...]
I (11:41:40.714) aMQTT: Console Publish: Topic: SiteID/00003030F93349BC/Fuelpoint/4/PumpStatus Payload: Status=13&TransType=1&State=3&TransactionSeqNo=123456&Hose=2&Product=4&Money=12345.67&Volume=987654.32&UnitPrice=12.34&LimitTyp=1&LimitAmount=100.00
E (11:41:21.585) mesh_netif: C Send with err code 16392 ESP_ERR_MESH_ARGUMENT
I (11:41:41.723) aMQTT: Console Publish: Topic: SiteID/00003030F93349BC/Fuelpoint/5/PumpStatus Payload: Status=14&TransType=1&State=3&TransactionSeqNo=123456&Hose=2&Product=5&Money=12345.67&Volume=987654.32&UnitPrice=12.34&LimitTyp=1&LimitAmount=100.00
...
```

michaelsimp, 2024-11-03:
Regarding Q1 and Q@ from 31 October (above), I still can't get stopping and restarting the wifi_mesh to work.

Per your advice I tried just calling function `esp_mesh_stop()` - this is triggered from my CLI console
This results in lots of ESP_LOG messages:  information warning and error .

and if I just call `esp_mesh_start()` the app generally crashes.  This is the same as when I called my original more complex restart procedure (see below):

See WifiStopStar.txt logs attached
[WifiStopStart.txt](https://github.com/user-attachments/files/17608702/WifiStopStart.txt)

Can you have a rethink and test yourself if necessary and advise how I can make this work.  I have 2 needs:

1. to try to fix any wifi mesh issues without rebooting the entire app.  Sometimes a node will just not crash, but it just stops all wifi mesh functionailty
2. to suspend wifimesh coms when not required

```
void wifiMeshStart() {
    ESP_LOGW(TAG, "Wifi Mesh switch on");
    /*  mesh initialization */
    ESP_ERROR_CHECK(esp_mesh_init());
    ESP_ERROR_CHECK(esp_mesh_set_max_layer(CONFIG_MESH_MAX_LAYER));
    ESP_ERROR_CHECK(esp_mesh_set_vote_percentage(1));
    ESP_ERROR_CHECK(esp_mesh_set_ap_assoc_expire(10));
    /* set blocking time of esp_mesh_send() to 30s, to prevent the esp_mesh_send() from permanently for some reason */
    ESP_ERROR_CHECK(esp_mesh_send_block_time(30000));
    mesh_cfg_t cfg = MESH_INIT_CONFIG_DEFAULT();
#if !MESH_IE_ENCRYPTED
    cfg.crypto_funcs = NULL;
#endif
    /* mesh ID */
    memcpy((uint8_t *) &cfg.mesh_id, MESH_ID, MAC_SIZE);
    /* router */
    cfg.channel = CONFIG_MESH_CHANNEL;

    cfg.router.ssid_len = strlen(meshProvisionData.ssid);
    memcpy((uint8_t *) &cfg.router.ssid, meshProvisionData.ssid, cfg.router.ssid_len);
    memcpy((uint8_t *) &cfg.router.password, meshProvisionData.password, strlen(meshProvisionData.password));
    
    ESP_ERROR_CHECK(esp_mesh_set_ap_authmode((wifi_auth_mode_t) CONFIG_MESH_AP_AUTHMODE));
    cfg.mesh_ap.max_connection = CONFIG_MESH_AP_CONNECTIONS;
    cfg.mesh_ap.nonmesh_max_connection = CONFIG_MESH_NON_MESH_AP_CONNECTIONS;
    memcpy((uint8_t *) &cfg.mesh_ap.password, CONFIG_MESH_AP_PASSWD, strlen(CONFIG_MESH_AP_PASSWD));
    ESP_ERROR_CHECK(esp_mesh_set_config(&cfg));
    /* mesh start */
    ESP_ERROR_CHECK(esp_mesh_start());
    ESP_LOGI(TAG, "WiFi Mesh started successfully");
}
```

zhangyanjiaoesp, 2024-11-04:
> Hi Zhangyanjiaoesp No its still crashes. Quite reproducible if your get two nodes connected and then power off the root for a couple of seconds and back on. See attached 3Nov24-1.zip with my mesh_netif.c source file and the bin and elf files for you to check. Also Crash1.txt with the MESH_NODE crashing line 182 IntegerDivideByZero after the test above. [3Nov24-1.zip](https://github.com/user-attachments/files/17608485/3Nov24-1.zip)

@michaelsimp Because I am not always able to reproduce the problem when testing with VSode, I have provided you with the following debug version:
1. Replace the wifi lib in the `idf/components/espw_ifi/lib/esp32s3` path with the wifi lib in the attachment
2. add the patch in `dhcpserver.c`
[wifi_lib_s3_1104.zip](https://github.com/user-attachments/files/17615268/wifi_lib_s3_1104.zip)
[0001-fix-dhcp-add-debug-log-for-dhcp-server.zip](https://github.com/user-attachments/files/17615271/0001-fix-dhcp-add-debug-log-for-dhcp-server.zip)

zhangyanjiaoesp, 2024-11-04:
`Send with err code 16392 ESP_ERR_MESH_ARGUMENT` this log indicates that the parameter error passed into `esp_mesh_send()` function.

michaelsimp, 2024-11-04:
For mesh, stop ->init ->config ->start...
I am not sure what this means and how I stop it crashing.  
The only WiFi comms I am doing during the test is via the MQTT API with an MQTT5 TLS connection.
Before I call esp_mesh_stop(), I set a flag to stop publishing any more MQTT messages.  MQTT is still active and I guess heartbeating with the broker, so I will try disconnecting from the broker tomorrow and retest.

The ->init ->config ->start code I call after `esp_mesh_stop()` is the exact same code as I call to initially start the WiFi mesh network.  But It is ALWAYS successful on initial startup and ALWAYS fails after calling `esp_mesh_stop()`.

Would you be able to add some simple code to the example project "ip_internal_network" which after say 30 seconds of running, calls `esp_mesh_stop()` and after 10 seconds restarts the mesh again to show me how it should work. Note I can't stop and restart the mesh on either the MESH_ROOT or MESH_NODE.

zhangyanjiaoesp, 2024-11-04:
> Could this also be related to the crashing issue as it is also in mesh_netif.

I'm not sure, but I think it shouldn't matter.

zhangyanjiaoesp, 2024-11-04:
> For mesh, stop ->init ->config ->start... I am not sure what this means and how I stop it crashing. The only WiFi comms I am doing during the test is via the MQTT API with an MQTT5 TLS connection. [...]

I think it would be better if you could provide a demo to reproduce your problem, and we can help check if there are any issues with your code.

michaelsimp, 2024-11-04:
Hi
I installed the wifi_lib_s3_1104 folder you sent me today and did a clean build and installed in my two test nodes.  Does this just have diagnostics, or some improvements which chould fix the crashing?  The reason I ask is I could not reproduce the crashing after quite a few cycles approx 20.

To be sure I restored the original library to double test.  But I found it much harder to reproduce the crash today than previously with the original library. Eventually after quite a few power cycIes I caught one - see attached OrigLibraryCrash.txt

So I restored the new wifi_lib_s3_1104 library again and worked really hard at trying to crash it.  I tried many many times maybe 50+ and just could not break it.

[5Nov-1.zip](https://github.com/user-attachments/files/17624838/5Nov-1.zip)
This contains my "ip_internal_network.elf file with the new library installed for your reference.

What I did see from time to time was the **MESH_NODE** would start logging quite a bit of diagnostics form "wifi" in between the normal "mesh_main" messages. You might have expected this to occur, eg:
```
I (197688) mesh_main: Tried to publish layer:2 IP:10.0.0.2
I (197688) mesh_mqtt: sent publish returned msg_id=54636
I (197758) wifi:>>>intv = 102400, max = 0
I (197858) wifi:>>>intv = 102400, max = 0
I (197958) wifi:>>>intv = 102400, max = 0
I (198058) wifi:>>>intv = 102400, max = 0
I (198158) wifi:>>>intv = 102400, max = 0
I (198268) wifi:>>>intv = 102400, max = 0
I (198268) mesh_main: Received Routing table [0] 30:30:f9:33:72:b8
I (198268) mesh_main: Received Routing table [1] 48:ca:43:9b:53:d8
I (198368) wifi:>>>intv = 102400, max = 0
I (198468) wifi:>>>intv = 102400, max = 0
I (198568) wifi:>>>intv = 102400, max = 0
I (198678) wifi:>>>intv = 102400, max = 0
I (198778) wifi:>>>intv = 102400, max = 0
I (198878) wifi:>>>intv = 102400, max = 0

```
Two example scenarios of this are:
**newdiagtrace 1.txt:**
This is from the MESH_NODE
Power off and on the MESH_ROOT quickly
MeshNode starts logging heaps of stuff eg see line 1429 onwards
When MESH_ROOT comes back (same Root reestablishes itself after quick power cycle), the MESH_NODE keeps logging stuff.
Then when MESH_ROOT power cycled again, the extra logging on the MESH_NODE stops

**newdiagtrace 2.txt:**
This is MESH_NODE
Restart the MESH_ROOT multiple times until MESHNODE starts logging heaps of stuff
See line 312
Wait for this MESH_NODE to timeout, disconnect and then establish itself as MESH_ROOT line 475
Logging stops

Neither is causing a problem, I just wanted to give you all the information.

I am very interested to know whether the diagnostics had the potential to fix the crashing as it seems to have from my testing.  Please advise next steps.

zhangyanjiaoesp, 2024-11-05:
@michaelsimp 
Yes, I added a fix to the wifi lib I provided to you. `wifi:>>>intv = 102400, max = 0` this log was added by me to locate the problem. I'm glad to hear that the problem has been solved.

Have you tested the router reboot case? Does the root node crash still exist?

michaelsimp, 2024-11-05:
Yes I have tested the router a few times.  I will need to do some more structured testing but it is looking very good now. Thank you for all your help.  Do you know what the forma release patch will be and when yet?

I also need to resolve the WIFI and MESH stop and restart but I am tied up for the next couple of days.
Can we keep this open until I am ready please.

zhangyanjiaoesp, 2024-11-05:
The patch will merge into master branch firstly and then backported into release/v5.3, we will merge it ASAP.

> Can we keep this open until I am ready please.

Of cause, we can keep it open.

michaelsimp, 2024-11-05:
Hi
I rebuilt my application with the wifi_lib_s3_1104 folder and did some more testing on my larger network of 6 nodes spread around the office. They all have the same software installed.
I tried resetting the MESH_ROOT node and watching the MESH_Network re-establish.
A lot of the time this worked perfectly but two nodes in particular would sometimes not come back online. 

So I reinstalled the "ip_internal_network" app on 4 nodes spread over two locations in the office.
It is still much better than before, but it does still crash.
[Crash 6.txt](https://github.com/user-attachments/files/17639106/Crash.6.txt)

Attached as an example on my MESH ROOT. 
Reset MESH_ROOT
Starts up as MESH_ROOT again and then crashes at line 193.  You have my elf file from yesterdays post

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x42013528  PS      : 0x00060430  A0      : 0x82013d17  A1      : 0x3fca8280
A2      : 0x00000006  A3      : 0x000034c7  A4      : 0x00060420  A5      : 0x3fca8324
A6      : 0x3fca56ec  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x00000006  A11     : 0x00000020  A12     : 0x00060a20  A13     : 0x00060623
A14     : 0x00000009  A15     : 0x0000abab  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0000000d  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff

Backtrace: 0x42013525:0x3fca8280 0x42013d14:0x3fca82a0 0x4201f6c1:0x3fca82d0 0x420234ba:0x3fca8300 0x42016819:0x3fca8320

ELF file SHA256: a6faa33f6
```
I need to find why my application is more prone to crashing than "ip_internal_network".  I suspect it is just busier coms wise.  The only communications it does in this crash test is MQTT5 through the MQTT API.  I subscribe to 3 topics on start up and then no more subscriptions (so none during the MESH_ROOT reboot.  I only publish using `esp_mqtt_client_enqueue`
So I am at a bit of a loss to know what to look at.  Do you have any suggestions for me or should I wait for another hopeful fix to the library.

zhangyanjiaoesp, 2024-11-06:
@michaelsimp 

> 2\. 0001-fix-dhcp-add-debug-log-for-dhcp-server.zip

Have you added this patch?

michaelsimp, 2024-11-06:
Hi.
No sorry I missed this instruction your post.  I have just applied it now and am rebuilding.
I have found a reason for my application crashing, but am not sure if it is new as a result of the earlier patch.  I have not noticed up up until now, but these things happen some time.

The crash is caused by calling esp_wifi_sta_get_rssi(&currentRSSI); when the node is of type MESH_IDLE.  This causes a crash which I have not seen before.  The simple fix I have applied is:
```
if (esp_mesh_get_type() != MESH_IDLE) {
        ESP_ERROR_CHECK(esp_wifi_sta_get_rssi(&currentRSSI));
}
```
I understand why requesting RSSI if you don't have a parent doesn't make sense, but it shouldn't really cause a crash.  Maybe something at add to your backlog.

michaelsimp, 2024-11-07:
My application is looking a lot healthier now, but I still have a couple of issues to resolve.

I was developing and testing with a fixed MESH_ROOT as I can keep this at my desk, while my MESH_NODES are spread around the office.  However my field deployment will need to not have a fixed root for redundancy.

About 5 days ago I posted how I was having my MESH_NODE and MESH_LEAFs scan for closer parents if the RSSI < -60dBm.  This is using code taken from the **"manual_networking"** sample project.  This works perfectly and is a really good addition, when I have a fixed MESH_ROOT.

But for my field deployment where I need any node to become MESH_ROOT, my code has a problem.

After running my scan and parent swap, on only one MESH_NODE, if I restart the MESH_ROOT, none of my 6 nodes will become MESH_ROOT.  Some nodes might have 4 children showing in their routing table but their Type will be MESH_NODE or MESH_IDLE.

This is my manual scan initiate software

```
    ESP_ERROR_CHECK(esp_mesh_set_self_organized(0, 0));
    esp_wifi_scan_stop();
    wifi_scan_config_t scan_config = { 0 }; // mesh softAP is hidden
    scan_config.show_hidden = 1;
    scan_config.scan_type = WIFI_SCAN_TYPE_PASSIVE;
    esp_wifi_scan_start(&scan_config, 0);

```
On MESH_EVENT_SCAN_DONE I added mesh_scan_done_handler(scan_done->number) as in "manual_networking".

I modified `mesh_scan_done_handler()` from "manual_networking" so that it will not stop on the first eligible parent. It keeps scanning the entire list and takes the one with the highest RSSI.

I tried adding

    `ESP_ERROR_CHECK(esp_mesh_set_self_organized(1, 0));`
after the MESH_NODE gets a new parent, but this still doesn't help.

I don't fully understand the API and this example.  
Is it possible to start with a self organized network and then make MESH_NODES manually scan for better parents (higher RSSI) and use `esp_mesh_set_parent` like in `mesh_scan_done_handler` of manual_networking, and **not** have a fixed root.

Or do I have to manually scan and set the MESH_ROOT as well.

Can you give me a steer as to how to fix this problem or a better way to meet by goal of forming a MESH with the lowest possible RSSI figures and redundancy of any NODE being able to become MESH_ROOT.

Thanks.

zhangyanjiaoesp, 2024-11-07:
@michaelsimp 

> This is my manual scan initiate software
>
> ```
>     ESP_ERROR_CHECK(esp_mesh_set_self_organized(0, 0));
>     esp_wifi_scan_stop();
>     wifi_scan_config_t scan_config = { 0 }; // mesh softAP is [...]
When do you call these operations?

> After running my scan and parent swap, on only one MESH_NODE, if I restart the MESH_ROOT, none of my 6 nodes will become MESH_ROOT. Some nodes might have 4 children showing in their routing table but their Type will be MESH_NODE or MESH_IDLE.

Is your root node explicitly specified? Which functions have been called? Do you mean that after the root node restarts, it can no longer remain the root node? And that the original node cannot form the network anymore? Could you provide some logs?

> I tried adding
> 
> ```
> `ESP_ERROR_CHECK(esp_mesh_set_self_organized(1, 0));`
> ```
What functionality are you trying to achieve by calling this function?

By the way, to better address the issue, please provide as much comprehensive log information as possible when describing the problem. It would be even more helpful if you could provide a demo.

michaelsimp, 2024-11-08:
Hi
I thought it best to go back to the Espressif example project "manual_networking".  These have been build with the 2 earlier updates to the IDF library.  This project does not use "mesh_netif.c"

I made no changes to the source.
I selected device ESP32-S3. 
Using Menu Config I:
- changed the Flash memory size to 16MB
- set Example Configuration device type to "MESH_SET_NODE" as this matches my objectives with the network automatically assigning a MESH_ROOT.
- entered my Router SSID and password

I did a clean build and tested.  See attachments for log files, line numbers and summary at top of each file.
[Manual test 8Nov.zip](https://github.com/user-attachments/files/17671257/Manual.test.8Nov.zip)

Test one, power 2 nodes up at the same time
Log file "manual test 1" Node crashes, reboots and then becomes root
Log file "manual test 1B" Node won't become root even though the other node is delayed with a crash and restart.  It displays disconnect message for a period of time until the other node becomes MESH_ROT and then it connects to it.

Test two, power 2 nodes one becomes MESH_ROOT and the other connects to it.
Restart the MESH_ROOT node for a couple of cycles until both nodes get in a state where neither becomes MESH_ROOT and so there are no connections made.
Log file "manual test 2A: Crashes once and then restarts.  Does not become MESH_ROOT and does not connect as there is no MESH_ROOT
Log file "manual test 2B: Start but does not become MESH_ROOT and does not connect as there is no MESH_ROOT

I stopped the log files after 72 seconds so I wouldn't lose the start. But I left the 2 terminals running while I wrote this up and they have now been going for 1416228ms or 1416 seconds or over 23 minutes

This pretty much mirrors what I am seeing with my application.
If I keep resetting nodes, they do start up with one becoming MESH_ROOT and the other connecting, but it is VERY easy to get it to fault as described above.  Please try it for yourself.

My objectives are to have a self organized network WITHOUT fixing my MESH_ROOT.  But if a MESH_NODE connects to a parent with an RSSI < -60dBm, I want them to scan and find eligible parents with better RSSI and switch to them using `esp_mesh_set_parent()`.  The reason is I have found the mesh network comms to be flaky if my RSSI figures get less than -65dBm.  The stronger the better.  

Most of the time (an for the testing I am doing and reporting to you) I am only using MQTT comms for nodes to an external broker, 

But I have also developed and OTA solution, using HTTPS to update my MESH_ROOT and then have my MESH_ROOT push OTA updates to my MESH_NODEs using esp_mesh_send() and recv_cb.  I have developed a protocol and broadcast packets of 1024 bytes at a time, and the MESH_NODEs respond after a small delay X their node number to avoid collisions.  I have Ack messages and a retry mechanism for dropped packets and this works well if RSSI > -65dBm.  BUT less than -65dBm the dropped packets become to much.  I drop these nodes from the OTA process and complete the broadcast to the rest.  I then follow up with P2P messages to the nodes I dropped which works better over lower RSSI connections.  But none of the OTA stuff is active in my application during the tests I have been running, this is just for background information.

zhangyanjiaoesp, 2024-11-08:
@michaelsimp 
1. The crash issue is caused by the error check you added when calling the `esp_mesh_scan_get_ap_ie_len()` function. After the AP is retrieved, it returns -1.
[image]

2.  Since you have set this node as "MESH_SET_NODE" in the menuconfig, it will only look for a mesh AP when searching for a parent. The mesh AP's mesh IE length is non-zero, and the parent of the root can only be a router,  router does not contain the mesh IE. Refer to the code in the example:
[image]

    > Log file "manual test 1" Node crashes, reboots and then becomes root

    So, your description is incorrect. In "manual test1," the device (**30:30:f9:33:72:b8**) did not become the root; instead, after rebooting, it became a third layer node. The log is : `I (7672) mesh_main: <MESH_EVENT_PARENT_CONNECTED>layer:0-->3, parent:48:ca:43:9b:5d:21, ID:77:77:77:77:77:76`

3.  In "manual test 1B", the node (**48:ca:43:9b:53:d8, we call B**) has designated **30:30:f9:33:72:b8 (we call A)** as its parent, it will only connect to this parent and will not attempt to connect to any other node. When the node A crashes and reboots, the node B will fail to find the AP and will return a 201 error. It will only successfully reconnect once the parent has fully recovered.
`W (6278) mesh_main: <PARENT>ESPM_3372B8, layer:2/23, assoc:0/6, 1, 30:30:f9:33:72:b9, channel:1, rssi:-19`
 
[image]

4. In "manual test 2A", the node A initially selected `ESPM_9B54C0` as its parent, `I (27638) mesh_main: <MESH_EVENT_PARENT_CONNECTED>layer:0-->2, parent:48:ca:43:9b:54:c1<layer2>, ID:77:77:77:77:77:76`. 
After rebooting, it chose `ESPM_9B53D8` as its parent instead. `W (6258) mesh_main: <PARENT>ESPM_9B53D8, layer:3/22, assoc:0/6, 0, 48:ca:43:9b:53:d9, channel:1, rssi:-32`. The connection fail shows the parent has changed to `IDLE`.
  ```
  I (9888) wifi:state: init -> auth (0xb0)
  I (9898) wifi:state: auth -> init (0x65c0)
  I (9898) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1, snd_ch_cfg:0x0
  I (9898) mesh: [wifi]disconnected reason:101(idle), continuous:1/max:12, non-root, vote(,stopped)<><>
  I (9898) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:101
  ```
5. In "manual test 2B", the node B selects node A as its parent.

michaelsimp, 2024-11-08:
Hi.  Thanks for your response.

My apologies, I thought I was working on the original "manual_networking project, but this was my test version which I did modify.

I thought MESH_SET_NODE meant I could set the MESH_NODES parent (to a closer node with RSSI), but I think you are saying it is designed set the MESH_TYPE as MESH_NODE only which I guess this explains why I don't later get a MESH_ROOT.  Please confirm.

So can you give me a lead on how I achieve my objective as I documented in my last post.  I am looking for a hybrid between example "ip_internal_networking" where everything is automatic and example "manual_networking" which is designed to manually set a fixed MESH_ROOT and MESH_NODES.

I am guessing my problem lies in my use of `esp_mesh_set_parent()` where this example always passes in a type of "my_type" so a MESH_NODE will be set to type MESH_NODE.  Does this then stop the node scanning to become MESH_ROOT.

I am guessing some how I need to restart the scanning for the router with highest RSSI for the voting process but I don't know how to tell when I have lost the MESH_ROOT in the network, and then how to trigger the rescan and vote to enable a MESH_NODE or MESH_IDLE to become MESH_ROOT again.  This all happens automatically on start up before my use of esp_mesh_set_parent().

I really appreciate your help to better understand this.

zhangyanjiaoesp, 2024-11-08:
1. In the "manual_networking" example, the "MESH_SET_NODE" meant its type will be MESH_NONE, it can't be ROOT
2. To achieve your goal, I suggest initially using a self-organized network. Once the network is formed, if a node detects that its parent’s RSSI < -60dBm, it can initiate a scanning process to reselect a new parent. When the root node disappears, an `MESH_EVENT_NETWORK_STATE` event will be triggered. The nodes can then read this event and call `esp_mesh_set_self_organized(true, true)` to vote a new root. 
3. Actually, in a self-organized network, after the initial network formation, non-root nodes will periodically scan the surrounding nodes and select a better parent, dynamically adjusting the network, as described in the [document](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/esp-wifi-mesh.html#parent-node-switching) I mentioned previously. I believe this dynamic adjustment aligns with your requirements.
Have you observed this behavior? The related logs are as follows: 
```
I (23076) mesh: 5170<active>parent layer:1(node), channel:5, rssi:-21, assoc:1(cnx rssi threshold:-120)my_assoc:0
I (24576) mesh: 5969<scan>parent layer:1, rssi:-20, assoc:1(cnx rssi threshold:-120)
I (24576) mesh: [SCAN][ch:5]AP:1, other(ID:0, RD:0), MAP:1, idle:0, candidate:1, root:1, topMAP:0[c:2,i:2][ac:84:c6:7e:31:2b]<weak>
I (24586) mesh: 7391[weak]try rssi_threshold:-120, backoff times:0, max:5<-78,-82,-85>
I (24596) mesh: 716[monitor]no change, parent:4c:11:ae:eb:83:45, rssi:-20
I (24596) mesh: 2012<arm>parent monitor, my layer:2(cap:6)(node), interval:5231ms, retries:3<>
```
if there is no better parent, the node will remain with its current parent;
```
I (24596) mesh: 716[monitor]no change, parent:4c:11:ae:eb:83:45, rssi:-20
```
if a better parent is found, it will switch to that parent.
```
mesh: 710[monitor]new parent:30:ae:a4:80:1e:ed<layer:2, rssi:-33, assoc:0>current parent:30:ae:a4:00:48:59<layer:3, rssi:-31, assoc:4>
```

michaelsimp, 2024-11-08:
Thank you for this information
Is the MESH_EVENT_NETWORK_STATE  event only triggered by a MESH_ROOT node disappearing or is there additional testing I need to do before calling `esp_mesh_set_self_organized(true, true)` to vote a new root. Its just the name of this event "MESH_EVENT_NETWORK_STATE" sounds pretty broad and not at all like MESH_ROOT disappeared?

I had read in the docs about the self-organized network non-root nodes periodically scanning the surrounding nodes and selecting a better parent, but I have had multiple cases where a node  at one end of the network location wise has connected with a node at the opposite end of the network location wise and resulting in RSSI figures < -70dBm when there are much closer nodes in between, and they have NOT automatically switched.  How often should it scan and switch? 

When I added my manual scan and switching code (with the fixed MESH_ROOT) it formed a much better network and the performance was very noticeably better.

I will try this over the weekend and report back.

michaelsimp, 2024-11-10:
Hi

To recap,

I have created a task which runs every 60 seconds to check if the node RSSI < -60dBm and if so it calls wifiMeshScan(), to scan for a better parent.  For now I have commented this out and instead call wifiMeshScan() (see below) manually from my CLI console .
This task also monitors a selfOrganizeReactivateTimer which calls esp_mesh_set_self_organized(true, false) after the scan and switch parent is completed.

```
static uint8_t meshTaskPrescale = MESH_SCAN_INTERVAL / 2; // rescan in 10 seconds
void vWifiMeshTaskCode(void* args) {
    ESP_LOGI(TAG, "vWifiMeshTaskCode started");
    ESP_ERROR_CHECK(taskRegisterToWatch(xWifiMeshHandle, "appWiFiMesh")); // register task for watchdog
    while (true) {
        sleepMs(1 * MILLISECOND_TICKS_PER_SECOND);
        taskPatWdog(xWifiMeshHandle); // pat watchdog
#ifdef ENABLE_PARENT_SEARCH
        // if (meshTaskPrescale++ > MESH_SCAN_INTERVAL) {
        //     meshTaskPrescale = 0;
        //     if ((!esp_mesh_is_root()) && (currentRSSI < RSSI_RESCAN_THRESHOLD)) 
        //         wifiMeshScan();
        // }

        if (selfOrganizeReactivateTimer > 0) {
            selfOrganizeReactivateTimer--;
            if (selfOrganizeReactivateTimer == 0) {
                    ESP_LOGI(TAG, "Reactivate self organization");
                    ESP_ERROR_CHECK(esp_mesh_set_self_organized(true, false));
            }
        }
#endif
    }
}
```

Here is my wifiMeshScan():
```
void wifiMeshScan() { // for closer parent
    if (esp_mesh_is_root()) {
        ESP_LOGE(TAG, "Error: This node is MESH_ROOT");
        return;
    }
#ifndef ENABLE_PARENT_SEARCH
    ESP_LOGE(TAG, "WifiMeshScan for closer parent disabled");
    return;
#endif // ENABLE_PARENT_SEARCH

    if (esp_mesh_get_type() != MESH_IDLE) {
        esp_err_t err;
        err = esp_wifi_sta_get_rssi(&currentRSSI);
        if (err == ESP_OK) {
            ESP_LOGI(TAG, "WifiMeshScan for closer parent. Current parent RSSI: %d", currentRSSI);
        } else {
            ESP_LOGE(TAG, "WifiMeshScan for closer parent. Get RSSI error: %s", esp_err_to_name(err));
        }
    } else {
        currentRSSI = -100; // if not parent, default low RSSI        
    }
    selfOrganizeReactivateTimer = SELF_ORGANIZE_REACTIVATE_LONG_TIME; // enable long reenable timer as backup in case scan does not complete
    ESP_ERROR_CHECK(esp_mesh_set_self_organized(false, false));
    esp_wifi_scan_stop();
    wifi_scan_config_t scan_config = { 0 }; // mesh softAP is hidden
    scan_config.show_hidden = 1;
    scan_config.scan_type = WIFI_SCAN_TYPE_PASSIVE;
    esp_wifi_scan_start(&scan_config, 0);
}

```

and in my mesh_event_handler I have added:
```
[... 123 lines cut ...]
None of these log files show the `MESH_EVENT_NETWORK_STATE`  firing and calling `esp_mesh_set_self_organized(true, true)`
So as a test I added a CLI command to call  `esp_mesh_set_self_organized(true, true)` and tried to use it to rectify the problem above.  It does work some of the time creating a MESH_ROOT and bringing some nodes back, but not all nodes.  Some stay stuck on MESH_IDLE.

In summary:
```
    esp_mesh_set_self_organized(false, false);
    esp_wifi_scan_stop();
    wifi_scan_config_t scan_config = { 0 }; // mesh softAP is hidden
    scan_config.show_hidden = 1;
    scan_config.scan_type = WIFI_SCAN_TYPE_PASSIVE;
    esp_wifi_scan_start(&scan_config, 0);
```
When I find a better parent:
```
    esp_mesh_set_parent(&parent, (mesh_addr_t *)&parent_assoc.mesh_id, my_type, my_layer)
    ....
    esp_mesh_set_self_organized(true, false);
```
When root disappears:
```
   case MESH_EVENT_NETWORK_STATE:  (and if is_rootless)
   esp_mesh_set_self_organized(true, true);
```
The problem:
    `MESH_EVENT_NETWORK_STATE` often does not fire and call `esp_mesh_set_self_organized(true, true)`

zhangyanjiaoesp, 2024-11-11:
> Thank you for this information Is the MESH_EVENT_NETWORK_STATE event only triggered by a MESH_ROOT node disappearing or is there additional testing I need to do before calling [...]

1. The `MESH_EVENT_NETWORK_STATE` event is triggered by a Layer 2 nodes and propagates progressively through the network to the final node. The event only records a "rootless" state underneath.
[image]

3. The scanning period is not fixed, but is at least 4.5 seconds.

zhangyanjiaoesp, 2024-11-11:
> 
> I end up with no MESH_ROOT and MESH_NODES on deep layers all point directly or indirectly to a MESH_IDLE node. See my logs attached.
> [10Nov.zip](https://github.com/user-attachments/files/17689819/10Nov.zip)

Could you provide a more complete log? I need to understand the network state before the root reboot. Specifically, how Node1, Node2, and Node3 were connected, and what the behavior was after the reboot.

michaelsimp, 2024-11-11:
Hi
See logs attached, from boot till end.
[11Nov.zip](https://github.com/user-attachments/files/17695226/11Nov.zip)

I have all 4 nodes in my office for convenience.
I have disabled the periodic scan for closer wifi and invoke this from my CLI console - see the command "wifi scan".  It looks like I only sent this on nodes 2 and 3 this time, but the results are pretty typical of when it goes wrong:

- Event `<MESH_EVENT_NETWORK_STATE>` did not fire so I didn't call `esp_mesh_set_self_organized(true, true);`
- ended up with broken mesh

Ignore the watchdog in MQTT. I think the library blocks a bit when WiFi is not up.

- test2 root.txt: MAC: 30:30:f9:33:72:b8 MESH_ROOT powered off at the end of the file and left off
- test2 node1.txt:  MAC: 48:ca:43:9b:53:d8 After MESH_ROOT powered of, this MESH_NODE becomes MESH_IDLE
- test2 node2.txt:  MAC: 48:ca:43:9b:5c:68 After MESH_ROOT powered of, this MESH_NODE layer 2 becomes MESH_NODE layer 3 parent of 48:ca:43:9b:53:d9
- test2 node3.txt:  MAC: 30:30:f9:33:49:bc After MESH_ROOT powered of, this MESH_NODE becomes MESH_NODE layer 4 parent of 48:ca:43:9b:5c:69

michaelsimp, 2024-11-12:
Hi Zhangyanjiaoesp
Just checking on the status of this to make sure you have everything you need and that you will get back to me with some ideas soon.  Thanks

zhangyanjiaoesp, 2024-11-12:
@michaelsimp 
Is this the expected behavior? 
1. Initially, the network is self-organized, forming the initial network. 
2. Then, non-root nodes disable self-organization and periodically scan for nearby APs, selecting a better parent. 
3. When the root node disappears, the other nodes re-enable self-organization, reselect the root, and restore the network.

michaelsimp, 2024-11-12:
Answering your questions:
_When would you invoke the scan via the CLI console? Is it at any given moment?_
YES

_Can you disable the MQTT task watchdog? I'm not sure if it will interfere with the entire process._
OK

_Why are there so many reboot operations in the root log?_
Sorry I should have deleted up until the last one.  I did a few tests, some successful before the final one I sent you.  Reference the last test run from line 1672 onwards which matches the node traces I sent.

_Is this the expected behavior?_
_Initially, the network is self-organized, forming the initial network._
Yes 

_Then, non-root nodes disable self-organization and periodically scan for nearby APs, selecting a better parent._
Yes

_When the root node disappears, the other nodes re-enable self-organization, reselect the root, and restore the network._
Yes.  This part is the problem as OFTEN `MESH_EVENT_NETWORK_STATE` does not fire and `call esp_mesh_set_self_organized(true, true)`.  If I call `call esp_mesh_set_self_organized(true, true)` manually from my CLI, it doesn't fix the broken mesh.

If it helps, I could make a cut down project and send it to you to test and play with.  It doesn't need any special hardware, I am just using ESP32-S3-WROOM-1 dev boards with 16MB Flash

zhangyanjiaoesp, 2024-11-12:
> If it helps, I could make a cut down project and send it to you to test and play with. It doesn't need any special hardware, I am just using ESP32-S3-WROOM-1 dev boards with 16MB Flash

Sure, it would be better if you could provide a demo. That way, I can test and debug it locally.

michaelsimp, 2024-11-12:
Hi

See attached wincut folder structure of my project minus the build directory.

[wincut.zip](https://github.com/user-attachments/files/17711153/wincut.zip)

I have cut out the application but left all of the platform stuff.  Its still quite big but you will probably only be interested in the source at:
 ..\wincut\main\platform\espidf\wifi where wifiMesh.cpp is developed out of the example project "ip_internal_network"

Typing "help" will bring up a menu.

Your targets probably won't be wifi provisioned yet but if it is you can run "wifi unprovision" to clear it out.  On restart it will display the provisioning screen on the console with the QRCode for the ESP provisioning app.

After provisioning (all your nodes), they will start up normally and form a self organized mesh.

Type "mesh" to display the nodes WiFi mesh attributes including the Mesh routing Table as follows:

W (00:14:26.671) aWifiMesh: MESH_ROOT  Layer: 1  MAC: 30:30:f9:33:49:bc  IP: 192.168.1.13
I (00:14:26.674) aWifiMesh: Router VONETS  BSSID (MAC): 00:17:13:20:bd:74
I (00:14:26.686) aWifiMesh: Vote: 100%
I (00:14:26.687) aWifiMesh: Network is: self organized
I (00:14:26.688) aWifiMesh: Network Root is: not fixed
I (00:14:26.700) aWifiMesh: Number of nodes: 4
I (00:14:26.701) aWifiMesh: Max layers: 6
I (00:14:26.713) aWifiMesh: Mesh Parent MAC: 00:17:13:20:bd:74  RSSI: -45  Channel: 3
I (00:14:26.715) aWifiMesh: Mesh routing Table records: 4   Record size: 6   Table size: 24
I (00:14:26.728) aWifiMesh:   MAC: 30:30:f9:33:49:bc
I (00:14:26.729) aWifiMesh:   MAC: 30:30:f9:33:72:b8
I (00:14:26.742) aWifiMesh:   MAC: 48:ca:43:9b:5c:68
I (00:14:26.744) aWifiMesh:   MAC: 48:ca:43:9b:53:d8

The MQTT app will try to connect to a Mosquito Broker on 192.168.1.2.  If you want to disable this the MQTT is in the file:
..\wincut\main\host\mqtt5app.cpp. Comment out the last line of code which starts my MQTT task.

Type "wifi scan" on a NON MESH_ROOT to get it to search for a better parent and switch.  It won't run on a MESH_ROOT.

Reboot the MESH_ROOT to reproduce the broken Mesh issue.

Type "wifi root" to run call esp_mesh_set_self_organized(true, true)

[43 more replies]

closed 2024-11-13 by espressif-bot