espressif/esp-idf #14721: Hardware Baud Rate Detection for ESP32 S3

LYNHQQ, 2024-10-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I used the following code for baud rate detection:
uint32_t uart_baud_detect()
{
    // static uint16_t delay_cnt = 0;
    if(REG_GET_BIT(UART_CONF0_REG(1), UART_AUTOBAUD_EN) == 0) // Enable auto baud rate detection
    {
        REG_WRITE(UART_RX_FILT_REG(1), (2 << UART_GLITCH_FILT_S) | UART_GLITCH_FILT_EN);
        REG_WRITE(UART_LOWPULSE_REG(1), 4095); // Reset register to max value
        REG_WRITE(UART_HIGHPULSE_REG(1), 4095); // Reset register to max value
        REG_SET_BIT(UART_CONF0_REG(1), UART_AUTOBAUD_EN); // Enable autobaud    
    }
    while(REG_GET_BIT(UART_CONF0_REG(1), UART_AUTOBAUD_EN) && REG_READ(UART_RXD_CNT_REG(1)) < 300)
    {
        
    }
    uint32_t low_period  = REG_READ(UART_LOWPULSE_REG(1));
    uint32_t high_period = REG_READ(UART_HIGHPULSE_REG(1));
    REG_CLR_BIT(UART_CONF0_REG(1), UART_AUTOBAUD_EN); // Disable autobaud
    REG_CLR_BIT(UART_RX_FILT_REG(1), UART_GLITCH_FILT_EN); // Disable glitch filtering
    printf("high_period:%ld\n", high_period);
    printf("low_period:%ld\n", low_period);
    return 80000000 / low_period;
}

However, I found that during use, it is only effective the first time. When I change the baud rate on the sending end, it still gives the value detected during the first power-up and does not change."

ammaree, 2024-10-21:
Any success in solving this problem?

I am hoping that you are making progress to resolve this since AutoBaud is on my todo list as well...

LYNHQQ, 2024-10-23:
> 解决这个问题有什么成功吗？
> 
> 我希望您在解决这个问题方面取得进展，因为 AutoBaud 也在我的待办事项列表中......
#include <soc/soc.h>
#include "soc/uart_reg.h"
uint32_t uart_baud_detect()
{
    // static uint16_t delay_cnt = 0;
    if(REG_GET_BIT(UART_CONF0_REG(1),UART_AUTOBAUD_EN)==0)//使能自动波特率检测
    {
        REG_WRITE(UART_RX_FILT_REG(1), (2 << UART_GLITCH_FILT_S) | UART_GLITCH_FILT_EN);
        REG_WRITE(UART_LOWPULSE_REG(1), 4095); // reset register to max value
        REG_WRITE(UART_HIGHPULSE_REG(1), 4095); // reset register to max value
        REG_SET_BIT(UART_CONF0_REG(1), UART_AUTOBAUD_EN); // enable autobaud    
    }
    while(REG_GET_BIT(UART_CONF0_REG(1), UART_AUTOBAUD_EN) && REG_READ(UART_RXD_CNT_REG(1))<300)
    {
        
    }
    uint32_t low_period  = REG_READ(UART_LOWPULSE_REG(1));
    uint32_t high_period = REG_READ(UART_HIGHPULSE_REG(1));
    REG_CLR_BIT(UART_CONF0_REG(1), UART_AUTOBAUD_EN); // disable autobaud
    REG_CLR_BIT(UART_RX_FILT_REG(1), UART_GLITCH_FILT_EN); // disable glitch filtering
    printf("high_period:%ld\n",high_period);
    printf("low_period:%ld\n",low_period);
    return 80000000/low_period;
}
Here is my progress on the automatic baud rate. The following function retrieves values from AutoBaud and returns the closest actual baud rate. However, during testing, I found that it only works on the first attempt. For subsequent loop checks, it tends to remain unchanged, even if you change the input baud rate。OF COURSE, YOU CAN ALSO REFER TO THE AUTOBAUD IMPLEMENTED BY ELRS.

closed 2025-04-02 by espressif-bot

eriksl, 2025-04-02:
Interesting. The commit includes all sorts of stuff relating to I2C. It looks like the patch isn't quite clean?

eriksl, 2025-04-23:
This means?

igrr, 2025-04-23:
> The commit includes all sorts of stuff relating to I2C. It looks like the patch isn't quite clean?

That's correct — previously, `uart_test_i2c_master_freq` test case used LL functions related to UART auto-baud detection as a way of checking the clock frequency of I2C driver. (I2C CLK output was routed to UART RX.)

After the refactoring and the introduction of the public UART baud rate detection API, I2C test cases had to be updated accordingly.

> This means?

I guess you are referring to the backlink — it is automatically generated by this third-party pull request — https://github.com/blog2i2j/espressif.._..esp-idf/pull/14. This someone's esp-idf fork, it's not related to espressif's workflow.