espressif/esp-idf #14722: fatal error: mbedtls/atca_mbedtls_wrap.h: No such file or  directory

hitecSmartHome, 2024-10-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3.0

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

To compile

### What is the actual behavior?

Does not compile. `atca_mbedtls_wrap.h` does not exists.

### Steps to reproduce.

define CONFIG_ESP_TLS_USE_SECURE_ELEMENT

### Build or installation Logs.

```plain
Compiling .pio\build\esp-wrover-kit\esp_adc\adc_continuous.c.o
C:\Users\Pc\.platformio\packages\framework-espidf\components\esp-tls\esp_tls_mbedtls.c:33:10: fatal error: mbedtls/atca_mbedtls_wrap.h: No such file or 
directory
   33 | #include "mbedtls/atca_mbedtls_wrap.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\esp-wrover-kit\esp-tls\esp_tls_mbedtls.c.o] Error 1
Compiling .pio\build\esp-wrover-kit\esp_adc\esp32\adc_dma.c.o
======================= [FAILED] Took 226.36 seconds =======================
```

closed 2024-10-15 by hitecSmartHome

hitecSmartHome, 2024-11-20:
I think I disabled use secure element in mbedtls in the menuconfig but iam not sure

AdityaHPatwardhan, 2024-11-21:
Hi @hitecSmartHome,
I think this was due to `CONFIG_MBEDTLS_ATCA_HW_ECDSA_SIGN` option being disabled at your end. This option internally sets `MBEDTLS_ECDSA_SIGN_ALT` without which the above failure is observed in `esp-cryptoauthlib`
Can you please share your sdkconfig to confirm the above.

@torntrousers 
As you see, we didn't close the issue, it was closed by the OP and we thought it was closed as it has been self-resolved.

hitecSmartHome, 2024-11-21:
Yes, It has been self resolved. I remember I tweaked some mbedtls options in the menuconfig at that time and I enabled or disabeld something I tought I needed and got this error. Really need more description for the options in the menuconfig. A lot of things are not labeled properly. Or lacks description