espressif/esp-idf #14723: Wrong condition in HTTP version check routine (esp_http_server component)

valkuc, 2024-10-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

ESP32-C3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

ESP HTTP server component should refuse to execute HTTP 1.0 request.

### What is the actual behavior?

ESP HTTP server component executing HTTP 1.0 request.

### Steps to reproduce.

Use curl to issue HTTP 1.0 request to ESP board:

`curl -0 -v http://esp-board.local`

### More Information.

There is an error in condition here https://github.com/espressif/esp-idf/blob/6e5a178b3120dced7fa5c29c655cc22ea182df3d/components/esp_http_server/src/httpd_parse.c#L89
Logical `AND` should be replaced with 'OR':

```
if ((parser->http_major != 1) || (parser->http_minor != 1))
```

nileshkale123, 2024-10-16:
Hello @valkuc 

Thank you for bringing this change to our attention. We've raised a merge request, which will be reflected in the main repository soon.

closed 2024-10-28 by espressif-bot

AxelLin, 2024-11-02:
@nileshkale123 The release branches also need fix.

nileshkale123, 2024-11-04:
Sure. It's already in progress.

Thanks