espressif/esp-idf #14726: [Use 2MB PSRAM] API xQueueCreate(uxQueueLength, uxItemSize) failed to create at ESP-IDF v5.3.1 , but v5.0.7 is okay.

Sky-Soo-Ap, 2024-10-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

ESP32-D0WDR2-V3-V3 (revision v3.1)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I expected API xQueueCreate to create a larger queue length, like the following codes.

#define EXTERNAL_2M_PSRAM   2000000     // ~2M PSRAM
#define REMAIN_FREE_HEAP_SIZE_FOR_FIRWARE   200000  // ~200Kbytes

    struct notifyEventStruct notifyEventTx;
    uint32_t min_free_heap = esp_get_minimum_free_heap_size();
    if (min_free_heap > EXTERNAL_2M_PSRAM) {
        uint32_t notify_queue_size = (min_free_heap - REMAIN_FREE_HEAP_SIZE_FOR_FIRWARE) / sizeof(notifyEventTx);
        notifyEventQueue = xQueueCreate(notify_queue_size, sizeof(notifyEventTx));
    }

### What is the actual behavior?

The API xQueueCreate failed to create queue at ESP-IDF v5.3.1, but v5.0.7 is okay.
The only difference is the ESP-IDF version.
[sdkconfig.txt](https://github.com/user-attachments/files/17375074/sdkconfig.txt)

### Steps to reproduce.

As described above.

ESP-Marius, 2024-10-16:
FreeRTOS functions were changed to always allocate objects to internal memory (since some of these objects can be accessed when cache is disabled, thus it is not safe to put them in PSRAM by default).

If you want to allocate them in PSRAM you can specify that explicitly, see docs [here](https://docs.espressif.com/projects/esp-idf/en/v5.3.1/esp32/api-reference/system/freertos.html#freertos-heap)

Sky-Soo-Ap, 2024-10-16:
Hi Marius,
Thank you for providing solutions.
It works after changing to using API xQueueCreateWithCaps(UBaseType_t uxQueueLength, UBaseType_t uxItemSize, UBaseType_t uxMemoryCaps).
And the third parameter is 
#define MALLOC_CAP_SPIRAM    (1<<10) ///< Memory must be in SPI RAM
So, I closed this issue.

closed 2024-10-16 by Sky-Soo-Ap