espressif/esp-idf #14729: BLE Init Malloc failed

emesamwong, 2024-10-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am developing a matter device with LCD & touch panel (lvgl component V7.11.0). When starting the Matter Provisioning (BLE & WiFi), it contains error as below:
[image]
the LCD buff size is 240 * 240 * 2 / 6
Here is the image size info:
[image]

Since I am aware that the static IRAM is 100% used, I tried to decrease the LCD buffer size, there was no error but the Matter Commissioning was not working, and I could not scan the device. the LCD buff size is 240 * 240 * 2 / 10
[image]
Here is the image size info:
[image]

When the LCD buff size is 240 * 240 * 2 / 15, scan the device successfully but still pairing failed
[image][image]

Here is the image size info:
[image]

Chip-tool Log show: CHIP Error 0x00000003: Incorrect state
[ChipTool Log.txt](https://github.com/user-attachments/files/17376495/ChipTool.Log.txt)

The LCD buffer is using DRAM.
According to my testing, seems the BLE/WiFi/Matter-related resource crashes with LCD which contains a huge buffer.

1. How to check the BLE/WiFi/Matter-related resource sizes like how many sizes of IRAM/DRAM for BLE/WiFi/Matter-releated.
2. How to fix the "BLE_INIT: Malloc failed" error and pair the Devive to Matter nwk via chip-tool?
3. Would it be another issue that led to the pairing failure? 

Please me know if more information is required. Thank you

Environment:
esp-matter:  SHA: 2f692ac10a
connectedhomeip: SHA: 8e3d98c0
esp-idf: SHA: 482a8fb (Tag: v5.1.2)
ESP32-S3-WROOM-1 N16R8
Linux 
VS code

bryghtlabs-richard, 2024-10-15:
Can you try moving things out to PSRAM to free up D/IRAM?

emesamwong, 2024-10-16:
@bryghtlabs-richard I tried to enable the PSRAM as below:
[image]
However, the project build failed as below:
[image]

Once I disabled the "Allow .bss segment placed in external memory", the project build was successful but the BLE pairing issue was not fixed. As I know, the buffer is set to PSRAM when the option is enabled. Is that correct?

Is any configuration required for the PSRAM?

bryghtlabs-richard, 2024-10-16:
I'm not sure. That configuration looks similar to mine. You could try reducing "maximum malloc size(in bytes) to always put in internal memory", maybe try 2048?

You might also look into what error code esp_nimble_hci_init() is returning. Is it due to out of memory? Or another reason?

emesamwong, 2024-10-17:
@bryghtlabs-richard  I found the fix of PSRAM in #14729, and updated the ESP-IDF to v5.1.4 SHA: 632e0c2a9. 
Now I build successfully when the PSRAM config is enabled as in the pic above. Also,  I updated the "maximum malloc size(in bytes) to always put in internal memory" to 2048.
 
Before I enabled the PSRAM config, I checked the error code esp_nimble_hci_init() which is no "ESP_ERR_NO_MEM" (0x101)
[image]
 
 
After the IDF is upgraded and the PSRAM configuration enabled, the DUT always Reboots without any error code. 
[image]
The only hits is the reset reason which is "TG1WDT_SYS_RST". 
Any idea about this Reboot situation? 
What is the meaning of "Saved PC:0x40375f25"?

bryghtlabs-richard, 2024-10-17:
>TG1WDT_SYS_RST

This is a watchdog reset. You might read https://esp32.com/viewtopic.php?t=6469

Is this a new hardware design? Have you confirmed you have enough power available?

I'm not sure why there's no backtrace, but if you have a debugger available, try setting a breakpoint at "Saved PC" - I believe the value of this is capture when an exception like TG1WDT_SYS_RST occurs.