espressif/esp-idf #14730: Wi-Fi connection gets stuck on a specific Wi-Fi network

matoha, 2024-10-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board, incl. ESP32-S3-WROOM-1U

### Power Supply used.

USB

### What is the expected behavior?

Hi,

We have been successfully using the ESP32-S3 for a range of our products. However, at a specific customer network, we're unable to get the devices to connect and as it is a building-wide Wi-Fi, we're unable to change the Wi-Fi settings. The device connection gets stuck in a connecting mode and no connection happens. I've attached logs below for a successful vs unsuccessful connection, both taken using the same device.

The only difference I can see is that the network that works  report `wifi:connect_op: status=0, auth=5, cipher=3` whereas the one that doesn't work shows `wifi:connect_op: status=0, auth=5, cipher=12` . However, I'm unable to find what cipher=12 stands for as the driver is closed-source. The network with issues is b/g/n WPA2-PSK with CCMP/GCMP/CCMP256/GCMP256 support.

I would be grateful if you could point me to any issues the logs might be indicating?

Thanks
Martin

### What is the actual behavior?

The device never connects to Wi-Fi.

### Steps to reproduce.

The source code is based on the https://github.com/espressif/esp-idf/tree/release/v4.4/examples/wifi/getting_started/station , with some modifications. I'm unable to provide a way to remotely reproduce this as this is restricted to a very specific Wi-Fi network.

### Debug Logs.

```plain
[Wi-Fi connection is attempted to a network that doesn't work]
[Wi-Fi starts]

I (2666) wifi:wifi driver task: 3fcb07dc, prio:23, stack:6656, core=0
I (2711) wifi:wifi firmware version: eeaa27d
I (2724) wifi:wifi certification version: v7.0
I (2737) wifi:config NVS flash: enabled
I (2752) wifi:config nano formating: disabled
I (2765) wifi:Init data frame dynamic rx buffer num: 32
I (2778) wifi:Init management frame dynamic rx buffer num: 32
I (2801) wifi:Init static tx buffer num: 16
I (2814) wifi:Init tx cache buffer num: 32
[... 142 lines cut ...]
I (6953) wifi:state: init -> auth (b0)
D (6967) wifi:start 1s AUTH timer
D (6980) wifi:clear scan ap list
D (6995) wifi:recv auth: seq=2, status=0
I (7008) wifi:state: auth -> assoc (0)
D (7023) wifi:restart connect 1s timer for assoc
D (7056) wifi:recv assoc: type=0x10
D (7070) wifi:filter: set rx policy=6
I (7084) wifi:state: assoc -> run (10)
D (7097) wifi:start 10s connect timer for 4 way handshake
I (7123) wifi:connected with MatohaNet, aid = 3, channel 11, BW20, bssid = 04:95:e6:e4:5f:50
I (7138) wifi:security: WPA2-PSK, phy: bgn, rssi: -72
D (7152) wifi:remove all except 04:95:e6:e4:5f:50 from rc list
D (7164) wifi:clear blacklist
D (7188) wifi:filter: set rx policy=7
I (7202) wifi:pm start, type: 1
I (7216) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 14, mt_pti: 25000, mt_time: 10000
D (7229) wifi:Send sta connected event
D (7242) wifi:connect status 1 -> 5
D (7254) wifi:obss scan is disabled
D (7268) wifi:start obss scan: obss scan is stopped
I (7282) wifi:BcnInt:102400, DTIM:1

[Wi-Fi works here as expected]
```

Alvin1Zhang, 2024-10-16:
Thanks for reporting.

nishanth-radja, 2024-10-16:
@matoha have you enabled the suiteB support with 192bit key in the component config->supplicant in the menuconfig ?

hansw123, 2024-10-18:
@matoha 
auth 5 means wpa2_psk
cipher=12 means  AES_256_GCM  which looks not support in your wifi network
chiper = 3 means AES_CCM

matoha, 2024-10-18:
Hi @hansw123 ,

Thanks for the reply. Is there a way to disable cipher=12 on the ESP side, and force cipher=3, since it doesn't seem to be working? I've reviewed the documentation but can't find anything.

Thanks
Martin

hansw123, 2024-10-22:
@matoha 
sorry ,I made a mistake ,this log means ap need sta support AES_256_GCM,but station don\t support
just open ESP_WIFI_GCMP_SUPPORT in menuconfig can be ok

closed 2025-01-15 by Alvin1Zhang