espressif/esp-idf #14731: wifi softAP does not work on ESP32-S2FNR2 board

erfty125, 2024-10-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

1 I installed esp-idf refering to https://docs.espressif.com/projects/esp-idf/en/v5.3.1/esp32s2/get-started/index.html 
(linux debian12)
2 I flashed the softAP example from example directory and AP did not appeared
I tried flashing simular wifi AP example in arduino IDE and it works. Also wifi station example from esp-idf also works.

nopnop2002, 2024-10-16:
Can't find "WiFi SSID" in the list of SSIDs?

erfty125, 2024-10-16:
> Can't find "myssid" in the list of AP?

yes

nopnop2002, 2024-10-16:
Can you publish the runtime logs?

erfty125, 2024-10-16:
> Can you publish the runtime logs?

I dont know how to do this, because esp32-s2 appears in lsusb only in download mode

nopnop2002, 2024-10-16:
```
idf.py build monitor -p port
```

erfty125, 2024-10-16:
> ```
> idf.py build monitor -p port
> ```

ls /dev shows the same when i comnect the board to laptop

hansw123, 2024-10-17:
@erfty125 
maybe you can psot the Terminal logs for the entire process include  export  idf-env  build project and flash and the whole process

erfty125, 2024-10-17:
https://pastebin.com/XNkX3rGt

erfty125, 2024-10-17:
> @erfty125 maybe you can psot the Terminal logs for the entire process include export idf-env build project and flash and the whole process

https://pastebin.com/XNkX3rGt

hansw123, 2024-10-18:
@erfty125 
1. first you can use uart to flash the bin to device ,which can be set in menuconfig -----Component config > ESP System Settings > Channel for console output (default is UART)
2. when you use USB. you can open ESP_CONSOLE_USB_SERIAL_JTAG in menuconfig
3. you can follow the tips in log 
Error: ESP32-S2FNR2 (revision v1.0) chip was placed into download mode using GPIO0.
esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
To suppress this note, set --after option to 'no_reset'.
coal@coal:~/esp/softAP$

erfty125, 2024-10-18:
> @erfty125
>
> 1. first you can use uart to flash the bin to device ,which can be set in menuconfig -----Component config > ESP System Settings > Channel for console output (default is UART)
> 2. when you [...]

ok, i set usb chanbel for output, i can now read logs
here the logs

coal@coal:~/esp/softAP$ idf.py monitor
Executing action: monitor
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Detecting chip type... ESP32-S2
Running idf_monitor in directory /home/coal/esp/softAP
Executing "/home/coal/.espressif/python_env/idf5.3_py3.11_env/bin/python /home/coal/esp/esp-idf/tools/idf_monitor.py -p /dev/ttyACM0 -b 115200 --toolchain-prefix xtensa-esp32s2-elf- --target esp32s2 --revision 0 /home/coal/esp/softAP/build/wifi_softAP.elf -m '/home/coal/.espressif/python_env/idf5.3_py3.11_env/bin/python' '/home/coal/esp/esp-idf/tools/idf.py'"...
--- esp-idf-monitor 1.5.0 on /dev/ttyACM0 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
I (30) boot: ESP-IDF v5.3.1 2nd stage bootloader
I (30) boot: compile time Oct 18 2024 15:35:15
I (30) boot: chip revision: v1.0
I (30) boot.esp32s2: SPI Speed      : 80MHz
I (30) boot.esp32s2: SPI Mode       : DIO
I (30) boot.esp32s2: SPI Flash Size : 4MB
I (30) boot: Enabling RNG early entropy source...
I (31) boot: Partition Table:
I (31) boot: ## Label            Usage          Type ST Offset   Length
I (32) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (33) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (34) boot:  2 factory          factory app      00 00 00010000 00100000
I (35) boot: End of partition table
I (35) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=1f010h (126992) map
I (61) esp_image: segment 1: paddr=0002f038 vaddr=3ffc9390 size=00fe0h (  4064) load
I (62) esp_image: segment 2: paddr=00030020 vaddr=40080020 size=7a894h (501908) map
I (163) esp_image: segment 3: paddr=000aa8bc vaddr=3ffca370 size=028b0h ( 10416) load
I (165) esp_image: segment 4: paddr=000ad174 vaddr=40024000 size=15388h ( 86920) load
I (198) boot: Loaded app from partition at offset 0x10000
I (199) boot: Disabling RNG early entropy source...
I (220) main_task: Started on CPU0
I (220) main_task: Calling app_main()
I (230) wifi softAP: ESP_WIFI_MODE_AP
I (240) wifi:wifi driver task: 3ffd93e0, prio:23, stack:6656, core=0
I (250) wifi:wifi firmware version: ccaebfa
I (250) wifi:wifi certification version: v7.0
I (250) wifi:config NVS flash: enabled
I (250) wifi:config nano formating: disabled
I (250) wifi:Init data frame dynamic rx buffer num: 32
I (250) wifi:Init static rx mgmt buffer num: 5
I (250) wifi:Init management short buffer num: 32
I (250) wifi:Init dynamic tx buffer num: 32
I (260) wifi:Init static rx buffer size: 1600
I (260) wifi:Init static rx buffer num: 10
I (260) wifi:Init dynamic rx buffer num: 32
I (260) wifi_init: rx ba win: 6
I (260) wifi_init: accept mbox: 6
I (260) wifi_init: tcpip mbox: 32
I (260) wifi_init: udp mbox: 6
I (260) wifi_init: tcp mbox: 6
I (260) wifi_init: tcp tx win: 5760
I (260) wifi_init: tcp rx win: 5760
I (260) wifi_init: tcp mss: 1440
I (260) wifi_init: WiFi IRAM OP enabled
I (260) wifi_init: WiFi RX IRAM OP enabled
I (260) phy_init: phy_version 2600,6a8ccb9,Jun  4 2024,16:49:07
I (310) wifi:mode : softAP (80:65:99:e9:42:89)
I (420) wifi:Total power save buffer number: 16
I (420) wifi:Init max length of beacon: 752/752
I (420) wifi:Init max length of beacon: 752/752
I (430) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
I (430) wifi softAP: wifi_init_softap finished. SSID:myssid password:mypassword channel:1
I (430) main_task: Returned from app_main()

erfty125, 2024-10-18:
> > @erfty125
> > 
> > 1. first you can use uart to flash the bin to device ,which can be set in menuconfig -----Component config > ESP System Settings > Channel for console output (default is UART)
> > 2. when you use USB. you can open ESP_CONSOLE_USB_SERIAL_JTAG in menuconfig
> > 3. you can follow the tips in log
> >    Error: ESP32-S2FNR2 (revision v1.0) chip was placed into download mode using GPIO0.
> >    esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
> >    To suppress this note, set --after option to 'no_reset'.
> >    coal@coal:~/esp/softAP$
> 
> ok, i set usb channel for output, i can now read logs here the logs
> 
> coal@coal:~/esp/softAP$ idf.py monitor Executing action: monitor Serial port /dev/ttyACM0 Connecting... Detecting chip type... Unsupported detection protocol, switching and trying again... Detecting chip type... ESP32-S2 Running idf_monitor in directory /home/coal/esp/softAP Executing "/home/coal/.espressif/python_env/idf5.3_py3.11_env/bin/python /home/coal/esp/esp-idf/tools/idf_monitor.py -p /dev/ttyACM0 -b 115200 --toolchain-prefix xtensa-esp32s2-elf- --target esp32s2 --revision 0 /home/coal/esp/softAP/build/wifi_softAP.elf -m '/home/coal/.espressif/python_env/idf5.3_py3.11_env/bin/python' '/home/coal/esp/esp-idf/tools/idf.py'"... --- esp-idf-monitor 1.5.0 on /dev/ttyACM0 115200 --- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H I (30) boot: ESP-IDF v5.3.1 2nd stage bootloader I (30) boot: compile time Oct 18 2024 15:35:15 I (30) boot: chip revision: v1.0 I (30) boot.esp32s2: SPI Speed : 80MHz I (30) boot.esp32s2: SPI Mode : DIO I (30) boot.esp32s2: SPI Flash Size : 4MB I (30) boot: Enabling RNG early entropy source... I (31) boot: Partition Table: I (31) boot: ## Label Usage Type ST Offset Length I (32) boot: 0 nvs WiFi data 01 02 00009000 00006000 I (33) boot: 1 phy_init RF data 01 01 0000f000 00001000 I (34) boot: 2 factory factory app 00 00 00010000 00100000 I (35) boot: End of partition table I (35) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=1f010h (126992) map I (61) esp_image: segment 1: paddr=0002f038 vaddr=3ffc9390 size=00fe0h ( 4064) load I (62) esp_image: segment 2: paddr=00030020 vaddr=40080020 size=7a894h (501908) map I (163) esp_image: segment 3: paddr=000aa8bc vaddr=3ffca370 size=028b0h ( 10416) load I (165) esp_image: segment 4: paddr=000ad174 vaddr=40024000 size=15388h ( 86920) load I (198) boot: Loaded app from partition at offset 0x10000 I (199) boot: Disabling RNG early entropy source... I (220) main_task: Started on CPU0 I (220) main_task: Calling app_main() I (230) wifi softAP: ESP_WIFI_MODE_AP I (240) wifi:wifi driver task: 3ffd93e0, prio:23, stack:6656, core=0 I (250) wifi:wifi firmware version: ccaebfa I (250) wifi:wifi certification version: v7.0 I (250) wifi:config NVS flash: enabled I (250) wifi:config nano formating: disabled I (250) wifi:Init data frame dynamic rx buffer num: 32 I (250) wifi:Init static rx mgmt buffer num: 5 I (250) wifi:Init management short buffer num: 32 I (250) wifi:Init dynamic tx buffer num: 32 I (260) wifi:Init static rx buffer size: 1600 I (260) wifi:Init static rx buffer num: 10 I (260) wifi:Init dynamic rx buffer num: 32 I (260) wifi_init: rx ba win: 6 I (260) wifi_init: accept mbox: 6 I (260) wifi_init: tcpip mbox: 32 I (260) wifi_init: udp mbox: 6 I (260) wifi_init: tcp mbox: 6 I (260) wifi_init: tcp tx win: 5760 I (260) wifi_init: tcp rx win: 5760 I (260) wifi_init: tcp mss: 1440 I (260) wifi_init: WiFi IRAM OP enabled I (260) wifi_init: WiFi RX IRAM OP enabled I (260) phy_init: phy_version 2600,6a8ccb9,Jun 4 2024,16:49:07 I (310) wifi:mode : softAP (80:65:99:e9:42:89) I (420) wifi:Total power save buffer number: 16 I (420) wifi:Init max length of beacon: 752/752 I (420) wifi:Init max length of beacon: 752/752 I (430) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1 I (430) wifi softAP: wifi_init_softap finished. SSID:myssid password:mypassword channel:1 I (430) main_task: Returned from app_main()

erfty125, 2024-10-18:
> @erfty125
>
> 1. first you can use uart to flash the bin to device ,which can be set in menuconfig -----Component config > ESP System Settings > Channel for console output (default is UART)
> 2. when you [...]

i set usb channel for output
logs https://pastebin.com/29nV2c5j

hansw123, 2024-10-21:
@erfty125 
feel free to reopen

erfty125, 2024-10-21:
> @erfty125 feel free to reopen

Ok, i checked config and the problem was caused by using wpa3(by default) so i wasnt able to see AP on phone

closed 2024-12-17 by Alvin1Zhang