espressif/esp-idf #14735: [ESP-WIFI-MESH] esp_mesh_delete_group_id() not working as expected

mmrein, 2024-10-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

### Expected behavior:

Delete number of addresses matching those provided by pointer, leaving other addresses intact.

### Actual behavior:

Selected address is deleted, but resulting list of addresses after deletion is incorrect.

### Steps to reproduce:

1. Get some example where esp-wifi-mesh is used and initialized, for example: https://github.com/espressif/esp-idf/tree/v5.3.1/examples/mesh/internal_communication
2. Add following code to the end of main (after all init is done)
3. Flash to device and see log output

```c
// XXX Debug test group addr
    int addrCount = 0;
    mesh_addr_t *group_list = NULL;
    esp_err_t ret = ESP_FAIL;

// Initialize and set the list of 3 group addresses
    uint8_t group_ids[3][6] = {{0x01, 0x00, 0x5e, 0x00, 0x00, 0x01}, {0x01, 0x00, 0x5e, 0x00, 0x00, 0x02}, {0x01, 0x00, 0x5e, 0x00, 0x00, 0x03}};
    esp_mesh_set_group_id((mesh_addr_t *)group_ids, 3);

// Check current list
    addrCount = esp_mesh_get_group_num();
    ESP_LOGI(MESH_TAG, "Group addresses count: %i", addrCount);
    // Print addresses
    group_list = heap_caps_malloc(sizeof(mesh_addr_t) * addrCount, MALLOC_CAP_8BIT);
    esp_mesh_get_group_list((mesh_addr_t*)group_list, addrCount);
    for(int i=0; i<addrCount; i++) {
    	ESP_LOGI(MESH_TAG, "Group address %i: " MACSTR, i, MAC2STR((uint8_t*)&group_list[i]));
    }
    free(group_list);

// Delete last address
    ret = esp_mesh_delete_group_id((const mesh_addr_t*)group_ids[2], 1);
    ESP_LOGI(MESH_TAG, "Delete last address: %s", esp_err_to_name(ret));

// Check current list
    addrCount = esp_mesh_get_group_num();
    ESP_LOGI(MESH_TAG, "Group addresses count: %i", addrCount);
    // Print addresses
    group_list = heap_caps_malloc(sizeof(mesh_addr_t) * addrCount, MALLOC_CAP_8BIT);
    esp_mesh_get_group_list((mesh_addr_t*)group_list, addrCount);
    for(int i=0; i<addrCount; i++) {
    	ESP_LOGI(MESH_TAG, "Group address %i: " MACSTR, i, MAC2STR((uint8_t*)&group_list[i]));
    }
    free(group_list);
```

### Expected Log output:

```
I (2564) mesh_main: Group addresses count: 3 
I (2564) mesh_main: Group address 0: 01:00:5e:00:00:01 
I (2564) mesh_main: Group address 1: 01:00:5e:00:00:02 
I (2574) mesh_main: Group address 2: 01:00:5e:00:00:03 
I (2574) mesh_main: Delete last address: ESP_OK 
I (2584) mesh_main: Group addresses count: 2 
I (2584) mesh_main: Group address 0: 01:00:5e:00:00:01 
I (2594) mesh_main: Group address 1: 01:00:5e:00:00:02 
```

### Actual Log output:

```
I (2564) mesh_main: Group addresses count: 3 
I (2564) mesh_main: Group address 0: 01:00:5e:00:00:01 
I (2564) mesh_main: Group address 1: 01:00:5e:00:00:02 
I (2574) mesh_main: Group address 2: 01:00:5e:00:00:03 
I (2574) mesh_main: Delete last address: ESP_OK 
I (2584) mesh_main: Group addresses count: 2 
I (2584) mesh_main: Group address 0: 01:00:5e:00:00:02 
I (2594) mesh_main: Group address 1: 00:00:00:00:00:00 
```

zhangyanjiaoesp, 2024-10-22:
@mmrein Thanks for reporting this issue, we have found the root cause, and will merge the fix ASAP.

closed 2024-11-04 by espressif-bot

mmrein, 2024-11-15:
I've just seen [v5.4-beta1](https://github.com/espressif/esp-idf/releases/tag/v5.4-beta1) release and the fix does not seem included. Will it make to final v5.4 stable?

zhangyanjiaoesp, 2024-11-18:
yes, the final version will include this change.