espressif/esp-idf #14736: If mem_block_symbols and SOC_RMT_MEM_WORDS_PER_CHANNEL do not match, rmt_new_tx_channel will result in an error.

nopnop2002, 2024-10-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32S3 Development board

### Power Supply used.

USB

### What is the expected behavior?

rmt_new_tx_channel() does not result in an error

### What is the actual behavior?

rmt_new_tx_channel() results in an error

### Steps to reproduce.

1. Step
flash this code.
```
#include <stdio.h>
#include <inttypes.h>
#include "driver/rmt_tx.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"

static const char *TAG = "MAIN";

void app_main()
{
    rmt_channel_handle_t tx_channels [ 3 ] = { NULL };
    gpio_num_t tx_gpio_number [ 3 ] = { GPIO_NUM_16, GPIO_NUM_17, GPIO_NUM_18 };
    for (int i = 0; i < 3; i++) {
        ESP_LOGI(TAG, "setting up channel %d", i);
        rmt_tx_channel_config_t tx_chan_config = {
            .gpio_num = tx_gpio_number[i],
            .clk_src = RMT_CLK_SRC_DEFAULT,
            .resolution_hz = 1 * 1000 * 1000,  // 1MHz clock
#if 1
            .mem_block_symbols = 64,
#else
            .mem_block_symbols = 48,
#endif
            .trans_queue_depth = 1,
        };
    ESP_ERROR_CHECK(rmt_new_tx_channel(&tx_chan_config, &tx_channels[i]));
    }
}
```

### Debug Logs.

```plain
I (496) MAIN: setting up channel 2
E (586) rmt: rmt_tx_register_to_group(167): no free tx channels
E (586) rmt: rmt_new_tx_channel(303): register channel failed
```

### More Information.

If SOC_RMT_MEM_WORDS_PER_CHANNEL is 48 and mem_block_symbols is 64, 2 blocks of memory are required, but there is a bug when using 2 blocks of memory.

[https://github.com/espressif/esp-idf/bl ... #L148-L149](https://github.com/espressif/esp-idf/blob/master/components/esp_driver_rmt/src/rmt_tx.c#L148-L149)

florentbr, 2024-10-17:
Your example is trying to allocate 6 TX channels of memory space (48 * 2 * 3). ESP32S3 has only 4 TX channels of 48 words each.

nopnop2002, 2024-10-18:
thank you for your reply.
We were able to confirm the number of TX channels of the ESP32 series.
```
$ grep -rn "#define SOC_RMT_TX_CANDIDATES_PER_GROUP" *
esp32/include/soc/soc_caps.h:275:#define SOC_RMT_TX_CANDIDATES_PER_GROUP 8  /*!< Number of channels that capable of Transmit in each group */
esp32c3/include/soc/soc_caps.h:240:#define SOC_RMT_TX_CANDIDATES_PER_GROUP       2  /*!< Number of channels that capable of Transmit */
esp32c5/mp/include/soc/soc_caps.h:305:#define SOC_RMT_TX_CANDIDATES_PER_GROUP       2  /*!< Number of channels that capable of Transmit */
esp32c5/beta3/include/soc/soc_caps.h:310:#define SOC_RMT_TX_CANDIDATES_PER_GROUP       2  /*!< Number of channels that capable of Transmit */
esp32c6/include/soc/soc_caps.h:307:#define SOC_RMT_TX_CANDIDATES_PER_GROUP       2  /*!< Number of channels that capable of Transmit */
esp32c61/include/soc/soc_caps.h:302:#define SOC_RMT_TX_CANDIDATES_PER_GROUP       2  /*!< Number of channels that capable of Transmit */
esp32h2/include/soc/soc_caps.h:301:#define SOC_RMT_TX_CANDIDATES_PER_GROUP       2  /*!< Number of channels that capable of Transmit */
esp32p4/include/soc/soc_caps.h:371:#define SOC_RMT_TX_CANDIDATES_PER_GROUP       4  /*!< Number of channels that capable of Transmit in each group */
esp32s2/include/soc/soc_caps.h:257:#define SOC_RMT_TX_CANDIDATES_PER_GROUP       4  /*!< Number of channels that capable of Transmit in each group */
esp32s3/include/soc/soc_caps.h:266:#define SOC_RMT_TX_CANDIDATES_PER_GROUP       4  /*!< Number of channels that capable of Transmit in each group */
```

closed 2024-10-18 by nopnop2002