espressif/esp-idf #14739: ESP_ERR_MBEDTLS_PK_PARSE_KEY_FAILED when use mqtt connect with client_key_password

gl-liuqingbo, 2024-10-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

你好，我在 **esp-idf/tools/test_apps/protocols/mqtt/publish_connect_test** 例程下，修改ca.crt,client_pwd.crt,client_pwd.key为自签名的证书，也修改mqtt_cfg.credentials.authentication.key_password为密钥的密码，然后连接本地的mqtt服务器，但是一直得到这样的错误：
```E (34252) esp-tls-mbedtls: mbedtls_pk_parse_keyfile returned -0x1280
E (34252) esp-tls-mbedtls: Failed to set client pki context
E (34262) esp-tls-mbedtls: Failed to set client configurations, returned [0x8019] (ESP_ERR_MBEDTLS_PK_PARSE_KEY_FAILED)
E (34272) esp-tls: create_ssl_handle failed
E (34272) esp-tls: Failed to open new connection
E (34282) transport_base: Failed to open a new connection
E (34292) mqtt_client: Error transport connect
```
我的证书和密钥是没问题的，这一点我在使用另一个mqtt客户端时候验证过了。
我注意到当tls调用mbedtls_pk_parse_keyfile时，返回了MBEDTLS_ERR_PEM_UNKNOWN_ENC_ALG这样的错误，我不知道是当前esp-tls还不支持密钥密码的模式，还是说我缺少了某些配置？希望得到解答

gl-liuqingbo, 2024-10-17:
```
E (23092) esp-tls-mbedtls: mbedtls_pk_parse_keyfile returned -0x1280
E (23092) esp-tls-mbedtls: Failed to set client pki context
E (23092) esp-tls-mbedtls: Failed to set client configurations, returned [0x8019] (ESP_ERR_MBEDTLS_PK_PARSE_KEY_FAILED)
E (23112) esp-tls: create_ssl_handle failed
E (23112) esp-tls: Failed to open new connection
E (23122) transport_base: Failed to open a new connection
E (23132) mqtt_client: Error transport connect
 ```

gl-liuqingbo, 2024-11-06:
Hello, is there any relevant help, I would be greatly appreciated

euripedesrocha, 2024-11-06:
Hi @gl-liuqingbo from the error you have issues in your key file. Mbedtls error indicates that you have an unsupported algorithm.

gl-liuqingbo, 2024-11-06:
@euripedesrocha Thank you for your reply. I used openssl to generate the key through common RSA. My esp32s3 was able to parse the certificate without encrypting the key, but was unable to parse the certificate when encrypting the key. And this certificate can be successfully resolved when I use mosquitto mqtt client. I am very confused, is my esp32s3 missing any configuration?

euripedesrocha, 2024-11-06:
I would check the encryption algorithm used and if it is supported in mbedtls.

gl-chenxiaosheng, 2025-01-10:
Hi, @euripedesrocha 

How's your checkup going? Below is the command to encrypt the key, help if it is supported?
`openssl genrsa -out client_encypt.key -passout pass:test -des3 2048`

euripedesrocha, 2025-01-13:
@gl-chenxiaosheng I didn't run any investigation on this due to the missing information. Looking in the command provided, I noticed the usage of des3 algorithm that is known to be insecure. Have you checked if it is supported in mbedtls configuration/version in use? I suggest you try a different one.

Alvin1Zhang, 2025-03-28:
@gl-liuqingbo Thanks for reporting, would you please help share if any further updates? Thanks.