espressif/esp-idf #14744: How to prevent the entire project from being recompiled due to modifying CMakeLists.txt

UradaSources, 2024-10-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I want to use a library as an idf component but it doesn't come with its own cmake file for idf. There are some cross-board files in the library, which I don't need, but I'm not sure which ones I don't need. I can only keep modifying the idf_component_register.SRCS parameters to test, which is a terrible experience. Every time I add even one file to my SOURCE_FILES, it will cause the entire project of more than 1400 source files to be recompiled. I have never hated c++ so much. Is there any way to solve this problem? I also use arduino-esp32 as my component. Maybe there is a way to pre-compile some things as static libraries to reduce compilation time?

```cmake
set(SOURCE_FILES
    "src/Arduino_DataBus.cpp"
    ...)

idf_component_register(
    INCLUDE_DIRS "src"
    SRCS ${SOURCE_FILES}
    PRIV_REQUIRES "arduino"
)
```

dobairoland, 2024-10-18:
Hi @UradaSources. If I understood your situation correctly then in this case the best option would be to make sure that you are bulding with `ccache` enable. That means it will use the previously build source files from the cache if it hasn't been modified.

Depending on your OS and ESP-IDF version, it might be disabled by default. You can force it with the ``--ccache` command line argument:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-py.html#global-options

dobairoland, 2024-10-29:
I'm closing this for inactivity. Please let us know if this needs further discussion.

closed 2024-10-29 by dobairoland