espressif/esp-idf #14747: Console over IP ?

diplfranzhoepfinger, 2024-10-17:


nopnop2002, 2024-10-18:
Is this what you mean by Serial Console?

[Console Component](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/console.html#console)

diplfranzhoepfinger, 2024-10-18:
> By serial console, do you mean Console Component?
> 
> [Console Component](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/console.html#console)
> 

No

> Or do you mean the output destination of ESP_LOGx?

Yes. 
Log by ESP_LOGx and printf.

nopnop2002, 2024-10-18:
How to redirect ESP_LOGx.

https://github.com/nopnop2002/esp-idf-net-logging

diplfranzhoepfinger, 2024-10-18:
> How to redirect ESP_LOGx.
> 
> https://github.com/nopnop2002/esp-idf-net-logging

that is what i searched. 
Continue Discussion here: https://github.com/nopnop2002/esp-idf-net-logging/issues/5

sudeep-mohanty, 2024-10-18:
Closing the issue as this feature-request could be handled outside of IDF. Please feel free to reopen the ticket or open a new one for further support. Thank you.

closed 2024-10-18 by sudeep-mohanty