espressif/esp-idf #14748: How do I enable both 2.4G and 5G AP capabilities on the ESP32-C5

Sail-211010, 2024-10-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

How do I enable both 2.4G and 5G AP capabilities on the ESP32-C5，
And in the use of automatic switching or need to be in the program at a certain interval to switch channels

MaxwellAlan, 2024-10-18:
Hi @Sail-211010 

ESP32-C5 only support DBSC (Dual-Band Single Concurrent) not support DBDC (Dual-Band Dual Concurrent).

Thanks.

Sail-211010, 2024-10-18:
OK,Understood. Thank you.

closed 2024-10-21 by MaxwellAlan