espressif/esp-idf #14749: error : scene_recall at esp-idf/components/bt/esp_ble_mesh/mesh_models/server/time_scene_server.c:613

hipal31, 2024-10-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.3

### Espressif SoC revision.

ESP32C3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32C3

### Power Supply used.

USB

### What is the expected behavior?

In my BLE-MESH Device, after http OTA, scene server (scene recall) should work fine.

### What is the actual behavior?

After OTA, during scene recall it's crashing.
scene_recall at D:/Frameworks/esp-idf-v5.1.1/components/bt/esp_ble_mesh/mesh_models/server/time_scene_server.c:613

### Steps to reproduce.

1. Configure a ble-mesh node and add it in scene
2. Store scene_state and scenes in nvs
3. After reboot, restore scene data from NVS
4. do ota
5. After restoring , during scene recall it's giving error.

### Debug Logs.

```plain
I (1078) REEVA_NVS: Restore, data: 01 00 01 00 78 c6
I (1088) REEVA_BLE: Restore, net_idx 0x0001, app_idx 0x0001
I (1098) REEVA_BLE: ESP_BLE_MESH_NODE_SET_UNPROV_DEV_NAME_COMP_EVT, err_code 0
W (1098) BLE_MESH: bt_mesh_prov_enable, Already
I (1108) REEVA_BLE: ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code -120
I (1118) REEVA_BLE: BLE Mesh Node initialized
I (1118) REEVA_NVS: Restore, key "SCENE_STATE", length 16
I (1128) REEVA_NVS: Restore, data: 32 00 00 00 3c 46 c9 3f db ad 00 00 00 00 00 00
I (1138) REEVA_BLE: Restored SCENE_STATES
I (1138) REEVA_NVS: Restore, key "SCENE_ADDR", length 400
I (1148) REEVA_NVS: Restore, data: 69 ea 00 00 64 4c c9 3f db ad 00 00 4c 4c c9 3f
I (1158) REEVA_NVS: Restore, data: 00 00 00 00 34 4c c9 3f 00 00 00 00 1c 4c c9 3f
I (1158) REEVA_NVS: Restore, data: 00 00 00 00 04 4c c9 3f 00 00 00 00 ec 4b c9 3f
I (1168) REEVA_NVS: Restore, data: 00 00 00 00 d4 4b c9 3f 00 00 00 00 bc 4b c9 3f
I (1178) REEVA_NVS: Restore, data: 00 00 00 00 a4 4b c9 3f 00 00 00 00 8c 4b c9 3f
I (1188) REEVA_NVS: Restore, data: 00 00 00 00 74 4b c9 3f 00 00 00 00 5c 4b c9 3f
[... 151 lines cut ...]
I (488) cpu_start: Compile time:     Oct 18 2024 10:48:01
I (494) cpu_start: ELF file SHA256:  5167f8bb75f2d8f0...
I (500) cpu_start: ESP-IDF:          v5.1.3-1-g24f7f3b17c-dirty
I (506) cpu_start: Min chip rev:     v0.3
I (511) cpu_start: Max chip rev:     v1.99
I (516) cpu_start: Chip rev:         v0.3
I (521) heap_init: Initializing. RAM available for dynamic allocation:
I (528) heap_init: At 3FCA8050 len 00017FB0 (95 KiB): DRAM
I (534) heap_init: At 3FCC0000 len 0001C710 (113 KiB): DRAM/RETENTION
I (541) heap_init: At 3FCDC710 len 00002950 (10 KiB): DRAM/RETENTION/STACK
I (549) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM
I (556) spi_flash: detected chip: generic
I (560) spi_flash: flash io: dio
I (564) sleep: Configure to isolate all GPIO pins in sleep state
I (570) sleep: Enable automatic switching of GPIO sleep configuration
I (578) coexist: coex firmware version: 77cd7f8
I (583) coexist: coexist rom version 9387209
I (588) app_start: Starting scheduler on CPU0
I (593) main_task: Started on CPU0
I (593) main_task: Calling app_main()
I (613) RESET: REASON : 4
0x40048b82: software_reset in ROM
I (673) REEVA_NVS: Open namespace done, name "reeva_mesh"
I (683) REEVA_NVS: Restore, key "LIGHT_DATA", length 5
```

hipal31, 2024-10-18:
Other error logs:
```
I (541) cpu_start: Max chip rev:     v1.99
I (546) cpu_start: Chip rev:         v0.3
I (551) heap_init: Initializing. RAM available for dynamic allocation:
I (558) heap_init: At 3FCA93A0 len 00016C60 (91 KiB): DRAM
I (564) heap_init: At 3FCC0000 len 0001C710 (113 KiB): DRAM/RETENTION
I (571) heap_init: At 3FCDC710 len 00002950 (10 KiB): DRAM/RETENTION/STACK
I (579) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM
I (586) spi_flash: detected chip: generic
I (590) spi_flash: flash io: dio
I (594) sleep: Configure to isolate all GPIO pins in sleep state
I (600) sleep: Enable automatic switching of GPIO sleep configuration
I (607) esp_core_dump_uart: Init core dump to UART
I (613) coexist: coex firmware version: 77cd7f8
I (618) coexist: coexist rom version 9387209
I (623) app_start: Starting scheduler on CPU0
I (628) main_task: Started on CPU0
I (628) main_task: Calling app_main()
I (648) RESET: REASON : 9
I (648) RESET: COUNT : 0
E (658) REEVA_BLE: Read WATT failed, err ESP_ERR_NVS_NOT_FOUND
E (658) REEVA_BLE: Read OPERATE_WATT failed, err ESP_ERR_NVS_NOT_FOUND
E (658) REEVA_BLE: Read OP_MAX_PER failed, err ESP_ERR_NVS_NOT_FOUND
E (668) REEVA_BLE: Read SMOOTH_START failed, err ESP_ERR_NVS_NOT_FOUND
E (678) REEVA_BLE: Read MINIMUM_LEVEL failed, err ESP_ERR_NVS_NOT_FOUND
E (678) REEVA_BLE: Read DEFAULT_LEVEL failed, err ESP_ERR_NVS_NOT_FOUND
E (688) REEVA_BLE: Read DEFAULT_COLOR failed, err ESP_ERR_NVS_NOT_FOUND
E (698) REEVA_BLE: Read MIN_COLOR failed, err ESP_ERR_NVS_NOT_FOUND
E (698) REEVA_BLE: Read MAX_COLOR failed, err ESP_ERR_NVS_NOT_FOUND
I (708) REEVA_NVS: Open namespace done, name "reeva_mesh"
I (718) REEVA_NVS: Restore, key "LIGHT_DATA", length 5
I (718) REEVA_NVS: Restore, data: e8 03 00 00 02
I (728) LIGHT: Restored...
I (728) LIGHT: color_temperature = 0, brightness = 1000
I (738) REEVA_NVS: Store, key "LIGHT_DATA", length 5
I (738) REEVA_NVS: Store, data: e8 03 00 00 02
I (748) REEVA_BLE: Initializing...
I (748) BLE_INIT: BT controller compile version [9359a4d]
I (758) BLE_INIT: Bluetooth MAC: 84:f7:03:16:37:ae

I (768) phy_init: phy_version 1150,7c3c08f,Jan 24 2024,17:32:21
I (818) REEVA_NVS: Open namespace done, name "reeva_mesh"
I (818) REEVA_NVS: Restore, key "HEART_TIME", length 8
I (818) REEVA_NVS: Restore, data: 58 02 ff ff ff ff 00 00
I (828) REEVA_BLE: HEARTBEAT TIME INTERVAL = 600 seconds
I (838) REEVA_NVS: Restore, key "SCENE_STATE", length 16
I (838) REEVA_NVS: Restore, data: 32 00 00 00 3c 46 c9 3f db ad 00 00 00 00 00 00
I (848) REEVA_BLE: Restored SCENE_STATES
I (858) REEVA_NVS: Restore, key "SCENE_ADDR", length 400
I (858) REEVA_NVS: Restore, data: 69 ea 00 00 64 4c c9 3f db ad 00 00 4c 4c c9 3f
I (868) REEVA_NVS: Restore, data: 00 00 00 00 34 4c c9 3f 00 00 00 00 1c 4c c9 3f
I (878) REEVA_NVS: Restore, data: 00 00 00 00 04 4c c9 3f 00 00 00 00 ec 4b c9 3f
I (888) REEVA_NVS: Restore, data: 00 00 00 00 d4 4b c9 3f 00 00 00 00 bc 4b c9 3f
I (888) REEVA_NVS: Restore, data: 00 00 00 00 a4 4b c9 3f 00 00 00 00 8c 4b c9 3f
I (898) REEVA_NVS: Restore, data: 00 00 00 00 74 4b c9 3f 00 00 00 00 5c 4b c9 3f
I (908) REEVA_NVS: Restore, data: 00 00 00 00 44 4b c9 3f 00 00 00 00 2c 4b c9 3f
I (918) REEVA_NVS: Restore, data: 00 00 00 00 14 4b c9 3f 00 00 00 00 fc 4a c9 3f
I (928) REEVA_NVS: Restore, data: 00 00 00 00 e4 4a c9 3f 00 00 00 00 cc 4a c9 3f
I (938) REEVA_NVS: Restore, data: 00 00 00 00 b4 4a c9 3f 00 00 00 00 9c 4a c9 3f
I (938) REEVA_NVS: Restore, data: 00 00 00 00 84 4a c9 3f 00 00 00 00 6c 4a c9 3f
I (948) REEVA_NVS: Restore, data: 00 00 00 00 54 4a c9 3f 00 00 00 00 3c 4a c9 3f
I (958) REEVA_NVS: Restore, data: 00 00 00 00 24 4a c9 3f 00 00 00 00 0c 4a c9 3f
I (968) REEVA_NVS: Restore, data: 00 00 00 00 f4 49 c9 3f 00 00 00 00 dc 49 c9 3f
I (978) REEVA_NVS: Restore, data: 00 00 00 00 c4 49 c9 3f 00 00 00 00 ac 49 c9 3f
I (978) REEVA_NVS: Restore, data: 00 00 00 00 94 49 c9 3f 00 00 00 00 7c 49 c9 3f
I (988) REEVA_NVS: Restore, data: 00 00 00 00 64 49 c9 3f 00 00 00 00 4c 49 c9 3f
I (998) REEVA_NVS: Restore, data: 00 00 00 00 34 49 c9 3f 00 00 00 00 1c 49 c9 3f
I (1008) REEVA_NVS: Restore, data: 00 00 00 00 04 49 c9 3f 00 00 00 00 ec 48 c9 3f
I (1018) REEVA_NVS: Restore, data: 00 00 00 00 d4 48 c9 3f 00 00 00 00 bc 48 c9 3f
[... 436 lines cut ...]
I (603) RESET: REASON : 9
I (603) RESET: COUNT : 0
E (613) REEVA_BLE: Read WATT failed, err ESP_ERR_NVS_NOT_FOUND
E (613) REEVA_BLE: Read OPERATE_WATT failed, err ESP_ERR_NVS_NOT_FOUND
E (613) REEVA_BLE: Read OP_MAX_PER failed, err ESP_ERR_NVS_NOT_FOUND
E (623) REEVA_BLE: Read SMOOTH_START failed, err ESP_ERR_NVS_NOT_FOUND
E (633) REEVA_BLE: Read MINIMUM_LEVEL failed, err ESP_ERR_NVS_NOT_FOUND
E (633) REEVA_BLE: Read DEFAULT_LEVEL failed, err ESP_ERR_NVS_NOT_FOUND
E (643) REEVA_BLE: Read DEFAULT_COLOR failed, err ESP_ERR_NVS_NOT_FOUND
E (653) REEVA_BLE: Read MIN_COLOR failed, err ESP_ERR_NVS_NOT_FOUND
E (653) REEVA_BLE: Read MAX_COLOR failed, err ESP_ERR_NVS_NOT_FOUND
I (663) REEVA_NVS: Open namespace done, name "reeva_mesh"
I (673) REEVA_NVS: Restore, key "LIGHT_DATA", length 5
I (673) REEVA_NVS: Restore, data: e8 03 00 00 02
I (683) LIGHT: Restored...
I (683) LIGHT: color_temperature = 0, brightness = 1000
I (693) REEVA_NVS: Store, key "LIGHT_DATA", length 5
I (693) REEVA_NVS: Store, data: e8 03 00 00 02
I (703) REEVA_BLE: Initializing...
I (703) BLE_INIT: BT controller compile version [9359a4d]
I (713) BLE_INIT: Bluetooth MAC: 84:f7:03:16:37:ae

I (723) phy_init: phy_version 1150,7c3c08f,Jan 24 2024,17:32:21
I BOD: Brownout detector was triggered
```

forx157, 2024-10-21:
hi, @hipal31 
After you restored the scene data from the flash, how do you use this data to initialize the time_scene server model?

hipal31, 2024-11-07:
@forx157 Please find code snippet
```
static esp_ble_mesh_scene_register_t scenes[50] = {
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)},
    {.scene_value = NET_BUF_SIMPLE(10)}};

static esp_ble_mesh_scenes_state_t scene_state = {
    .scene_count = ARRAY_SIZE(scenes),
    .scenes = scenes,
};
ESP_BLE_MESH_MODEL_PUB_DEFINE(SCENE_SETUP_SRV_pub, 1 + 6, ROLE_NODE);
static esp_ble_mesh_scene_setup_srv_t scene_setup_server = {
    .rsp_ctrl.get_auto_rsp = ESP_BLE_MESH_SERVER_AUTO_RSP,
    .rsp_ctrl.set_auto_rsp = ESP_BLE_MESH_SERVER_AUTO_RSP,
    .state = &scene_state,
};

ESP_BLE_MESH_MODEL_PUB_DEFINE(SCENE_SRV_pub, 1 + 6, ROLE_NODE);
static esp_ble_mesh_scene_srv_t scene_server = {
    .rsp_ctrl.get_auto_rsp = ESP_BLE_MESH_SERVER_AUTO_RSP,
    .rsp_ctrl.set_auto_rsp = ESP_BLE_MESH_SERVER_AUTO_RSP,
    .state = &scene_state,
};

static esp_ble_mesh_model_t root_models[] = {
    ESP_BLE_MESH_MODEL_CFG_SRV(&config_server),
    ESP_BLE_MESH_MODEL_SCENE_SETUP_SRV(&SCENE_SETUP_SRV_pub, &scene_setup_server),
    ESP_BLE_MESH_MODEL_SCENE_SRV(&SCENE_SRV_pub, &scene_server),
};
```

forx157, 2025-03-31:
`scene_value` is a pointer, you should recover the content stored by the pointer, not the value of this pointer, and update the pointer when the content has been successfully recovered.

closed 2025-05-05 by hipal31