espressif/esp-idf #14750: ESP32-C5 BLUFI example connects to 5.0G WIFI Failed, 2.4GHz successfully.

williamleeag, 2024-10-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-dev-3602-ga97a7b0962

### Espressif SoC revision.

ESP32-C5 (revision v0.0)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C5-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

Successfully connect to 5G Network

### What is the actual behavior?

Not successfully connect to 5G Network

### Steps to reproduce.

1. Run newest version of BLUFI example
2. Run EspBluFi App 1.6.3

### Debug Logs.

```plain
ESP-ROM:esp32c5-20240329
Build:Mar 29 2024
rst:0x1 (POWERON),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPI mode:DIO, clock div:1
load:0x40855ab0,len:0x1664
load:0x4084c7a0,len:0xcdc
load:0x4084e9a0,len:0x2ecc
entry 0x4084c7aa
I (23) boot: ESP-IDF v5.4-dev-3602-ga97a7b0962 2nd stage bootloader
I (23) boot: compile time Oct 18 2024 18:14:38
I (24) boot: chip revision: v0.0
I (25) boot: efuse block revision: v0.0
I (29) boot.esp32c5: SPI Speed      : 80MHz
I (32) boot.esp32c5: SPI Mode       : DIO
I (36) boot.esp32c5: SPI Flash Size : 2MB
I (40) boot: Enabling RNG early entropy source...
I (44) boot: Partition Table:
I (47) boot: ## Label            Usage          Type ST Offset   Length
I (53) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (60) boot:  1 phy_init         RF data          01 01 0000f000 00001000
[... 189 lines cut ...]
W (17949) wifi:(vht)BSS bandwidth:160MHz
I (17949) wifi:(he)max.RxAMPDULenExponentExtension:3, max.RxAMPDUpre-EOFpaddingBytes:8388607(0x7fffff)
I (17959) wifi:(mac)omc_ul_mu_data_disable_rx:1
I (17959) wifi:(phy)ppe_thresholds_present:1, nominal_packet_padding:0
I (17969) wifi:(phy)dcm tx(constellation:0, nss:0), dcm rx(constellation:1, nss:0)
I (17979) wifi:(phy)rx_mcs_map:0xffaa, tx_mcs_map:0xffaa, stbc_tx:0, bfmer(su:1, mu:1), ldpc:1, max_rate:172
I (17989) wifi:(phy)nsts:3, ru_index_bitmap:0x5(242:1, 484:0, 996:1, 2*996:0)
I (17989) wifi:(phy)threshold_bits:36, nsts_num:3, ru_num:2, he_cap->len:39, ppe_threshold_len:17(6,11,4)
I (17999) wifi:(ppe)RU242, NSTS0, PPE16:7, PPE8:7, nominal_packet_padding:0
W (18009) wifi:(hal)co_hosted_bss:0, max_indicator:0, bitmask:0xff, mBSSIDsEnable:0
I (18019) wifi:(opr)len:7, TWT Required:0, VHT Opr Info Present:0, 6GHz Opr Info Present:0, Co-Hosted BSS:0(max_indicator:0), Basic MCS and NSS:0xfffc
I (18029) wifi:(opr)len:7, Default PE Duration:4, TXOP RTS Threshold:1023(0 us), ER-SU-Disable:0
I (18039) wifi:(opr)len:7, BSS Color:31, disabled:0, Partial BSS Color:0
I (18039) wifi:(spr)len:2, ctrl:0x3, PSR Disallowed:1, Non-SRG OBSS PD SR Disallowed:1
I (18049) wifi:(spr)len:2, ctrl:0x3, Non-SRG Offset Present:0, SRG Info Present:0
W (18059) wifi:(regdomain)len:42, country:CN, env:all, ngroup:13
I (18279) wifi:new:<48,0>, old:<6,0>, ap:<255,255>, sta:<48,0>, prof:1, snd_ch_cfg:0x13
I (18279) wifi:(connect)dot11_authmode:0x3, pairwise_cipher:0x3, group_cipher:0x3
I (18279) wifi:state: init -> auth (0xb0)
I (19289) wifi:state: auth -> init (0x200)
I (19299) wifi:Coexist: Wi-Fi connect fail, apply reconnect coex policy

I (19299) wifi:new:<48,0>, old:<48,0>, ap:<255,255>, sta:<48,0>, prof:1, snd_ch_cfg:0x0
I (19299) BLUFI_EXAMPLE: BLUFI WiFi starts reconnection
```

nishanth-radja, 2024-10-23:
@williamleeag Can we have a sniffer capture during the connection to analyze further.

williamleeag, 2024-10-24:
> @williamleeag Can we have a sniffer capture during the connection to analyze further.

I found that when I chose the 5Ghz wifi and then click start connect to that wifi, there was even no packet sent from esp, only when connecting to a 2.4Ghz wifi, there were packets sent, I am not that familiar with this kind of tools, can you tell me how to set the filter to obtain all possible packets sent from esp?
70	13.311667	Espressif_f9:67:74	Broadcast	ARP	42	Who has 10.0.0.155? (ARP Probe)
74	13.824215	Espressif_f9:67:74	Broadcast	ARP	42	Who has 10.0.0.155? (ARP Probe)
77	14.030204	Espressif_f9:67:74	Broadcast	ARP	42	Who has 10.0.0.155? (ARP Probe)
89	15.053146	Espressif_f9:67:74	Broadcast	ARP	42	ARP Announcement for 10.0.0.155
110	15.975769	Espressif_f9:67:74	Broadcast	ARP	42	ARP Announcement for 10.0.0.155
125	16.998580	Espressif_f9:67:74	Broadcast	ARP	42	ARP Announcement for 10.0.0.155
444	58.060287	Espressif_f9:67:74	Broadcast	ARP	42	ARP Announcement for 10.0.0.155
471	64.102164	Espressif_f9:67:74	Broadcast	ARP	42	Who has 10.0.0.155? (ARP Probe)
474	64.511442	Espressif_f9:67:74	Broadcast	ARP	42	Who has 10.0.0.155? (ARP Probe)
478	65.024644	Espressif_f9:67:74	Broadcast	ARP	42	Who has 10.0.0.155? (ARP Probe)
484	66.048910	Espressif_f9:67:74	Broadcast	ARP	42	ARP Announcement for 10.0.0.155
494	66.968852	Espressif_f9:67:74	Broadcast	ARP	42	ARP Announcement for 10.0.0.155
503	67.992921	Espressif_f9:67:74	Broadcast	ARP	42	ARP Announcement for 10.0.0.155
1032	118.065288	Espressif_f9:67:74	Broadcast	ARP	42	ARP Announcement for 10.0.0.155

nishanth-radja, 2024-10-24:
@williamleeag On the linux laptop with wireshark installed. pls follow these steps to get the capture.
1.Ifconfig <wireless-interface-name> down
2.Iwconfig <wireless-interface-name> mode monitor
3. Ifconfig <wireless-interface-name> up
4. killall wpa_supplicant
5. Iwconfig <wireless-interface-name> channel <AP-channel> , From the logs,it looks like it is in channel 48
6. wireshark

williamleeag, 2024-10-24:
> @williamleeag On the linux laptop with wireshark installed. pls follow these steps to get the capture. 1.Ifconfig down 2.Iwconfig mode monitor 3. Ifconfig up 4. killall wpa_supplicant 5. Iwconfig channel , From the logs,it looks like it is in channel 48 6. wireshark

Hi, I am using the windows version Wireshark and I thought that the mac of the same esp board should be the same?
eth.src == 60:55:f9:f9:67:74,I set this filter and there is no packet captured during connecting to 5Ghz WIFI and can only connect to a 2.4Ghz WIFI.

If this issue cannot be reproduced on your computer? Thanks.

Espressif-liuuuu, 2024-10-25:
Hi @williamleeag , we can reproduce this issue on our side. There are some issues if running Wi-Fi + BT / sleep with ESP32C5 ECO0 chips on Wi-Fi low channels. Everything shall be OK if you configure AP to 5G channel 149. Sorry for that, please try 5G high channels at first.

williamleeag, 2024-10-28:
> Hi @williamleeag , we can reproduce this issue on our side. There are some issues if running Wi-Fi + BT / sleep with ESP32C5 ECO0 chips on Wi-Fi low channels. Everything shall be OK if you configure AP to 5G channel 149. Sorry for that, please try 5G high channels at first.

Thanks. May I ask if it will be solved in a way of hardware or software?

Espressif-liuuuu, 2024-10-28:
> Thanks. May I ask if it will be solved in a way of hardware or software?

Hi, we will fix this issue in later hardware versions of chips.

williamleeag, 2025-01-07:
Hi, I used another version of IC which named MB... instead of MA.... and update the new sdk, release version of idf5.4.
Two questions:
1.If this issue solved in this version already?
2.If any changes in SPI since my code is not working when sending SPI messages to other devices, it works before, and I have checked the connection.

Espressif-liuuuu, 2025-01-08:
> Hi, I used another version of IC which named MBN8R4 instead of MAN8R4 and update the new SDK to release version of idf5.4. Two questions: 1.If this issue solved in this version already? 2.If any changes in SPI since my code is not working when sending SPI messages to other devices, it works before, and I have checked the connection.

Hi @williamleeag , thanks for support.  
The coexistence issue is resloved on this chip (C5 ECO1) now.

williamleeag, 2025-01-08:
> > Hi, I used another version of IC which named MBN8R4 instead of MAN8R4 and update the new SDK to release version of idf5.4. Two questions: 1.If this issue solved in this version already? 2.If any changes in SPI since my code is not working when sending SPI messages to other devices, it works before, and I have checked the connection.
> 
> Hi @williamleeag , thanks for support. The coexistence issue is resolved on this chip (C5 ECO1) now.

So, if there are any possible changes or setting that I need to modify that may cause SPI master interface failed?
[sdkconfig.zip](https://github.com/user-attachments/files/18344082/sdkconfig.zip)

wanckl, 2025-01-09:
@williamleeag Hello
C5 is now `preview` menas it not fully supported, can you try using `master` where the SPI should work.

williamleeag, 2025-01-09:
> @williamleeag Hello C5 is now `preview` menas it not fully supported, can you try using `master` where the SPI should work.

Hi, 
I tried the master version where the SPI still not working, I am wondering if there is any pin changed in the EC01 version? Thanks for your reply.

williamleeag, 2025-02-18:
> [@williamleeag](https://github.com/williamleeag) Which commit you tried on master the SPI still fail??
> 
> I tried just now it is OK with above code, what fail message you got?

should be v5.5-dev-1050-gb5ac4fbdf9, when I shift from the ECO0(v5.4-dev-3602-ga97a7b0962) to EC01(v5.5-dev-1050-gb5ac4fbdf9), all the code about spi are not working,here are the related code.
```
    spi_bus_config_t buscfg = {
        .mosi_io_num = SPI_MOSI_PIN,
        .miso_io_num = SPI_MISO_PIN,
        .sclk_io_num = SPI_SCLK_PIN,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
        .max_transfer_sz = 50,
    };

    spi_device_interface_config_t devcfg = {
        .clock_speed_hz = 40 * 1000 * 1000,
        .mode = 0,
        .spics_io_num = SPI_CS_PIN,
        .queue_size = 10,
        .cs_ena_posttrans = 5,
    };

    ESP_ERROR_CHECK(spi_bus_initialize(MOTOR_SPI_HOST, &buscfg, SPI_DMA_CH_AUTO));
    ESP_ERROR_CHECK(spi_bus_add_device(MOTOR_SPI_HOST, &devcfg, &spi_handle));
    ESP_LOGI(TAGSPI, "SPI initialized.");
static void spi_task(void *pvParameters)
{
    spi_request_t request;
    while (1) {
        if (xQueueReceive(spi_queue, &request, portMAX_DELAY)) {
            if (request.is_read) {
                spi_transaction_t t;
                memset(&t, 0, sizeof(t));
                t.length = 16;
                t.tx_data[0] = request.cmd;
                t.flags = SPI_TRANS_USE_TXDATA | SPI_TRANS_USE_RXDATA;

                ESP_ERROR_CHECK(spi_device_transmit(spi_handle, &t));

                if (request.read_value) {
                    *request.read_value = t.rx_data[1];
                }
            } else {
                spi_transaction_t t;
                memset(&t, 0, sizeof(t));
                t.length = 16;
                t.tx_data[0] = SET_CMD(request.cmd);
                t.tx_data[1] = request.data;
                t.flags = SPI_TRANS_USE_TXDATA;
                ESP_ERROR_CHECK(spi_device_transmit(spi_handle, &t));
            }
        }
    }
    vTaskDelete(NULL);
}
static esp_err_t queue_spi_cmd(uint8_t cmd, uint8_t data, bool is_read, uint8_t *read_value)
{
    spi_request_t request = {
        .cmd = cmd,
        .data = data,
        .is_read = is_read,
        .read_value = read_value
    };

    if (xQueueSend(spi_queue, &request, portMAX_DELAY) == pdPASS) {
        return ESP_OK;
    } else {
        return ESP_FAIL;
    }
}
static esp_err_t queue_send_spi_cmd(uint8_t cmd, uint8_t data)
{
    return queue_spi_cmd(cmd, data, false, NULL);
}
static esp_err_t queue_read_spi_cmd(uint8_t cmd, uint8_t *read_value)
{
    return queue_spi_cmd(cmd, 0, true, read_value);
}
static esp_err_t queue_send_and_verify_spi_cmd(uint8_t cmd, uint8_t data)
{
    uint8_t read_back_value;
    uint8_t retry_count = 0;

    while (retry_count < MAX_SPI_ATTEMPTS) {
        queue_send_spi_cmd(cmd, data);
        
        queue_read_spi_cmd(cmd, &read_back_value);

        if (read_back_value == data) {
            ESP_LOGI(TAGSPI, "Register 0x%02X set successfully to 0x%02X.", cmd, data);
            return ESP_OK;
        }

        ESP_LOGW(TAGSPI, "Verification failed for 0x%02X. Expected 0x%02X but got 0x%02X. Retrying...", cmd, data, read_back_value);
        retry_count++;
    }

    ESP_LOGE(TAGSPI, "Failed to set register 0x%02X after %d attempts.", cmd, MAX_SPI_ATTEMPTS);
    return ESP_FAIL;
}
```

wanckl, 2025-02-19:
@williamleeag 
Have tried your code on your commit with esp32c5 board, It works OK,     
And other test cases in `components\esp_driver_spi\test_apps\master` also works fine,    

So what fail message or apperance on your side? do you have logic analyser to check the signal?

williamleeag, 2025-02-26:
> Hi, I tried the master version where the SPI still not working, I am wondering if there is any pin changed in the EC01 version? Thanks for your reply.

Now we know the reason why it is not working, just because the pin of ECO1 version is different from the ECO0...........................................................................

wanckl, 2025-02-28:
@williamleeag 🔢 ,,,OK, congrat for finding the issue,,,

If no other question, you can close this issue, thanks!

closed 2025-02-28 by williamleeag