espressif/esp-idf #14751: Cannot build i2s_basic demos on Windows

xobs, 2024-10-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-dev-3602-ga97a7b0962

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

I should be able to build and run commands like `idf.py menuconfig`

### What is the actual behavior?

```
[9:17:41 pm] ~/esp/esp-idf/examples/peripherals/i2s/i2s_basic/i2s_std> idf.py build
Executing action: all (aliases: build)
Running cmake in directory C:\Users\Sean\esp\esp-idf\examples\peripherals\i2s\i2s_basic\i2s_std\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:\Espressif\python_env\idf5.4_py3.10_env\Scripts\python.exe -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 C:\Users\Sean\esp\esp-idf\examples\peripherals\i2s\i2s_basic\i2s_std"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.40.0.windows.1")
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 14.2.0
-- The CXX compiler identification is GNU 14.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Espressif/tools/xtensa-esp-elf/esp-14.2.0_20240906/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Espressif/tools/xtensa-esp-elf/esp-14.2.0_20240906/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Espressif/tools/xtensa-esp-elf/esp-14.2.0_20240906/xtensa-esp-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
CMake Error at build/component_requires.temp.cmake:4 (__component_set_property):
  Syntax error in cmake code at

    C:/Users/Sean/esp/esp-idf/examples/peripherals/i2s/i2s_basic/i2s_std/build/component_requires.temp.cmake:4

  when parsing string

    .;C:\Users\Sean\esp\esp-idf/examples/peripherals/i2s/common

  Invalid character escape '\U'.
Call Stack (most recent call first):
  C:/Users/Sean/esp/esp-idf/tools/cmake/component.cmake:258 (include)
  C:/Users/Sean/esp/esp-idf/tools/cmake/build.cmake:643 (__component_get_requirements)
  C:/Users/Sean/esp/esp-idf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1, output of the command is in the C:\Users\Sean\esp\esp-idf\examples\peripherals\i2s\i2s_basic\i2s_std\build\log\idf_py_stderr_output_65328 and C:\Users\Sean\esp\esp-idf\examples\peripherals\i2s\i2s_basic\i2s_std\build\log\idf_py_stdout_output_65328
[9:17:51 pm] ~/esp/esp-idf/examples/peripherals/i2s/i2s_basic/i2s_std>
```

### Steps to reproduce.

1. Try ro build examples/peripherals/i2s/i2s_basic/i2s_std on Windows

### More Information.

This is likely caused by importing `$ENV{IDF_PATH}` directly into `INCLUDE_DIRS`, which makes the variable as follows:

```
__component_set_property(___idf_main INCLUDE_DIRS ".;C:\Users\Sean\esp\esp-idf/examples/peripherals/i2s/common")
```

It's complaining about `C:\U`

dothanhdat1413, 2024-10-29:
This is due to the reference address in Window. I fixed this by replacing the $ENV{IDF_PATH} in main/CMakeLists.txt with the real address in my computer. It works but I think that this way is just a temporary solution to run the example once.

idf_component_register(SRCS "${srcs}"
                    PRIV_REQUIRES esp_driver_i2s esp_driver_gpio
                    INCLUDE_DIRS "." "$ENV{IDF_PATH}/examples/peripherals/i2s/common")

replace $ENV{IDF_PATH} with the address in computer

L-KAYA, 2024-12-10:
Hi @xobs, thanks for your feedback! Indeed, the absolute include path has slash & backslash issue on windows. It will be fixed by making the common dependencies an example common component.

cc @dothanhdat1413 , thanks for your answer!

closed 2024-12-20 by espressif-bot