espressif/esp-idf #14754: Build process tries to overwrite openocd stub binaries in the esp-idf installation

nomis, 2024-10-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-dev-3602-ga97a7b0962

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Building a project doesn't try to modify part of the esp-idf installation

### What is the actual behavior?

Build process tries to overwrite `/opt/esp/idf/components/esp_system/openocd_stub_bins/...` that the user has no access to

### Steps to reproduce.

Using the `espressif/idf:latest` docker image, build https://github.com/nomis/candle-dribbler/releases/tag/0.7.3 with `idf.py` for either ESP32-C6 or ESP32-H2 as a **non-root user**.

The build will fail when it tries to overwrite part of the esp-idf installation at `/opt/esp/idf/components/esp_system/openocd_stub_bins/stub_flash_idf_image.h`

### Build or installation Logs.

```plain
[2024-10-17T10:11:20.514Z] Activating ESP-IDF 5.4
[2024-10-17T10:11:20.514Z] * Checking python version ... 3.12.3
[2024-10-17T10:11:20.886Z] * Checking python dependencies ... OK
[2024-10-17T10:11:20.886Z] * Deactivating the current ESP-IDF environment (if any) ... OK
[2024-10-17T10:11:20.886Z] * Establishing a new ESP-IDF environment ... OK
[2024-10-17T10:11:21.259Z] * Identifying shell ... sh
[2024-10-17T10:11:21.259Z] * Detecting outdated tools in system ... OK - no outdated tools found
[2024-10-17T10:11:21.259Z] 
[2024-10-17T10:11:21.259Z] Done! You can now compile ESP-IDF projects.
[2024-10-17T10:11:21.259Z] Go to the project directory and run:
[2024-10-17T10:11:21.259Z] 
[2024-10-17T10:11:21.259Z]   idf.py build
[2024-10-17T10:11:21.259Z] + idf.py --version
[2024-10-17T10:11:23.746Z] ESP-IDF v5.4-dev-3602-ga97a7b0962
[2024-10-17T10:11:23.746Z] + idf.py set-target esp32c6
[2024-10-17T10:11:24.547Z] Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
[2024-10-17T10:11:24.547Z] Executing action: fullclean
[2024-10-17T10:11:24.547Z] Build directory '/home/jenkins/agent/workspace/candle-dribbler_main/build' not found. Nothing to clean.
[2024-10-17T10:11:24.547Z] Executing action: set-target
[2024-10-17T10:11:24.547Z] Set Target to: esp32c6, new sdkconfig will be created.
[2024-10-17T10:11:24.547Z] Running cmake in directory /home/jenkins/agent/workspace/candle-dribbler_main/build
[2024-10-17T10:11:24.547Z] Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/opt/esp/python_env/idf5.4_py3.12_env/bin/python -DESP_PLATFORM=1 -DIDF_TARGET=esp32c6 -DCCACHE_ENABLE=1 /home/jenkins/agent/workspace/candle-dribbler_main"...
[2024-10-17T10:11:24.547Z] -- Found Git: /usr/bin/git (found version "2.43.0")
[2024-10-17T10:11:26.627Z] -- ccache will be used for faster recompilation
[2024-10-17T10:11:26.627Z] -- The C compiler identification is GNU 14.2.0
[2024-10-17T10:11:26.627Z] -- The CXX compiler identification is GNU 14.2.0
[2024-10-17T10:11:26.627Z] -- The ASM compiler identification is GNU
[2024-10-17T10:11:26.627Z] -- Found assembler: /opt/esp/tools/riscv32-esp-elf/esp-14.2.0_20240906/riscv32-esp-elf/bin/riscv32-esp-elf-gcc
[2024-10-17T10:11:27.029Z] -- Detecting C compiler ABI info
[2024-10-17T10:11:27.029Z] -- Detecting C compiler ABI info - done
[2024-10-17T10:11:27.029Z] -- Check for working C compiler: /opt/esp/tools/riscv32-esp-elf/esp-14.2.0_20240906/riscv32-esp-elf/bin/riscv32-esp-elf-gcc - skipped
[... 69 lines cut ...]
[2024-10-17T10:11:36.528Z] [7/1058] Generating ../../ota_data_initial.bin
[2024-10-17T10:11:36.528Z] [8/1058] Generating ../../partition_table/partition-table.bin
[2024-10-17T10:11:36.528Z] Partition table binary generated. Contents:
[2024-10-17T10:11:36.528Z] *******************************************************************************
[2024-10-17T10:11:36.528Z] # ESP-IDF Partition Table
[2024-10-17T10:11:36.528Z] # Name, Type, SubType, Offset, Size, Flags
[2024-10-17T10:11:36.528Z] phy_init,data,phy,0x9000,4K,
[2024-10-17T10:11:36.528Z] zb_fct,data,fat,0xa000,4K,
[2024-10-17T10:11:36.528Z] otadata,data,ota,0xc000,8K,
[2024-10-17T10:11:36.528Z] nvs,data,nvs,0xe000,48K,
[2024-10-17T10:11:36.528Z] zb_storage,data,fat,0x1a000,152K,
[2024-10-17T10:11:36.528Z] coredump,data,coredump,0x40000,256K,
[2024-10-17T10:11:36.528Z] spiffs,data,spiffs,0x80000,1536K,
[2024-10-17T10:11:36.528Z] factory,app,factory,0x200000,2M,
[2024-10-17T10:11:36.528Z] app0,app,ota_0,0x400000,2M,
[2024-10-17T10:11:36.528Z] app1,app,ota_1,0x600000,2M,
[2024-10-17T10:11:36.528Z] *******************************************************************************
[2024-10-17T10:11:36.528Z] ninja: build stopped: subcommand failed.
[2024-10-17T10:11:36.927Z] ninja failed with exit code 1, output of the command is in the /home/jenkins/agent/workspace/candle-dribbler_main/build/log/idf_py_stderr_output_527 and /home/jenkins/agent/workspace/candle-dribbler_main/build/log/idf_py_stdout_output_527
[2024-10-17T10:11:36.927Z] + exit 1
```

### More Information.

The last successful build was with v5.4-dev-3489-g6e5a178b31 and it looks like f4eec4da48502ce725d6e74f5b5d9c8237aa7abf is the change that broke it.

igrr, 2024-10-19:
We have merged the fix for this yesterday, it should appear on GitHub upon next sync. Sorry for the inconvenience.

erhankur, 2024-12-16:
Fix is merged.

closed 2024-12-16 by erhankur