espressif/esp-idf #14757: components/esp_rom/patches/esp_rom_tlsf.c:24:10: fatal error: tlsf_control_functions.h: No such file or directory

SB211, 2024-10-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-dev-3433-g43b2074124

### Espressif SoC revision.

ESP32-C6

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6

### Power Supply used.

USB

### What is the expected behavior?

Compile hello_world from examples for target esp32-c6.

### What is the actual behavior?

[...]
Executing "ninja all"...
[4/1213] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[238/1213] Building C object esp-idf/esp_rom/CMakeFiles/__idf_esp_rom.dir/patches/esp_rom_tlsf.c.obj
FAILED: esp-idf/esp_rom/CMakeFiles/__idf_esp_rom.dir/patches/esp_rom_tlsf.c.obj 
/usr/local/riscv32-esp-elf/bin/riscv32-esp-elf-gcc ....
[...]
esp-idf/components/esp_rom/patches/esp_rom_tlsf.c:24:10: fatal error: tlsf_control_functions.h: No such file or directory
   24 | #include "tlsf_control_functions.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

Searching the folder were the ESP-IDF v5.4 resides for "tlsf_control_functions.h" does not reveal any such file.

### Steps to reproduce.

1. Step prepare hello_world from examples of IDF v5.4
2. Step idf.py set-target esp32c6
3. Step idf.py build
...

### Debug Logs.

```plain
no
```

### More Information.

no

sudeep-mohanty, 2024-10-21:
Hi @SB211, 
I do not see this issue at my end . Would you mind sharing your sdkconfig file as well?

sudeep-mohanty, 2024-10-23:
Closing the ticket as there does not seem to be any issues with ESP-IDF. Please feel free to re-open this issue or open a new one if you need further support. Thank you!

closed 2024-10-23 by sudeep-mohanty