espressif/esp-idf #14758: idf.py menuconfig on Ubuntu (linux) don't use full menu options (on windows machine run fine) for ESP32-S3 Chip

DrStone71, 2024-10-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

i have esp32-s3 and when use idf.py menuconfig on ubuntu system, program don't show full option (see image 1).

I have make new full install in this day of ESP-IDF 5.3.1 but with same issue.

When i make new full install on windows machine with same environment and ESP-IDF 5.3, idf.py menuconfig show full and correct options.

I try to import form window into ubuntu sdkconfig sdkconfig.ci and sdkconfig.old correctly configured form windows , but menuconfig don't show same parameter and reset parameter a standard configuration.

i try on all system same hello_world default Espressif esp32-s3 program, for make same environment on all system.

i in
[image]
[image]
sert screenshot of menuconfig with same esp32-s3 module correctly connect.

any idea ?
Best Regards

DrStone71, 2024-10-20:
i find solution, using sdiff -s of all file home of project:

into file HOME_OF_Project/CMakeLists.txt i remove:

set(COMPONENTS main) and idf.py menuconfig show all option.

########## PRIOR ##################

# The following lines of boilerplate have to be in your project's CMakeLists
# in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.16)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
set(COMPONENTS main)

project(esp32-32_i2s_Rec_e_Play_example)

############################

############### AFTER ###############

# The following lines of boilerplate have to be in your project's CMakeLists
# in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.16)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)

project(esp32-32_i2s_Rec_e_Play_example)

##############################

any idea ?

for me is possible close ISSUE.

Best Regards.

dobairoland, 2024-10-21:
@DrStone71 Have you run `idf.py set-target esp32s3` before running `idf.py menuconfig`?

dobairoland, 2024-10-21:
@fhrbata PTAL at the cmake part, especially about removing `set(COMPONENTS main)`.

DrStone71, 2024-10-21:
> @DrStone71 Have you run `idf.py set-target esp32s3` before running `idf.py menuconfig`?

yes, and I also ran the command idf.py reconfigured

DrStone71, 2024-10-21:
> @fhrbata PTAL at the cmake part, especially about removing `set(COMPONENTS main)`.

I honestly don't know how that configuration line ended up there. Maybe inherited from the original program. Night n there should be a warning when there are similar errors that block the basic functionality of the configurator?

Best Regards

fhrbata, 2024-10-21:
Hello @DrStone71,

 as you've already identified, the likely culprit is the `set(COMPONENTS main)` line in the project's `CMakeLists.txt`. Could it be that you have the `set(COMPONENTS main)` line added to your project only in the Ubuntu environment? When this is set, the build system considers only commonly required components and components explicitly added as requirements to the main component, including their direct and indirect dependencies. As a result Kconfig files for these components only are displayed in the menuconfig. The description of the `COMPONENTS` variable is described  [here](https://docs.espressif.com/projects/esp-idf/en/v5.3.1/esp32s3/api-guides/build-system.html#optional-project-variables).

Thank you

fhrbata, 2024-10-21:
> I honestly don't know how that configuration line ended up there. Maybe inherited from the original program. Night n there should be a warning when there are similar errors that block the basic [...]

Hello @DrStone71,

If I understand correctly, you were able to solve the issue, so maybe this can be closed. By setting the `COMPONENTS` cMake variable, the build system only includes components that are accessible from the main component dependencies and common components, as explained earlier. This is the current build system behavior. To have all components and their configurations available for your project, the `COMPONENTS` variable should not be set.

Thank you!

DrStone71, 2024-10-21:
Thanks for the valuable support, I read the article with interest.

I close the report.

closed 2024-10-21 by dobairoland