espressif/esp-idf #14759: ESP-IDF v5.3.0 Why i2c crashing continuously in interrupt ?

nesetaydinn, 2024-10-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: BREAK instr 
Core  0 register dump:
PC      : 0x4037b2ac  PS      : 0x00000016  A0      : 0x40374306  A1      : 0x3fc9dc90  
0x4037b2ac: _xt_kernel_exc at /home/nesh/esp/esp-idf/components/xtensa/xtensa_vectors.S:599
0x40374306: _KernelExceptionVector at ??:?

A2      : 0x00000000  A3      : 0x00000003  A4      : 0x00000003  A5      : 0x3fca8540  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x80383828  A9      : 0x3fc9dc60  
A10     : 0x00000001  A11     : 0x00000001  A12     : 0x3fca8540  A13     : 0x3fca8520  
A14     : 0x00060023  A15     : 0x00000003  SAR     : 0x0000001e  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  
0x40056f5c: memcpy in ROM
0x40056f72: memcpy in ROM

Backtrace: 0x4037b2a9:0x3fc9dc90 0x40374303:0x3fc9dca0 0x40384c99:0x3fc9dcc0 0x40382ff8:0x3fc9dce0 0x4037e276:0x3fc9dd00 0x403765f5:0x3fc9dd40 0x4037b6ed:0x3fc9dd60 0x4037fc7f:0x3fca8500 0x4202a1a2:0x3fca8520 0x4038440d:0x3fca8540 0x4038352e:0x3fca8560
0x4037b2a9: _xt_debugexception at /home/nesh/esp/esp-idf/components/xtensa/xtensa_vectors.S:483
0x40374303: _KernelExceptionVector at /home/nesh/esp/esp-idf/components/xtensa/xtensa_vectors.S:589
0x40384c99: xTaskRemoveFromEventList at /home/nesh/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3961
0x40382ff8: xQueueGenericSendFromISR at /home/nesh/esp/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1264
0x4037e276: i2c_isr_handler_default at /home/nesh/esp/esp-idf/components/driver/i2c/i2c.c:602
0x403765f5: shared_intr_isr at /home/nesh/esp/esp-idf/components/esp_hw_support/intr_alloc.c:445
0x4037b6ed: _xt_lowint1 at /home/nesh/esp/esp-idf/components/xtensa/xtensa_vectors.S:1240
0x4037fc7f: xt_utils_wait_for_intr at /home/nesh/esp/esp-idf/components/xtensa/include/xt_utils.h:82
 (inlined by) esp_cpu_wait_for_intr at /home/nesh/esp/esp-idf/components/esp_hw_support/cpu.c:55
0x4202a1a2: esp_vApplicationIdleHook at /home/nesh/esp/esp-idf/components/esp_system/freertos_hooks.c:58
0x4038440d: prvIdleTask at /home/nesh/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4344 (discriminator 1)
0x4038352e: vPortTaskWrapper at /home/nesh/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

My i2c init;
```c
    i2c_bus_port_h = I2C_MASTER_NUM;
    i2c_config_t conf;
    memset(&conf, 0, sizeof(conf));
    conf.mode = I2C_MODE_MASTER;
    conf.sda_io_num = PIN_I2C_MASTER_SDA;
    conf.scl_io_num = PIN_I2C_MASTER_SCL;
    conf.sda_pullup_en = GPIO_PULLUP_DISABLE;
    conf.scl_pullup_en = GPIO_PULLUP_DISABLE;
    conf.master.clk_speed = I2C_MASTER_FREQ_HZ;

    if (ESP_OK != i2c_param_config(i2c_bus_port_h, &conf))
    {
        while (1);
    }

    if (ESP_OK != i2c_driver_install(i2c_bus_port_h, conf.mode, 0, 0, ESP_INTR_FLAG_SHARED | ESP_INTR_FLAG_IRAM))
    {
        while (1);
    }
```

And sample using i2c example;
```c
    i2c_cmd_handle_t cmd = i2c_cmd_link_create();
    i2c_master_start(cmd);
    i2c_master_write_byte(cmd, (self->prop.dev_addr << 1) | I2C_MASTER_WRITE, true);
    i2c_master_write_byte(cmd, reg, true);
    i2c_master_start(cmd);
    i2c_master_write_byte(cmd, (self->prop.dev_addr << 1) | I2C_MASTER_READ, true);
    i2c_master_read_byte(cmd, res, I2C_MASTER_LAST_NACK);
    i2c_master_stop(cmd);
    bool state = (ESP_OK == i2c_master_cmd_begin(*self->bus_port, cmd, portMAX_DELAY));
    i2c_cmd_link_delete(cmd);
    return state;

```

mythbuster5, 2024-10-22:
Cannot reproduce so far. Please provide a simplest reproducable code if possible.

mythbuster5, 2025-02-17:
Seems there is no far more update, I will close this issue, if there is any update, feel free to reopen. Thanks!

closed 2025-02-17 by mythbuster5