espressif/esp-idf #14763: My ESP-IDF v5.2.2 is not the same as the github v5.2.2 tagged version (IEP-1267)

sfeutrier, 2024-10-22:
**Describe the bug**
I'm trying for hours to make the wifi provisioning sample to work, without any luck!
After investigation of the source code, I found out that everything related to "esp_wifi_start" (starting Wi-Fi in Station Mode) is commented out.
This is not the case here: https://github.com/espressif/esp-idf/blob/v5.2.2/components/wifi_provisioning/src/manager.c (line 1556) while in my folder "C:\Espressif\frameworks\esp-idf-v5.2.2\components\wifi_provisioning\src\manager.c", this is the line 1658... How ?
What is the correct way to solve my issue?

**To Reproduce**
Steps to reproduce the behavior:
I guess install Espressif IDE with esp-idf v5.2.2

**Expected behavior**
Folder files and tagged version files should be identical

**Screenshots**
What I have in my local version of esp-idf
[image]
What I should have when looking at the tag v5.2.2 in Github
[image]

sfeutrier, 2024-10-22:
I have a project that uses WiFi Mesh lite.
I have updated my Espressif-IDE to v3.1.0.
Now, every time I build, the two files are regenerated and thus WiFi does not work. Why the SDK can be modified by itself like this when I compile?
Also, I tried to update to v5.2.3 and it did not change anything... What a pain.

AndriiFilippov, 2024-10-22:
> Hello,
> 
> There is a workaround for this: just go in in the tab "Espressif" and open the "ESP-IDF manager" just install the esp-df version 5.2.2 at another place and select it. Note: there is an issue that Eclipse Plugging does not succeed to access to the default path to GIT on my side (in the WindowsApp folder), I have just selected another instance at the C root directory
> 
> However this is still a real problem as it makes you used a "bugged" version of the ESP-IDF without letting you know at all!

Hello, 

 WindowsApps folder is not the default path for Git when installed via the official Git installer or other typical methods.
Usually it is something like this: 
`C:\Program Files\Git\cmd\git.exe`
Ofc, you still can use Microsoft Store to install Git, however, it might lead to potential permission issues, as the WindowsApps folder is protected and access is restricted by OS.

AndriiFilippov, 2024-10-22:
> My project uses WiFi Mesh lite. I have recently updated my Espressif-IDE to v3.1.0. Now, every time I build, the two files are regenerated and thus WiFi does not work. Why the SDK can be modified by itself like this when I compile? Also, I tried to update to v5.2.3 and it did not change anything... What a pain.

Please, provide more details:   
Which two files are being regenerated that cause Wi-Fi to stop working? Does it just regenerate, or is it reset to default?
Does this happen with a specific example, or do you encounter it with any of them?

AndriiFilippov, 2024-10-22:
@sfeutrier hi, 
Thank you for the quick response. 
How do download ESP-IDF? 
Using Espressif IDE "ESP-IDF: Manager" option 
[image]
or Offline Installer 
[image]
or manually 
```
git clone -b v5.2.3 --recursive https://github.com/espressif/esp-idf.git esp-idf-v5.2.3
cd esp-idf-v5.2.3/
```

sfeutrier, 2024-10-23:
[image]

Your welcome!
Any of these it seems xD. I used the "Online installer" if I believe my Download folder and it installed the esp-idf by itself (I don't remember manually choosing anything during the installation to be honest).

sfeutrier, 2024-10-23:
Also, I installed the ESP-IDF v5.2.3 thanks to the ESP-IDF Manager and it has the same issue when I built.

kolipakakondal, 2024-10-23:
Hi @sfeutrier 

Unfortunately, I am unable to reproduce this problem to investigate further.

I have verified this with **ESP-IDF v5.2.2** and **v5.2.3** using the Espressif-IDE, running the **wifi_prov_mgr** example. I am able to provision the device without any issues, and I don’t see any code changes in the files mentioned in the respective IDF folders.

I’m not sure what might have caused this problem in your case.

[image]

sfeutrier, 2024-10-23:
Hello @kolipakakondal,

Thank you for testing!
I confirm that 2 of my colleagues have tested and even with the modified files, WiFi now works (maybe because we use WiFi Mesh lite, I don't know)
That's very strange you cannot reproduce the issue. We are 3 facing the same issue. Maybe because we downloaded the installer at the same moment. I don't know.
Is this supposed to happen that building a project modify files in the ESP-IDF sdk? It seems very unsecure.
Could it be an issue with our project or the Espressif-IDE? (I have updated the Espressif-IDE though and it did not change anything), maybe I need to reinstall it completly).

sfeutrier, 2024-10-24:
Hello,

I did the same test with "wifi_prov_mgr" and indeed, it does not modify any files in my esp-idf SDK.
BUT, if I import in my Espressif-IDE the sample "mesh_local_control", **it does modify my files! That means it is related to esp-mesh-lite!!**

igrr, 2024-10-24:
This is likely related to this line in esp-mesh-lite:

https://github.com/espressif/esp-mesh-lite/blob/5d22ed44dd85b71de5e96e8bd9b8fdda288e545f/components/mesh_lite/CMakeLists.txt#L21-L23

@tswen could you please take a look? Probably there is a better way to do this than applying a patch — this command may fail if esp-idf is in a read-only directory. Plus this messes with other projects which may use wifi_provisioning.

sfeutrier, 2024-10-24:
Hello @igrr,
Thanks for pointing that out! :)
I agree that it is a mess when you develop other projects that does not use mesh but provisionning (or just WiFi maybe). Plus it means if we are several on the same project, I cannot relate on the generated code because everyone can have a "different" SDK.

Also, note that since I install Espressif-IDE v3.1.0, more filesare now modified x). You can see below thelist.
[image]

tswen, 2024-10-25:
> This is likely related to this line in esp-mesh-lite:
>
> https://github.com/espressif/esp-mesh-lite/blob/5d22ed44dd85b71de5e96e8bd9b8fdda288e545f/components/mesh_lite/CMakeLists.txt#L21-L23
>
> @tswen [...]

Sure, we will reassess a better approach, and any new developments will be communicated here.

sfeutrier, 2024-10-25:
@tswen I guess it will take time to make this change. During this, I cannot guarantee that between all my colleagues and across projects, we have the same binary file built because the SDK itself is different...
Do you have any idea of deadline?

tswen, 2024-11-01:
@sfeutrier We plan to integrate the `wifi_provisioning` component into the `mesh-lite` component, maintaining content modified due to `mesh-lite` within it. Since I am currently busy with other tasks, this will take some time, and we expect to merge it into the `esp-mesh-lite` repository around mid-November.