espressif/esp-idf #14764: tusb_serial_device not working

LeonBirkel, 2024-10-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

ESP32-s3 (QFN56) (revision v0.2)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

 ESP32 S3 DevKitC 1 N16R8 ESP32 S3 WROOM1 N16R8 Development Board

### Power Supply used.

USB

### What is the expected behavior?

After importing the example, builing and flashing. I expect the pytest to run. 

### What is the actual behavior?

Pytest is failing (Debug Logs) 

### Steps to reproduce.

1. Import example
2. build and flash
3. run pytest

### Debug Logs.

```plain
PS C:\BA\bastion\hardware-trust-anchor\board-v5> pytest ./pytest_usb_device_serial.py
============================================================ test session starts ============================================================
platform win32 -- Python 3.13.0, pytest-8.3.3, pluggy-1.5.0
rootdir: C:\BA\bastion\hardware-trust-anchor\board-v5
plugins: embedded-1.11.6
collected 1 item

pytest_usb_device_serial.py F                                                                                                          [100%]

================================================================= FAILURES ==================================================================
______________________________________________________ test_usb_device_serial_example _______________________________________________________

self = <pytest_embedded.dut.Dut object at 0x0000022A6F335FD0>, pattern = 'USB initialization DONE', expect_all = False, not_matching = ()
args = (), kwargs = {}, patterns = ['USB initialization DONE'], res = []
debug_str = 'Not found "USB initialization DONE"\nBytes in current buffer (color code eliminated): \nPlease check the full log her...ers\\leonb\\AppData\\Local\\Temp\\pytest-embedded\\2024-10-22_09-27-25-840324\\test_usb_device_serial_example\\dut.log'

    @functools.wraps(func)
    def wrapper(
        self, pattern, *args, expect_all: bool = False, not_matching: List[Union[str, re.Pattern]] = (), **kwargs
[... 313 lines cut ...]
│bMaxPacketSize0    │ 64          │
├───────────────────┼─────────────┤
│idVendor           │ 0x303a      │
├───────────────────┼─────────────┤
│idProduct          │ 0x4001      │
├───────────────────┼─────────────┤
│bcdDevice          │ 0x100       │
├───────────────────┼─────────────┤
│iManufacturer      │ 0x1         │
├───────────────────┼─────────────┤
│iProduct           │ 0x2         │
├───────────────────┼─────────────┤
│iSerialNumber      │ 0x3         │
├───────────────────┼─────────────┤
│bNumConfigurations │ 0x1         │
└───────────────────┴─────────────┘
I (514) TinyUSB: TinyUSB Driver installed
I (514) example: USB initialization DONE
I (824) example: Line state changed on channel 0: DTR:0, RTS:0
I (2554) example: Line state changed on channel 0: DTR:1, RTS:1
I (12164) example: GGGGG 
I (12164) example: Data from channel 0:
I (12164) example: 0x3fc98b50   74 65 73 74                                       |test|
I (12164) example: TEST 
```

roma-jam, 2024-10-22:
Hi @LeonBirkel, 

could you provide all the commands, how do you configure `esp-idf` (install & export), build and flash example and run pytest. 

Thanks.

LeonBirkel, 2024-10-22:
Thank you for the fast reply.
Pytest: PS C:\BA\bastion\hardware-trust-anchor\board-v5> pytest ./pytest_usb_device_serial.py

Install & import is handled by the VsCode extension. I select Express installation.

For building and flashing I use the VScode UI, it prints this command into my console:
C:\Users\leonb\.espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe C:\Users\leonb\esp\v5.3.1\esp-idf\components\esptool_py\esptool\esptool.py -p COM9 -b 460800 --before default_reset --after hard_reset --chip esp32s3 write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 board-v5.bin 0x8000 partition_table/partition-table.bin 

Is this the information you need @roma-jam or did I misunterstand something?

peter-marcisovsky, 2024-10-23:
Hi @LeonBirkel Can you run any other pytest, even from a different example? 

I can see you are running the pytest outside of the esp-idf, could you try to run a pytest from within the esp-idf? 
It would be something like this:

```
peter@peter ➜  ~/esp/esp-idf/examples/peripherals/usb/device/tusb_serial_device git:(master) ✗ pytest --target esp32s3
==================================================================================== test session starts =====================================================================================
platform linux -- Python 3.10.12, pytest-8.3.3, pluggy-1.5.0 -- /home/peter/.espressif/python_env/idf5.4_py3.10_env/bin/python
cachedir: .pytest_cache
rootdir: /home/peter/esp/esp-idf
configfile: pytest.ini
plugins: ignore-test-results-0.2.2, rerunfailures-14.0, timeout-2.3.1, cpp-2.6.0, embedded-1.11.6
collected 1 item 
```

The pytest searches for a configuration file called `pytest.ini`, preset in the root dir of the esp-idf
In your log, the `pytest.ini` is missing, as well as plugins for pytest. I am wondering, if that could be an issue.

LeonBirkel, 2024-10-23:
I am on a different machine and OS now (ubuntu), and I get the a similar error. 

If I try to run pytest in the idf folder, I get following error: 
```
leon@LeonsNotebookMInt:~/esp/v5.3.1/esp-idf$ pytest --target esp32s3
ImportError while loading conftest '/home/leon/esp/v5.3.1/esp-idf/conftest.py'.
conftest.py:30: in <module>
    import common_test_methods  # noqa: F401
tools/ci/python_packages/common_test_methods.py:13: in <module>
    os.path.join(os.environ['IDF_PATH'], 'EnvConfig.yml'),
/usr/lib/python3.10/os.py:680: in __getitem__
    raise KeyError(key) from None
E   KeyError: 'IDF_PATH'
```

I used the VS Code Extension installer, might this be the reason? 
However, if I just call `pytest` in my VSCode directory I get following errors, which is a bit confusing since I did install the pytest dependencies with : $ install.sh --enable-pytest

```
leon@LeonsNotebookMInt:~/uni/WS24/BA/bastion/hardware-trust-anchor/v15$ pytest
======================================= test session starts =======================================
platform linux -- Python 3.10.12, pytest-7.0.1, pluggy-1.2.0
rootdir: /home/leon/uni/WS24/BA/bastion/hardware-trust-anchor/v15
plugins: timeout-2.1.0, embedded-1.11.6, xdist-2.5.0, xonsh-0.11.0, forked-1.4.0
collected 503 items / 47 errors / 456 selected                                                    

============================================= ERRORS ==============================================
___________ ERROR collecting components/components/fatfs/test_fatfsgen/test_fatfsgen.py ___________
ImportError while importing test module '/home/leon/uni/WS24/BA/bastion/hardware-trust-anchor/v15/components/components/fatfs/test_fatfsgen/test_fatfsgen.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
/usr/lib/python3.10/importlib/__init__.py:126: in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
components/components/fatfs/test_fatfsgen/test_fatfsgen.py:11: in <module>
    from test_utils import CFG, fill_sector, generate_test_dir_1, generate_test_dir_2
components/components/fatfs/test_fatfsgen/test_utils.py:8: in <module>
    import fatfsgen  # noqa E402  # pylint: disable=C0413
components/components/fatfs/fatfsgen.py:9: in <module>
    from fatfs_utils.boot_sector import BootSector
components/components/fatfs/fatfs_utils/boot_sector.py:6: in <module>
    from construct import Bytes, Const, Int8ul, Int16ul, Int32ul, PaddedString, Padding, Struct, core
E   ModuleNotFoundError: No module named 'construct'
__________ ERROR collecting components/components/fatfs/test_fatfsgen/test_fatfsparse.py __________
ImportError while importing test module '/home/leon/uni/WS24/BA/bastion/hardware-trust-anchor/v15/components/components/fatfs/test_fatfsgen/test_fatfsparse.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
/usr/lib/python3.10/importlib/__init__.py:126: in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
components/components/fatfs/test_fatfsgen/test_fatfsparse.py:11: in <module>
    from test_utils import compare_folders, fill_sector, generate_local_folder_structure, generate_test_dir_2
components/components/fatfs/test_fatfsgen/test_utils.py:8: in <module>
    import fatfsgen  # noqa E402  # pylint: disable=C0413
components/components/fatfs/fatfsgen.py:9: in <module>
    from fatfs_utils.boot_sector import BootSector
components/components/fatfs/fatfs_utils/boot_sector.py:6: in <module>
    from construct import Bytes, Const, Int8ul, Int16ul, Int32ul, PaddedString, Padding, Struct, core
E   ModuleNotFoundError: No module named 'construct'
____________ ERROR collecting components/components/fatfs/test_fatfsgen/test_utils.py _____________
ImportError while importing test module '/home/leon/uni/WS24/BA/bastion/hardware-trust-anchor/v15/components/components/fatfs/test_fatfsgen/test_utils.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
/usr/lib/python3.10/importlib/__init__.py:126: in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
components/components/fatfs/test_fatfsgen/test_utils.py:8: in <module>
    import fatfsgen  # noqa E402  # pylint: disable=C0413
components/components/fatfs/fatfsgen.py:9: in <module>
    from fatfs_utils.boot_sector import BootSector
components/components/fatfs/fatfs_utils/boot_sector.py:6: in <module>
    from construct import Bytes, Const, Int8ul, Int16ul, Int32ul, PaddedString, Padding, Struct, core
E   ModuleNotFoundError: No module named 'construct'
[... 599 lines cut ...]
ERROR components/components/openthread/openthread/tests/scripts/thread-cert/border_router/test_end_device_udp_reachability.py
ERROR components/components/openthread/openthread/tests/scripts/thread-cert/border_router/test_firewall.py
ERROR components/components/openthread/openthread/tests/scripts/thread-cert/border_router/test_manual_maddress.py
ERROR components/components/openthread/openthread/tests/scripts/thread-cert/border_router/test_manual_omr_prefix.py
ERROR components/components/openthread/openthread/tests/scripts/thread-cert/border_router/test_radvd_coexist.py
ERROR components/components/openthread/openthread/tests/scripts/thread-cert/border_router/test_trel_connectivity.py
ERROR components/components/openthread/openthread/tests/scripts/thread-cert/border_router/nat64/test_multi_border_routers.py
[... cut ...]
ERROR components/components/spi_flash/test_apps/mspi_test/pytest_mspi_test.py
ERROR components/components/spiffs/spiffs/py/sp_test.py
ERROR managed_components/espressif__tinyusb/examples/device/hid_generic_inout/hid_test.py
!!!!!!!!!!!!!!!!!!!!!!!!!!!! Interrupted: 47 errors during collection !!!!!!!!!!!!!!!!!!!!!!!!!!!!!
================================= 17 warnings, 47 errors in 3.59s =================================
leon@LeonsNotebookMInt:~/uni/WS24/BA/bastion/hardware-trust-anchor/v15$ 
```

peter-marcisovsky, 2024-10-23:
Hi, 

> If I try to run pytest in the idf folder, I get following error

By doing so (by running esp-idf/ pytest --target esp32s3) you ordered the pytest to run all the pytest*.py files within the esp-idf. Try to run a pytest from a hello world example, for start, it should be something like 
`peter@peter ➜  ~/esp/esp-idf/examples/get-started/hello_world git:(master) ✗ pytest --target esp32s3`

> However, if I just call pytest in my VSCode directory 

AFAIK, calling just `pytest` does not do anything useful, until you don't specify a target `pytest --targe tesp32xx`

> I get following errors, which is a bit confusing since I did install the pytest dependencies with : $ install.sh --enable-pytest

Please take a look at the idf pytest docs about installation and running the pytest locally [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/contribute/esp-idf-tests-with-pytest.html#running-tests-locally).
I think you are missing `. ./export.sh` call after `install.sh --enable-pytest`

> E   KeyError: 'IDF_PATH'

Looks like you do not have the IDF_PATH set

Just try `echo $IDF_PATH` it should return you a path to the esp-idf (if the path has been set)

Please follow [instructions](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/linux-macos-setup.html) how to install and setup esp-idf on your linux machine. 
Looks like the same problem as above, missing `. ./export.sh`

LeonBirkel, 2024-10-23:
I decided to just not use pytest. I followed all the setup docs. 
Maybe a missed something.  Thank you for your help and replies!