espressif/esp-idf #14765: NMEA parser not printing GPS data

hendraze, 2024-10-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am attempting to run the NMEA 0183 parser example however it is not functioning as expected? I have everything setup as the example describes and have selected the sentence types my gps module can output in menuconfig, however all I receive on my monitor is:

I (342) nmea_parser: NMEA Parser init OK 

and then...

W (10552) nmea_parser: Ring Buffer Full
W (20612) nmea_parser: Ring Buffer Full
W (29112) nmea_parser: Ring Buffer Full
W (38392) nmea_parser: Ring Buffer Full

I have increased the ring buffer size to the maximum in menuconfig and also tried increasing the stack size however neither seem to fix the issue. 

I am very much a beginner to all this so appreciate the answer is most likely very obvious, so apologies in advance!

I am using the adafruit esp32 feather v2 and the adafruit ultimate GPS v3.

nopnop2002, 2024-10-26:
You can use this to check NMEA messages received from GPS.
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/uart/uart_async_rxtxtasks

Alvin1Zhang, 2025-03-28:
Thanks for reporting, saw there is also discussion on forum https://esp32.com/viewtopic.php?t=42454, feel free to reopen or let us track on the forum post, thanks.

closed 2025-03-28 by Alvin1Zhang