espressif/esp-idf #14768: ULP RISC-V Interrupts/Timers

michaelboeding, 2024-10-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I’m working on using the ESP32's ULP (Ultra-Low Power) mode for activity tracking with an accelerometer, and I’m trying to figure out the best way to implement this.

Here’s what I want to achieve:

Mark Activity as Active/Inactive: I need to monitor each minute. If a wake-up interrupt occurs from the accelerometer, I mark that minute as "active." If no interrupt occurs, it should be marked as "inactive."

Timer for Inactive Periods: I want a timer running in the ULP that resets every minute, marking the period as "inactive" if there’s no wake-up interrupt.

Main Application Sync: When the main app wakes up, I want to back-calculate activity based on the ULP's runtime and sync it with a Unix timestamp to create accurate time-stamped data.

Question:

How can I set up the ULP to trigger at regular intervals (like every minute) for marking inactivity, while still allowing GPIO interrupts for activity detection? What’s the best way to handle the timing and interrupts together in this scenario? I would rather not be constantly polling but just have it act like a RTC timer to wake up the ULP if possible. And if a wake up interrupt occurs reset the timer. 

Another option may just be a way to have the interrupts setup and then have a software interrupt fired from the ULP at a set interval that I can track but it doesn't seem consistent when I tried to do this.  I was never able to get the software interrupt in the ULP to say fire every 60 seconds. So is there a way to get the run time clock in the ULP?

sudeep-mohanty, 2024-10-23:
Hi @michaelboeding,
May I know which target are you using? Is it the ESP32 or the ESP32-S2/S3, which support the ULP RISC-V core in addition to the ULP FSM core?

The ULP cores do not have a free-running timer like an RTC timer to keep track of time. However, the ULP can be programmed to wakeup every minute (or fractions of a minute) to check on the active/inactive status.
If you are using a target with the ULP RISC-V core, you could also use the [ulp_riscv_delay_cycles()](https://github.com/espressif/esp-idf/blob/a97a7b0962da148669bb333ff1f30bf272946ade/components/ulp/ulp_riscv/ulp_core/include/ulp_riscv_utils.h#L97) to create a 60 second wait period where the ULP does nothing but waits either for the accelerometer IO interrupt or a timeout to happen.

michaelboeding, 2024-10-24:
# ULP Code

```c++

#include "ulp_riscv.h"
#include "ulp_riscv_utils.h"
#include <stdint.h>
#include "ulp_riscv.h"
#include "ulp_riscv_utils.h"
#include "ulp_riscv_i2c_ulp_core.h"
#include "ulp_riscv.h"
#include "ulp_riscv_utils.h"
#include "ulp_riscv_i2c_ulp_core.h"
#include <stdint.h>
#include "esp_attr.h"
#include "ulp_riscv_utils.h"
#include "ulp_riscv_gpio.h"

#define WAKEUP_PIN_1 GPIO_NUM_1
#define WAKEUP_PIN_2 GPIO_NUM_2

#define ULP_RISCV_CYCLES_PER_SEC (ULP_RISCV_CYCLES_PER_MS * 1000)

int sw_int_cnt = 0;
int wake_by_sw = 0;
int wake_by_gpio = 0;

// SW Interrupt Handler 
void sw_int_handler(void *arg) {
    sw_int_cnt++;
    ulp_riscv_wakeup_main_processor();
}

// GPIO Interrupt Handler 
void gpio_int1_handler(void *arg){
    wake_by_gpio = 1;
    ulp_riscv_wakeup_main_processor();
}

void gpio_int2_handler(void *arg){
    wake_by_gpio = 2;
    ulp_riscv_wakeup_main_processor();
}

//method used to setup the gpio intterupts 
void setupGpioIntterupts(){
    /* Configure GPIO in input mode for interrupt */
    ulp_riscv_gpio_init(WAKEUP_PIN_1);
    ulp_riscv_gpio_input_enable(WAKEUP_PIN_1);

    ulp_riscv_gpio_init(WAKEUP_PIN_2);
    ulp_riscv_gpio_input_enable(WAKEUP_PIN_2);

    /* Register GPIO interrupt handler */
    ulp_riscv_gpio_isr_register(WAKEUP_PIN_1, ULP_RISCV_GPIO_INTR_POSEDGE, gpio_int1_handler, NULL);
    ulp_riscv_gpio_isr_register(WAKEUP_PIN_2, ULP_RISCV_GPIO_INTR_POSEDGE, gpio_int2_handler, NULL);
}

//method used to setup the software intterupt 
void setupSoftwareIntterupt(){
    //Register SW interrupt handler 
    ulp_riscv_enable_sw_intr(sw_int_handler, NULL);
}

int main(void) {
    wake_by_sw = 0;
    //setup the gpio intterupts 
    setupGpioIntterupts();
    //setup the software intterupt 
    setupSoftwareIntterupt();
[... 104 lines cut ...]
    }

    while (1) {
        // 60-second delay before entering deep sleep
        ESP_LOGI(TAG, "Waiting 20 seconds before entering deep sleep...");
        for (int i = 20; i > 0; i--) {
            ESP_LOGI(TAG, "%d seconds remaining...", i);
            vTaskDelay(1000 / portTICK_PERIOD_MS);
        }

        ESP_LOGI(TAG, "Perparing to enter deep sleep");

        vTaskDelay(100);

        ESP_ERROR_CHECK( esp_sleep_enable_ulp_wakeup());

        ESP_LOGI(TAG, "Going to sleep");
        // Enter deep sleep
        esp_deep_sleep_start();
    }
}

```

Hey @sudeep-mohanty Just for context this is what I currently have also on the ESP32-S3 dev kit and im getting for example 42 seconds on the interrupt above when it should be 60 seconds.

sudeep-mohanty, 2024-10-24:
Hi @michaelboeding,

> im getting for example 42 seconds on the interrupt above when it should be 60 seconds

This happens because the main CPU waits for about 20 seconds before going in to deepsleep while the ULP RISC-V core has already started running and entered it's 60 second delay the moment you call `ulp_riscv_run()`. On subsequent iterations, the wakeup time is random because the ULP is always running and will have it's subsequent 60 second timer expiration randomly irrespective of when the main CPU enters deepsleep. My suggestion here would be to have tighter coupling of when you'd want to start the 60 second wait period on the ULP. Perhaps you could achieve this by a shared variable which is set by the main core right before entering deepsleep .

> Also interesting link here where it looks like they could read the RTC from the ULP? https://esp32.com/viewtopic.php?f=13&t=18013

I was mistaken! Indeed, the ULP can read the RTC timer and you should be able to read the values just as shown in the forum post.

michaelboeding, 2024-10-24:
Hey @sudeep-mohanty ....that makes sense I should have put that together. But if it's random after the first cycle then that won't work for my use case I guess. That being said the example with the RTC timer seems to be in assembly, is there anyway I could get an example on how to use the RISC-V in C? Or could you point me in the direction of how I would do this in C and I can attempt to make it work and share the code here? I think an example of this would be very valuable in the RISC-V examples. 

Im assuming in that case I could use that to read the time of every interrupt in the ULP when it happens and then can back calculate the actual time when my system wakes up from sleep? Basically log the active with a timestamp and then once I have the unix timestamp go back and calculate which minutes should be active/inactive. I will have another timer running via the RTC to wake up from deep sleep at set intervals that I would do this in. But ideally if I could manage all of this in the ULP slow memory with the timestamp it would make my application use a lot less power.

sudeep-mohanty, 2024-10-25:
Hi @michaelboeding,
You could have a more deterministic way of triggering the 60 second wait on the ULP. And, you should be able to read the RTC timer registers from the ULP as well. Your ULP code could look like -

```
uint32_t main_app_in_sleep = 0; // Set by the main app before going to sleep
uint32_t t0;
uint32_t t1;

int main(void) {
    ...
    ...
    
    //loop used for timing the software intterupt 
    while (1) {
       while (!main_app_in_sleep) {
           // Wait for the main app to go to sleep before starting the 60 second wait
       } 
       ulp_riscv_delay_cycles(ULP_RISCV_CYCLES_PER_SEC * 60);
       //wake up the main cpu every minute 
       main_app_in_sleep = 0;
       ulp_riscv_wakeup_main_processor();
       
       // Read RTC timer
       SET_PERI_REG_MASK(RTC_CNTL_TIME_UPDATE_REG, RTC_CNTL_TIME_UPDATE);
       t0 = READ_PERI_REG(RTC_CNTL_TIME0_REG);
       t1 = READ_PERI_REG(RTC_CNTL_TIME1_REG);
    }
    return 0;
}
```

It is also possible for you to log the RTC time stamp in the RTC slow memory every time the ULP receives an interrupt, without waking up the main core. Once you do wake up from sleep, you could read the logged timestamps from the main core and do your calculations on them.

michaelboeding, 2024-10-25:
Hey @sudeep-mohanty I implemented this and it doesn't seem to work. It seems like my ULP locks up completely whenever it trys to call the following 

```c++
// Read RTC timer
SET_PERI_REG_MASK(RTC_CNTL_TIME_UPDATE_REG, RTC_CNTL_TIME_UPDATE);
rtc_timestamp_low = READ_PERI_REG(RTC_CNTL_TIME0_REG);
rtc_timestamp_high = READ_PERI_REG(RTC_CNTL_TIME1_REG);
```

Full ULP code 

```c++
#include "ulp_riscv.h"
#include "ulp_riscv_utils.h"
#include <stdint.h>
#include "ulp_riscv.h"
#include "ulp_riscv_utils.h"
#include "ulp_riscv_i2c_ulp_core.h"
#include "ulp_riscv.h"
#include "ulp_riscv_utils.h"
#include "ulp_riscv_i2c_ulp_core.h"
#include <stdint.h>
#include "esp_attr.h"
#include "ulp_riscv_utils.h"
#include "ulp_riscv_gpio.h"
#include "soc/rtc_cntl_reg.h"
#include "ulp_riscv_register_ops.h"

// https://github.com/espressif/esp-idf/tree/release/v5.2/examples/system/ulp/ulp_riscv

//timing 
#define INTERRUPT_PERIOD_MS 20  // Software interrupt period in milliseconds
#define SECONDS_TO_WAIT 15       // Desired wait time in seconds
#define INTERRUPT_COUNT ((SECONDS_TO_WAIT * 1000) / INTERRUPT_PERIOD_MS)

#define ULP_RISCV_CYCLES_PER_SEC (ULP_RISCV_CYCLES_PER_MS * 1000)

//pins 
//these are the intterupt pins from the accelerometer
#define WAKEUP_PIN_1 GPIO_NUM_1
#define WAKEUP_PIN_2 GPIO_NUM_2

int sw_int_cnt = 0;
int wake_by_sw = 0;
int wake_by_gpio = 0;
//timestamp from the ulp the current time from the current wakeup 
uint32_t rtc_timestamp_low = 0;
uint32_t rtc_timestamp_high = 0;
//timestamp from the ulp the previous time from the previous wakeup in ulp 
uint32_t rtc_timestamp_low_prev = 0;
uint32_t rtc_timestamp_high_prev = 0;

/**
 * @brief Read the lower 32 bits of the real time clock counter.
 * @note At 150kHz, this value will overflow every ~7.95h.
 * 
 * @return uint32_t 
 */
static inline uint32_t ulp_riscv_get_rtc_32() {
    SET_PERI_REG_MASK(RTC_CNTL_TIME_UPDATE_REG, RTC_CNTL_TIME_UPDATE);
    return READ_PERI_REG(RTC_CNTL_TIME0_REG);
}

/**
 * @brief Read the full real time clock counter value (48 bits).
 * 
 * @return uint64_t 
 */
static inline uint64_t ulp_riscv_get_rtc() {
    SET_PERI_REG_MASK(RTC_CNTL_TIME_UPDATE_REG, RTC_CNTL_TIME_UPDATE);
    return (((uint64_t) READ_PERI_REG(RTC_CNTL_TIME1_REG)) << 32) | READ_PERI_REG(RTC_CNTL_TIME0_REG);
}

// SW Interrupt Handler 
void sw_int_handler(void *arg) {
    sw_int_cnt++;
    //ulp_riscv_wakeup_main_processor();
}

// GPIO Interrupt Handler 
void gpio_int1_handler(void *arg){
    wake_by_gpio = 1;
    //set the previous timestamp 
    rtc_timestamp_low_prev = rtc_timestamp_low;
    rtc_timestamp_high_prev = rtc_timestamp_high;
    // Read RTC timer
    SET_PERI_REG_MASK(RTC_CNTL_TIME_UPDATE_REG, RTC_CNTL_TIME_UPDATE);
    rtc_timestamp_low = READ_PERI_REG(RTC_CNTL_TIME0_REG);
    rtc_timestamp_high = READ_PERI_REG(RTC_CNTL_TIME1_REG);
    ulp_riscv_wakeup_main_processor();
}

void gpio_int2_handler(void *arg){
    wake_by_gpio = 2;
    //set the previous timestamp 
    rtc_timestamp_low_prev = rtc_timestamp_low;
    rtc_timestamp_high_prev = rtc_timestamp_high;
    // Read RTC timer
    SET_PERI_REG_MASK(RTC_CNTL_TIME_UPDATE_REG, RTC_CNTL_TIME_UPDATE);
    rtc_timestamp_low = READ_PERI_REG(RTC_CNTL_TIME0_REG);
    rtc_timestamp_high = READ_PERI_REG(RTC_CNTL_TIME1_REG);
    ulp_riscv_wakeup_main_processor();
}

//method used to setup the gpio intterupts 
void setupGpioIntterupts(){
    /* Configure GPIO in input mode for interrupt */
    ulp_riscv_gpio_init(WAKEUP_PIN_1);
    ulp_riscv_gpio_input_enable(WAKEUP_PIN_1);

    ulp_riscv_gpio_init(WAKEUP_PIN_2);
    ulp_riscv_gpio_input_enable(WAKEUP_PIN_2);

    /* Register GPIO interrupt handler */
    ulp_riscv_gpio_isr_register(WAKEUP_PIN_1, ULP_RISCV_GPIO_INTR_POSEDGE, gpio_int1_handler, NULL);
    ulp_riscv_gpio_isr_register(WAKEUP_PIN_2, ULP_RISCV_GPIO_INTR_POSEDGE, gpio_int2_handler, NULL);
}

//method used to setup the software intterupt 
void setupSoftwareIntterupt(){
    //Register SW interrupt handler 
    ulp_riscv_enable_sw_intr(sw_int_handler, NULL);
}

int main(void) {
    wake_by_sw = 0;
    //set the previous timestamp 
    rtc_timestamp_low_prev = 0;
    rtc_timestamp_high_prev = 0;
    rtc_timestamp_low = 0;
    rtc_timestamp_high = 0;
    //setup the gpio intterupts 
    setupGpioIntterupts();
    //setup the software intterupt 
    setupSoftwareIntterupt();

    while (1) {
        /* Trigger SW interrupt */
        ulp_riscv_trigger_sw_intr();

        if ((sw_int_cnt % 5 == 0) && !wake_by_gpio) {
            wake_by_sw = 1;
            ulp_riscv_wakeup_main_processor();
        }
        ulp_riscv_delay_cycles(1000 * ULP_RISCV_CYCLES_PER_MS);
    }
    
    return 0;
}
```

michaelboeding, 2024-10-29:
Hey @sudeep-mohanty - again thanks for working on this. Are you reading the RTC in your example also? It seems when I read the RTC in my interrupt the ULP locks up...no errors really. Seems to work when I don't try to read the RTC. I can provide my main code also if that is helpful?

sudeep-mohanty, 2024-10-29:
Hi @michaelboeding,
I have the exact same code as you have shared, i.e, the RTC timer registers are read from the GPIO interrupts. And I have an updated main code which prints the RTC timer values which are populated by the ULP. I can run it on v5.3.1 as well. Here is an output -

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x5 (DSLEEP),boot:0x8 (SPI_FAST_FLASH_BOOT)
pro cpu reset by JTAG
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x12b8
load:0x403c8700,len:0x4
load:0x403c8704,len:0xaf4
load:0x403cb700,len:0x2bfc
entry 0x403c8898
W (89) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
ULP RISC-V woke up the main CPU!
ULP RISC-V SW Interrupt triggered 10 times.
Entering in deep sleep

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x5 (DSLEEP),boot:0x8 (SPI_FAST_FLASH_BOOT)
pro cpu reset by JTAG
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x12b8
load:0x403c8700,len:0x4
load:0x403c8704,len:0xaf4
load:0x403cb700,len:0x2bfc
entry 0x403c8898
W (89) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
ULP RISC-V woke up the main CPU!
ULP RISC-V SW Interrupt triggered 15 times.
Entering in deep sleep

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x5 (DSLEEP),boot:0x8 (SPI_FAST_FLASH_BOOT)
pro cpu reset by JTAG
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x12b8
load:0x403c8700,len:0x4
load:0x403c8704,len:0xaf4
load:0x403cb700,len:0x2bfc
entry 0x403c8898
W (89) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
ULP RISC-V woke up the main CPU!
ULP RISC-V GPIO Interrupt triggered.
rtc_time_low 2030968
rtc_time_high 0
rtc_time_low 2022030
rtc_time_high 0
Entering in deep sleep
```

michaelboeding, 2024-10-29:
Got it - I'm traveling today but I'll try to look at this tonight then. I had that exact same code running before and every time I tried to read that RTC in an interrupt it basically just hung and refused to do anything. Once I commented that RTC read out it all worked again.

sudeep-mohanty, 2024-10-29:
I see. There could be something different in our setups. When you have time, do let me know if you could run this at your end or not. If not, we'll try to narrow down the differences. Thanks again!

michaelboeding, 2024-10-29:
Sounds good will do! Thanks I'll let you know.

michaelboeding, 2024-10-30:
Just to keep you looped in I'm traveling till Friday and forgot my Dev kit. So I won't be able to test till then.

michaelboeding, 2024-11-03:
Hey @sudeep-mohanty ,

I have spent the whole day trying to debug this again. I seem to be running into some weird cases here. Below is my exact code

### MAIN.cpp

```c++

static const char *TAG = "MAIN";

extern const uint8_t ulp_main_bin_start[] asm("_binary_ulp_main_bin_start");
extern const uint8_t ulp_main_bin_end[]   asm("_binary_ulp_main_bin_end");

extern "C" {
    void app_main();
}

//method used to power on the board
void holdPowerOn() {
    gpio_config_t io_conf;
    // Configure GPIO_NUM_14 as output for powering on the entire board
    io_conf.intr_type = GPIO_INTR_DISABLE;
    io_conf.mode = GPIO_MODE_OUTPUT;
    io_conf.pin_bit_mask = (1ULL<<MCU_POWER_PIN); // GPIO_NUM_14 for board power
    io_conf.pull_down_en = GPIO_PULLDOWN_DISABLE;
    io_conf.pull_up_en = GPIO_PULLUP_DISABLE;
    gpio_config(&io_conf);
    // Set GPIO 14 to high to power on the entire board
    gpio_set_level(MCU_POWER_PIN, 1);
    //MARK: Used to configure the GPIO 8 to be off since this controlles the power to the cellular module and we want it off at first
    // Configure GPIO 8 as output and ensure it is set to off (low)
    io_conf.pin_bit_mask = (1ULL << CELLULAR_POWER_PIN); // GPIO 8 for your specific use case
    gpio_config(&io_conf); // Apply the configuration to GPIO 8
    // Set GPIO 8 to low to ensure it is off
    gpio_set_level(CELLULAR_POWER_PIN, 0);
    // Enable GPIO hold to retain the pin state during light-sleep
    gpio_hold_en(MCU_POWER_PIN);
}

//method used to setup ulp program and setup wakeup pins
static void setup_ulp_program(void){
    // Initialize ULP
    esp_err_t err = ulp_riscv_load_binary(ulp_main_bin_start, (ulp_main_bin_end - ulp_main_bin_start));
    ESP_ERROR_CHECK(err);
    //The first argument is the period index, which is not used by the ULP-RISC-V timer
    //The second argument is the period in microseconds, which gives a wakeup time period of: 20ms
    ulp_set_wakeup_period(0, 20000);
    /* Start the program */
    err = ulp_riscv_run();
    ESP_ERROR_CHECK(err);
}

//setup application here 
void app_main(){
    // Initialize NVS
    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_LOGI("APP", "NVS flash initialized");
    // Install ISR service
    gpio_install_isr_service(0);
    
    /*
    //Initialize NVS
    esp_err_t ret = nvs_flash_init();
    //check for errors
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES){
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);
    nvs_flash_erase();
[... 621 lines cut ...]
void Accelerometer::setup_gpio_interrupts() {
    // Configuration for GPIO 12
    gpio_config_t io_conf_1 = {};
    io_conf_1.intr_type = GPIO_INTR_ANYEDGE; // Set for any edge (both rising and falling)
    io_conf_1.pin_bit_mask = (1ULL << GPIO_NUM_12); // Use GPIO 12
    io_conf_1.mode = GPIO_MODE_INPUT;
    io_conf_1.pull_up_en = GPIO_PULLUP_DISABLE;
    gpio_config(&io_conf_1);

    // Configuration for GPIO 11
    gpio_config_t io_conf_2 = {};
    io_conf_2.intr_type = GPIO_INTR_ANYEDGE; // Set for any edge (both rising and falling)
    io_conf_2.pin_bit_mask = (1ULL << GPIO_NUM_11); // Use GPIO 11
    io_conf_2.mode = GPIO_MODE_INPUT;
    io_conf_2.pull_up_en = GPIO_PULLUP_DISABLE;
    gpio_config(&io_conf_2);

    // Add ISR handlers
    gpio_isr_handler_add(GPIO_NUM_12, gpio_isr_handler_1, (void*) this); // For GPIO 11
    gpio_isr_handler_add(GPIO_NUM_11, gpio_isr_handler_2, (void*) this); // For GPIO 12
}

```

I'm really stumped here. I also did uncomment the sw interrupt and it continued to fire as the program went on after the first run but the GPIO wakeup would never fire again after the first time. I think these were my original issues just masked by reading the RTC timer. Any help would be appreciated...or any ideas.

michaelboeding, 2024-11-03:
Heres also my sdkconfig file.

```bash

#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) 5.3.1 Project Configuration
#
CONFIG_SOC_MPU_MIN_REGION_SIZE=0x20000000
CONFIG_SOC_MPU_REGIONS_MAX_NUM=8
CONFIG_SOC_ADC_SUPPORTED=y
CONFIG_SOC_UART_SUPPORTED=y
CONFIG_SOC_PCNT_SUPPORTED=y
CONFIG_SOC_PHY_SUPPORTED=y
CONFIG_SOC_WIFI_SUPPORTED=y
CONFIG_SOC_TWAI_SUPPORTED=y
CONFIG_SOC_GDMA_SUPPORTED=y
CONFIG_SOC_AHB_GDMA_SUPPORTED=y
CONFIG_SOC_GPTIMER_SUPPORTED=y
CONFIG_SOC_LCDCAM_SUPPORTED=y
CONFIG_SOC_MCPWM_SUPPORTED=y
CONFIG_SOC_DEDICATED_GPIO_SUPPORTED=y
CONFIG_SOC_CACHE_SUPPORT_WRAP=y
CONFIG_SOC_ULP_SUPPORTED=y
CONFIG_SOC_ULP_FSM_SUPPORTED=y
CONFIG_SOC_RISCV_COPROC_SUPPORTED=y
CONFIG_SOC_BT_SUPPORTED=y
CONFIG_SOC_USB_OTG_SUPPORTED=y
CONFIG_SOC_USB_SERIAL_JTAG_SUPPORTED=y
CONFIG_SOC_CCOMP_TIMER_SUPPORTED=y
CONFIG_SOC_ASYNC_MEMCPY_SUPPORTED=y
CONFIG_SOC_SUPPORTS_SECURE_DL_MODE=y
CONFIG_SOC_EFUSE_KEY_PURPOSE_FIELD=y
CONFIG_SOC_EFUSE_SUPPORTED=y
CONFIG_SOC_SDMMC_HOST_SUPPORTED=y
CONFIG_SOC_RTC_FAST_MEM_SUPPORTED=y
CONFIG_SOC_RTC_SLOW_MEM_SUPPORTED=y
CONFIG_SOC_RTC_MEM_SUPPORTED=y
CONFIG_SOC_PSRAM_DMA_CAPABLE=y
CONFIG_SOC_XT_WDT_SUPPORTED=y
CONFIG_SOC_I2S_SUPPORTED=y
CONFIG_SOC_RMT_SUPPORTED=y
CONFIG_SOC_SDM_SUPPORTED=y
CONFIG_SOC_GPSPI_SUPPORTED=y
CONFIG_SOC_LEDC_SUPPORTED=y
CONFIG_SOC_I2C_SUPPORTED=y
CONFIG_SOC_SYSTIMER_SUPPORTED=y
CONFIG_SOC_SUPPORT_COEXISTENCE=y
CONFIG_SOC_TEMP_SENSOR_SUPPORTED=y
CONFIG_SOC_AES_SUPPORTED=y
CONFIG_SOC_MPI_SUPPORTED=y
CONFIG_SOC_SHA_SUPPORTED=y
CONFIG_SOC_HMAC_SUPPORTED=y
CONFIG_SOC_DIG_SIGN_SUPPORTED=y
CONFIG_SOC_FLASH_ENC_SUPPORTED=y
CONFIG_SOC_SECURE_BOOT_SUPPORTED=y
CONFIG_SOC_MEMPROT_SUPPORTED=y
CONFIG_SOC_TOUCH_SENSOR_SUPPORTED=y
CONFIG_SOC_BOD_SUPPORTED=y
CONFIG_SOC_CLK_TREE_SUPPORTED=y
CONFIG_SOC_MPU_SUPPORTED=y
CONFIG_SOC_WDT_SUPPORTED=y
CONFIG_SOC_SPI_FLASH_SUPPORTED=y
CONFIG_SOC_RNG_SUPPORTED=y
CONFIG_SOC_LIGHT_SLEEP_SUPPORTED=y
CONFIG_SOC_DEEP_SLEEP_SUPPORTED=y
CONFIG_SOC_LP_PERIPH_SHARE_INTERRUPT=y
CONFIG_SOC_PM_SUPPORTED=y
CONFIG_SOC_XTAL_SUPPORT_40M=y
CONFIG_SOC_APPCPU_HAS_CLOCK_GATING_BUG=y
[... 2219 lines cut ...]
# CONFIG_PPP_MPPE_SUPPORT is not set
# CONFIG_PPP_DEBUG_ON is not set
CONFIG_ESP32S3_TIME_SYSCALL_USE_RTC_SYSTIMER=y
CONFIG_ESP32S3_TIME_SYSCALL_USE_RTC_FRC1=y
# CONFIG_ESP32S3_TIME_SYSCALL_USE_RTC is not set
# CONFIG_ESP32S3_TIME_SYSCALL_USE_SYSTIMER is not set
# CONFIG_ESP32S3_TIME_SYSCALL_USE_FRC1 is not set
# CONFIG_ESP32S3_TIME_SYSCALL_USE_NONE is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options

```

michaelboeding, 2024-11-03:
Hey @sudeep-mohanty  I think I got everything working. 

To solve my first issue I needed to call  

```c++
ulp_riscv_halt() 
```
on any new wakeup. And then for the second I needed to call 

```c++
rtc_gpio_init(GPIO_NUM_11);
rtc_gpio_set_direction(GPIO_NUM_11, RTC_GPIO_MODE_INPUT_ONLY);
```

And im currently testing the RTC with an external crystal but I think it is working correctly in deep sleep also so far. As long as those solutions look good to you I will probably close this issue. Let me know - thanks

michaelboeding, 2024-11-04:
Hey @sudeep-mohanty it looks like we don't have the parallel resistor on the external crystal. I was able to run the code with the esp_rtc_get_time_us() and got the following I (1029) Main: RTC time: 1264849 us. Could this be the issue? WE can rev this.

sudeep-mohanty, 2024-11-06:
Hi @michaelboeding,
I wouldn't be the right person to comment on the hardware design. I suggest that you write to our hardware review team directly at https://www.espressif.com/en/contact-us/circuit-schematic-pcb-design-review and the business support team should get back to you.

sudeep-mohanty, 2024-12-06:
No update needed in ESP-IDF for this ticket. Closing it. Please feel free to reopen this ticket or open a new one for further support. Thanks!

closed 2024-12-06 by sudeep-mohanty