espressif/esp-idf #14769: ESP32C6无法使用ADC监视器

yolee599, 2024-10-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1-536-g5955059940

### Espressif SoC revision.

ESP32-C6FH4 (QFN32) (revision v0.1)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6-DevKitM-1 V1.0

### Power Supply used.

External 5V

### What is the expected behavior?

在使用  adc_continuous 的时候，想使用 adc_monitor 监控电压，电压超过阈值就执行回调。

### What is the actual behavior?

实际使用的时候没有报错，也没有发生回调。
[adc_test_main.c.txt](https://github.com/user-attachments/files/17486087/adc_test_main.c.txt)

### Steps to reproduce.

1. 编译 adc_test_main.c 文件。
2. 烧录。
3. 短接 IO2 和 3V3

### Debug Logs.

```plain
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x1528
load:0x4086c110,len:0xe2c
load:0x4086e610,len:0x2e58
entry 0x4086c11a
I (23) boot: ESP-IDF v5.3.1-536-g5955059940 2nd stage bootloader
I (23) boot: compile time Oct 23 2024 11:23:37
I (24) boot: chip revision: v0.0
I (24) boot.esp32c6: SPI Speed      : 80MHz
I (28) boot.esp32c6: SPI Mode       : DIO
I (32) boot.esp32c6: SPI Flash Size : 4MB
I (36) boot: Enabling RNG early entropy source...
I (40) boot: Partition Table:
I (43) boot: ## Label            Usage          Type ST Offset   Length
I (49) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (56) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (62) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (69) boot:  3 factory          factory app      00 00 00010000 00100000
I (75) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (82) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (88) boot: End of partition table
I (91) boot: Defaulting to factory image
I (95) esp_image: segment 0: paddr=00010020 vaddr=42020020 size=0a520h ( 42272) map
I (111) esp_image: segment 1: paddr=0001a548 vaddr=40800000 size=05ad0h ( 23248) load
I (118) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=1544ch ( 87116) map
I (137) esp_image: segment 3: paddr=00035474 vaddr=40805ad0 size=068a4h ( 26788) load
I (144) esp_image: segment 4: paddr=0003bd20 vaddr=4080c380 size=01830h (  6192) load
I (149) boot: Loaded app from partition at offset 0x10000
I (149) boot: Disabling RNG early entropy source...
I (162) cpu_start: Unicore app
I (170) cpu_start: Pro cpu start user code
I (170) cpu_start: cpu freq: 160000000 Hz
I (171) app_init: Application information:
I (171) app_init: Project name:     smart_lock_mcu
I (175) app_init: App version:      1
I (178) app_init: Compile time:     Oct 23 2024 11:28:55
I (183) app_init: ELF file SHA256:  6e2cecbf1...
I (188) app_init: ESP-IDF:          v5.3.1-536-g5955059940
I (193) efuse_init: Min chip rev:     v0.0
I (197) efuse_init: Max chip rev:     v0.99 
I (201) efuse_init: Chip rev:         v0.0
I (205) heap_init: Initializing. RAM available for dynamic allocation:
I (211) heap_init: At 4080EB90 len 0006DA80 (438 KiB): RAM
I (216) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
I (221) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (227) spi_flash: detected chip: generic
I (230) spi_flash: flash io: dio
I (234) sleep: Configure to isolate all GPIO pins in sleep state
I (239) sleep: Enable automatic switching of GPIO sleep configuration
I (245) coexist: coex firmware version: 4482466
I (250) coexist: coexist rom version 5b8dcfa
I (254) main_task: Started on CPU0
I (254) main_task: Calling app_main()
I (254) gpio: GPIO[22]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (264) gpio: GPIO[23]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (274) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1284) adc_test: Channel: 2, Value: 0x1, Vol: 0.268620
I (2284) adc_test: Channel: 2, Value: 0x1, Vol: 0.268620
I (3284) adc_test: Channel: 2, Value: 0x0, Vol: 0.000000
I (4284) adc_test: Channel: 2, Value: 0x0, Vol: 0.000000
I (5284) adc_test: Channel: 2, Value: 0x0, Vol: 0.000000
I (6284) adc_test: Channel: 2, Value: 0x1, Vol: 0.268620
I (7284) adc_test: Channel: 2, Value: 0x0, Vol: 0.000000
I (8284) adc_test: Channel: 2, Value: 0x0, Vol: 0.000000
I (9284) adc_test: Channel: 2, Value: 0x0, Vol: 0.000000
I (10284) adc_test: Channel: 2, Value: 0x0, Vol: 0.000000
I (11284) adc_test: Channel: 2, Value: 0x0, Vol: 0.000000
I (12284) adc_test: Channel: 2, Value: 0x1, Vol: 0.268620
I (13284) adc_test: Channel: 2, Value: 0x0, Vol: 0.000000
I (14284) adc_test: Channel: 2, Value: 0x0, Vol: 0.000000
I (15284) adc_test: Channel: 2, Value: 0x1, Vol: 0.268620
I (16284) adc_test: Channel: 2, Value: 0x1, Vol: 0.268620
I (17284) adc_test: Channel: 2, Value: 0x1, Vol: 0.268620
I (18284) adc_test: Channel: 2, Value: 0x1, Vol: 0.268620
I (19284) adc_test: Channel: 2, Value: 0x1, Vol: 0.268620
I (20284) adc_test: Channel: 2, Value: 0x0, Vol: 0.000000
I (21284) adc_test: Channel: 2, Value: 0x0, Vol: 0.000000
I (22284) adc_test: Channel: 2, Value: 0x1, Vol: 0.268620
I (23284) adc_test: Channel: 2, Value: 0x1, Vol: 0.268620
I (24284) adc_test: Channel: 2, Value: 0x0, Vol: 0.000000
I (25284) adc_test: Channel: 2, Value: 0x1, Vol: 0.268620
I (26284) adc_test: Channel: 2, Value: 0x1, Vol: 0.268620
I (27284) adc_test: Channel: 2, Value: 0x1, Vol: 0.268620
I (28284) adc_test: Channel: 2, Value: 0x0, Vol: 0.000000
I (29284) adc_test: Channel: 2, Value: 0x0, Vol: 0.000000
I (30284) adc_test: Channel: 2, Value: 0x390, Vol: 244.981685
I (31284) adc_test: Channel: 2, Value: 0xcd0, Vol: 881.074481
I (32284) adc_test: Channel: 2, Value: 0xff1, Vol: 1096.239316
I (33284) adc_test: Channel: 2, Value: 0xff1, Vol: 1096.239316
I (34284) adc_test: Channel: 2, Value: 0xff1, Vol: 1096.239316
I (35284) adc_test: Channel: 2, Value: 0xff1, Vol: 1096.239316
```

MageDelfador, 2024-10-26:
Same issue on ESP32S3

suda-morris, 2024-11-02:
[Uploading 0001-fix-esp_adc-fixed-adc-continue-monitor-don-t-work-is.patch.zip…]()

Please can try this patch for now.

MageDelfador, 2024-11-02:
> Uploading 0001-fix-esp_adc-fixed-adc-continue-monitor-don-t-work-is.patch.zip…
> 
> Please can try this patch for now.

Cannot download patch

MageDelfador, 2024-11-03:
@suda-morris Can you re-upload the patch?

suda-morris, 2024-11-04:
Sorry guys for my broken network. I updated the link again. @MageDelfador

3w3rt0n, 2024-11-05:
The fix resolved it here, thanks @suda-morris

[image]

The serial print has a delay in this code, the threshold occurs exactly when I change the value of the potentiometer.

closed 2024-11-14 by espressif-bot