espressif/esp-idf #14770: idf v5.2.3 memory corruption in wifi_prov_mgr

ph2355, 2024-10-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.3

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

In wifi_prov_mgr_configure_sta should not cause heap corruption. Tried the same program on v5.2.2 which gave no heap corruption.

### What is the actual behavior?

In function wifi_prov_mgr_configure_sta after calling esp_wifi_set_config(WIFI_IF_STA, wifi_cfg)  on line 1267 i get heap corruption. I checked with surrounding the call by heap_caps_check_integrity_all.

### Steps to reproduce.

1. Enable wifi SoftAP provisioning in sdkconfig.
2. esp_netif_create_default_wifi_ap()
3. esp_wifi_set_config(ESP_IF_WIFI_AP, &wifi_config);
    esp_wifi_start();
4. Then when i receive credentials i call wifi_prov_mgr_configure_sta

### Debug Logs.

```plain
W (09:23:50.885) wifi_prov_mgr: heap integrity: 1
W (101953) wifi:Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Setting breakpoint at 0x42140a84 and returning...
--- 0x42140a84: block_size at /home/peter/esp/v5.2.3/esp-idf/components/heap/tlsf/tlsf_block_functions.h:20
 (inlined by) block_is_last at /home/peter/esp/v5.2.3/esp-idf/components/heap/tlsf/tlsf_block_functions.h:31
 (inlined by) tlsf_walk_pool at /home/peter/esp/v5.2.3/esp-idf/components/heap/tlsf/tlsf.c:768
```

### More Information.

I added heap_integrity checks to  wifi_prov_mgr_configure_sta. The first line in debug logs is printed just before calling esp_wifi_set_config(WIFI_IF_STA, wifi_cfg). wifi_prov_mgr_configure_sta sometimes returns, sometimes crashes before returning. If the function returns, the call to heap integrity check immediately after returns 0. 
The crash in my program is caused by my monitor which calls heap_caps_get_info every second, so it does not always occur at the same time and same location.

hrushikesh430, 2024-12-02:
Hello @ph2355, 
Could you please tell, whether you are facing the issue while trying your own custom example (outside the esp-idf) or esp-idf wifi_prov_mgr example?

ph2355, 2024-12-03:
I am using my own custom code. I tried reproducing the error using wifi_prov_mgr example, but could not achieve the same behaviour. Heap corruption occurs in my custom code using idf version 5.2.3. If  I switch to 5.2.2 or any other earlier version, the error is not there. 
Has the wifi_prov_mgr code changed in v5.2.3?

Since I cannot reproduce the error with the esp-idf provided example, you may close this issue.

hrushikesh430, 2024-12-03:
Hello @ph2355, I was searching looking if any changes are made in v5.2.3, but found that changes are made regarding BLE provisioning (for [ref](https://github.com/espressif/esp-idf/commit/59aa45ad2336c381066b3e495d63209ce98a62c0)). Hope this will help you.

ph2355, 2024-12-04:
True, the issue is present with both BLE and SoftAP provisioning. I will investigate further and reopen if I find anything.

ph2355, 2024-12-09:
Upon further inspection I have found that setting wifi authmode to `WIFI_AUTH_WPA2_PSK` resolves the issue.
So in `wifi_prov_mgr_configure_sta` I added the line `wifi_cfg->sta.threshold.authmode = WIFI_AUTH_WPA2_PSK;`
This overrides the default `WIFI_AUTH_OPEN` option from which the esp wifi lib automatically deduces the authmode based on the password length. 

It turns out this line is the problem, which I did not see in versions prior to idfv5.2.3:
`wifi:Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2`

Now I know setting the authmode to WPA2 statically is not a solution, so the issue must be resolved in the wifi lib.

hrushikesh430, 2024-12-09:
> So in wifi_prov_mgr_configure_sta I added the line wifi_cfg->sta.threshold.authmode = WIFI_AUTH_WPA2_PSK;

Could you please me patch file to reproduce this issue locally, as I am unable to recreate the issue.

ph2355, 2024-12-09:
This patches wifi_provisioning component in esp-idf and resolves my issue. 

[patchfile.txt](https://github.com/user-attachments/files/18061055/patchfile.txt)

hrushikesh430, 2024-12-09:
hello @ph2355,  will check and let you know

hrushikesh430, 2024-12-10:
Hello @ph2355 , I just tried to reproduce your issue - 

1. I checkout to master
2. Applied you patchfile.txt on esp-idf
3. Set the transport mode to BLE for wifi_provisioning
4. Turned on the hotspot with WPA-2 security (whose credentials will be provisioned)
5. Scanned the QR code through BLE mobile app and selected the network from the list 
6. Entered the correct credentials (successfully provisioned).

Is this procedure correct to reproduce your issue or I missed or misunderstood something, if yes, suggest your changes.

ph2355, 2024-12-10:
The patch I provided resolves the issue for me, not causes it. I don't know how to reproduce the issue in any other firmware than my own. I tried reproducing the issue in wifi provisioning example without success.

hrushikesh430, 2024-12-10:
Hello @ph2355, could you please tell whether your are using esp-mesh-lite in your project? I was searching  for your issue got to know that this can be reason for this issue.

ph2355, 2024-12-10:
Nope. Just using WiFi provisioning using nimble. After provisioning is done BLE is deinitialized.

ph2355, 2024-12-10:
Ok but this was added in idfv5.4 if I am not mistaken. I will try to install 5.3 or 5.4 to see if the error persists.

hrushikesh430, 2024-12-10:
Okay, I will try to do backward compatibility and will inform you

hrushikesh430, 2024-12-10:
Hello @ph2355, Actually we do not support 4.4.8 version (according to esp-idf support [policy](https://github.com/espressif/esp-idf)), so can't add fix here. But we will add this fix this support for v5.3

ph2355, 2024-12-10:
4.4.8 version of idf? I am running 5.2.3.

hrushikesh430, 2024-12-10:
Sorry I just misunderstood, I will do back-port for 5.2.3

ph2355, 2024-12-10:
I don't think it will be any different. I tried with version 5.4 and still have the same issue. But using `CONFIG_EXAMPLE_PROV_ENABLE_APP_CALLBACK` I was able to set authmode to WPA2_PSK resolves the issue.

hrushikesh430, 2024-12-10:
Great, If anything requires or if any issues/doubts are there you can ask or we can close this issue

ph2355, 2024-12-10:
I still do not know the cause of this issue. I just know that setting authmode manually resolves it. At this time I am out of ideas what else to try, so this issue may be closed. Will look for the same issue in future versions of idf and report.

closed 2024-12-10 by ph2355

hrushikesh430, 2024-12-10:
Sure @ph2355, will inform you if we get similar issue in future. Thank you.