espressif/esp-idf #14775: 带负号时小转大的强制转换问题

hvhghv, 2024-10-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

idf v5.3.1

### Espressif SoC revision.

ESP32C3 ESP32S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32C3

### Power Supply used.

USB

### What is the expected behavior?

```
#include "freertos/FreeRTOS.h"
#include "cJSON.h"
#include "stdio.h"

void task(){
    char a = -5;
    printf("%d, %d, %f\n", a, (int)a, (float)a);

    cJSON *root = cJSON_CreateObject();
    cJSON_AddNumberToObject(root, "a", (double)a);
    printf("%s\n", cJSON_Print(root));

}

```

这段代码在esp32的表现与在windows-x64上的表现不同

### What is the actual behavior?

在esp32上，输出
```
251, 251, 251.000000
{
	"a":	251
}
```

在window-x64上，输出
```
-5, -5, -5.000000
{
        "a":    -5
}
```

### Steps to reproduce.

1. Step
在window上通过idf进行编译，烧录到开发板上，从串口中得到数据

dobairoland, 2024-10-25:
Datatypes like `char` and `int` are different on different platforms. Your Windows is 64-bit AMD64 and ESP32 is 32-bit Xtensa. At least the length of the datatypes will be different. If you want to achieve the same behavior then you can try `uint8_t`, `int8_t` and similar, with our without `u` (unsigned) and 16 or 32 instead of 8 for larger datatypes.

Lapshin, 2024-10-25:
@hvhghv , and for more portability consider using `PRIxx` macros in printf/scanf functions. (https://en.cppreference.com/w/c/types/integer#Format_macro_constants)

See the line for example: https://github.com/espressif/esp-idf/blob/d155a04fd50ec5303f2d980ca11c7de735e7eea7/examples/get-started/hello_world/main/hello_world_main.c#L40

closed 2025-03-07 by Alvin1Zhang