espressif/esp-idf #14777: gptimer_start(xxx): timer is not enabled yet

baldhead69, 2024-10-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

The gp_timer is working, but show this message all the time i start the timer.

Full description here:
http://esp32.io/viewtopic.php?f=13&t=42388&sid=6ba7d8d7963b2f6055212842abd2b835&start=10

Notes:
ESP32-S3
esp-idf v5.3.1

baldhead69, 2024-11-08:
No one ???

ruig25, 2025-03-10:
I use gptimer with callback as one-shot timer with restart on demand.

Trying to restart the timer get following error:

* ESP-IDF v5.3.1: timer is not enabled yet
* ESP-IDF v5.4: timer is not ready for a new start

```
static gptimer_handle_t sensor_read_timer = NULL;
uint16_t timert = ;
...
void sensor_reader_task(void *pvParameters)
{
  const TickType_t xMaxBlockTime = pdMS_TO_TICKS(20);
  while (1) {
    uint32_t ulNotificationValue = ulTaskNotifyTake(pdTRUE, xMaxBlockTime);
    if (ulNotificationValue > 0) {
      if (some sophisticated condition to restart the timer) {
        restart_single_shot_timer();
      }
  }
  vTaskDelete(NULL);
}
...
static bool IRAM_ATTR sensor_read_isr(gptimer_handle_t timer, const gptimer_alarm_event_data_t *edata, void *user_data)
{
  gptimer_stop(sensor_read_timer); // workaround to avoid errors ******************************************

  BaseType_t xHigherPriorityTaskWoken = pdFALSE;
  vTaskNotifyGiveFromISR(sensor_reader_task_handle, &xHigherPriorityTaskWoken);

  return (xHigherPriorityTaskWoken == pdTRUE);
}
...
void init() {
...
  gptimer_config_t timer_config;
  memset(&timer_config, 0, sizeof(timer_config));
  timer_config.clk_src = GPTIMER_CLK_SRC_DEFAULT; // Use default clock source
  timer_config.direction = GPTIMER_COUNT_UP;      // Timer counts up
  timer_config.resolution_hz = 1000000;            // 1 MHz resolution (1 tick = 1 µs)
  ESP_ERROR_CHECK(gptimer_new_timer(&timer_config, &sensor_read_timer));

  gptimer_alarm_config_t alarm_config;
  memset(&alarm_config, 0, sizeof(alarm_config));
  alarm_config.alarm_count = timert; // Fire after N microseconds
  alarm_config.reload_count = 0;
  alarm_config.flags.auto_reload_on_alarm = false; // One-shot mode (no auto-reload)
  ESP_ERROR_CHECK(gptimer_set_alarm_action(sensor_read_timer, &alarm_config));

  gptimer_event_callbacks_t cbs = {
    .on_alarm = sensor_read_isr
  };
  ESP_ERROR_CHECK(gptimer_register_event_callbacks(sensor_read_timer, &cbs, NULL));

  ESP_LOGI(TAG, "gptimer_enable about to be called...");
  ESP_ERROR_CHECK(gptimer_enable(sensor_read_timer));

  restart_single_shot_timer(); // first call
}
...
esp_err_t restart_single_shot_timer(void)
{
  esp_err_t ret = gptimer_set_raw_count(sensor_read_timer, 0);  // Reset counter to 0
  ret |= gptimer_start(sensor_read_timer); // Restart timer ---> error here without workaround above
  return ret;
}
...
```

marccane, 2025-07-01:
I think I found a fix/workaround. I'm on version 5.3.1 and using the timer with `.auto_reload_on_alarm = false`. 
To reload the timer I use `gptimer_set_alarm_action(gptimer, &alarm_config);` **without** calling `gptimer_start(gptimer);`. It works for me without printing any errors now.

(I have not been able to read the full description because the esp32.io page is down)