espressif/esp-idf #14778: HTTP Client download large file

GiulioDallaVecchia, 2024-10-24:
Hi All!

I'm trying to download a large file using a simple HTTP GET request. Of course, I need to read data from socket in chunk in order to save to an external memory. I have the issue that after some minute, while I'm reading the data from the TCP socket, the sever close the connection. 

What is the problem? Why that happens?

Thanks a lot for any suggestions!

Giulio

nileshkale123, 2024-10-25:
Hello @GiulioDallaVecchia,

Could you please provide a detailed description of the steps you followed? If possible, could you also share a patch or reproducible code with sdkconfig file by making the necessary changes in one of the IDF examples?

Additionally, I recommend enabling debug logs to help clarify the issue—if you could send those along, it would be very helpful.

GiulioDallaVecchia, 2024-10-25:
Hi @nileshkale123,

While examining the code of esp_http_client, I've noticed that the client downloads a file in chunks of data by calling recv() on the open socket, using the buffer size as an argument to read each block of data. Doing this, isn't there a possibility that the remote server might close the connection before we've read all the data from the socket due to a server-side timeout expiring?

Wouldn't it be better to download a large file using the "Range:" header in the HTTP request?

Thanks a lot!

nileshkale123, 2024-11-08:
Hello @GiulioDallaVecchia ,

We have already implemented OTA downloading using range requests in the feature called 'partial HTTP download.' You can find this in the [advanced_ota_example](https://github.com/espressif/esp-idf/tree/master/examples/system/ota/advanced_https_ota) example.

Please refer to [documentation](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/esp_https_ota.html#partial-image-download-over-https).

Please review this, and let us know if it does not meet your requirements.

nileshkale123, 2024-11-29:
Hello @GiulioDallaVecchia,

I am closing this issue as the requested support is already available in IDF. Relevant references have been provided in the messages above. Please feel free to reopen if needed.

closed 2024-11-29 by nileshkale123