espressif/esp-idf #14779: ULP doesn't build under Linux -  No such file or directory "ulp_riscv***.h"

203Null, 2024-10-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ESP IDF Version: Tested on 5.0.6 & 5.3.1

Try to build system/ulp examples under Linux and it yields those results.

`/mnt/c/espressif/v5.3.1/examples/system/ulp/ulp_riscv/gpio/main/ulp/main.c:20:10: fatal error: ulp_riscv.h: No such file or directory
   20 | #include "ulp_riscv.h"
      |          ^~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
Missing "ulp_riscv.h" file name found in the following component(s): ulp(/home/null/esp/esp-idf/components/ulp/ulp_riscv/include/ulp_riscv.h), ulp(/home/null/esp/esp-idf/components/ulp/ulp_riscv/include/ulp_riscv/ulp_riscv.h), ulp(/home/null/esp/esp-idf/components/ulp/ulp_riscv/include/esp32s2/ulp_riscv.h). Maybe one of the components needs to add the missing header directory to INCLUDE_DIRS of idf_component_register call in CMakeLists.txt.`

chegewara, 2024-10-24:
What is the target you are trying to build for?

203Null, 2024-10-24:
> What is the target you are trying to build for?

ESP32-S3

chegewara, 2024-10-24:
And logs are for esp32 S2

203Null, 2024-10-24:
If additional info is needed. Here's my attempt to port my code to github action. (Compiles under windows) https://github.com/203Null/MatrixOS/actions/runs/11508133151/job/32035734531

203Null, 2024-10-24:
> And logs are for esp32 S2

Compiles fine under ESP32-S2

203Null, 2024-10-24:
For some reason, after building under s2, the s3 build now started to work.

chegewara, 2024-10-25:
It must have been some misconfig.
When you change target everything is rebuild, including sdkconfig. Most likely that fixed the problem.

203Null, 2024-10-25:
> It must have been some misconfig. When you change target everything is rebuild, including sdkconfig. Most likely that fixed the problem.

I figure out the issue.

I always do 
`
idf.py set-target esp32-s3
idf.py build
`

For some reason, this builds for esp32 instead of esp32-s3. 

`
idf.py set-target esp32-s3 build
`

this builds it just fine. Is this the correct behaviour?

closed 2024-10-25 by 203Null