espressif/esp-idf #14780: esp-tls sock=54 timeout on HTTPS OTA of firmware with Amazon S3

gamename, 2024-10-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

ESP32-C3 (QFN32) (revision v0.3)

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP-IDF

### Power Supply used.

USB

### What is the expected behavior?

The microcontroller should download the new firmware on Amazon S3 via HTTPS and install it when OTA is called.

### What is the actual behavior?

OTA appears to begin correctly, but then fails with a `sock=54` timeout.  

### Steps to reproduce.

1. Start the microcontroller
2. Publish a message to the OTA MQTT topic.  The published message contains the MAC address + a URL.  The URL is the new firmware image.
3. The error is seen.

### Debug Logs.

```plain
Here is the log:

<snip>
I (17:56:39.659) MQTT: MQTT_EVENT_DATA
I (17:56:39.660) MQTT: TOPIC=farm/coop/snooper/update
I (17:56:39.660) MQTT: DATA={"10:91:a8:1f:cb:14": "https://charlies-farm-ota.s3.us-east-2.amazonaws.com/coop-snooper/test/v1.0.0-729f97d/firmware.bin"}
I (17:56:39.673) MQTT: DATA Length=123
W (17:56:39.678) COOP_COP_EGG: Received topic farm/coop/snooper/update
I (17:56:39.685) COOP_COP_EGG: Orphan timer reset successfully
I (17:56:39.692) COOP_COP_EGG: OTA task handle is NULL
I (17:56:39.698) OTA: Starting...
I (17:56:39.701) OTA: Using URL: https://charlies-farm-ota.s3.us-east-2.amazonaws.com/coop-snooper/test/v1.0.0-729f97d/firmware.bin
I (17:56:39.714) OTA: Disabling Wi-Fi power save mode during OTA...
I (20810) wifi:Set ps type: 0, coexist: 0

I (17:56:39.725) OTA: Turn off MQTT so we can perform the update
I (17:56:39.734) RGB_LED_MGR: Blink task started with msg: R=0, G=8192, B=0, interval=500
I (17:56:39.778) OTA: MQTT client stopped successfully.
I (17:56:39.779) OTA: MQTT client destroyed successfully.
I (17:56:39.779) OTA: OTA started
I (17:56:40.699) OTA: Connected to server
I (17:56:40.699) esp_https_ota: Starting OTA...
I (17:56:40.700) esp_https_ota: Writing to partition subtype 16 at offset 0x140000
I (17:56:40.706) OTA: Reading Image Description
I (17:56:40.715) OTA: Currently running firmware version: v1.0.0-1dfa40b-1-g729f97d-dirty
I (17:56:40.721) OTA: Verifying chip id of new image: 5
E (17:58:22.903) esp-tls: [sock=54] select() timeout
E (17:58:22.904) esp-tls: Failed to open new connection
E (17:58:22.905) transport_base: Failed to open a new connection
E (17:58:22.911) HTTP_CLIENT: Connection failed, sock < 0
E (17:58:22.917) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (17:58:22.925) esp_https_ota: Failed to establish HTTP connection
E (17:58:22.932) OTA: Complete data was not received.
I (17:58:22.938) OTA: OTA abort
E (17:58:22.942) OTA: ESP_HTTPS_OTA upgrade failed
I (124040) wifi:state: run -> init (0x0)
I (124050) wifi:pm stop, total sleep time: 15338858 us / 123191843 us

I (124050) wifi:<ba-del>idx:0, tid:0
I (124050) wifi:<ba-del>idx:1, tid:2
I (124060) wifi:new:<3,0>, old:<3,0>, ap:<255,255>, sta:<3,0>, prof:1, snd_ch_cfg:0x0
I (124110) wifi:flush txq
I (124110) wifi:stop sw txq
I (124110) wifi:lmac stop hw txq
I (17:58:23.014) OTA: Wi-Fi stopped successfully.
I (124110) wifi:Deinit lldesc rx mblock:10
I (17:58:23.023) OTA: Wi-Fi deinitialized successfully.
```
```

### More Information.

1. This is happening on multiple esp32c3 microcontrollers.
2. The OTA code is a near-verbatim copy of the [advanced_https_ota_example](https://github.com/espressif/esp-idf/blob/master/examples/system/ota/advanced_https_ota/main/advanced_https_ota_example.c)
3. My OTA code is [here](https://github.com/espressif/esp-idf/blob/master/examples/system/ota/advanced_https_ota/main/advanced_https_ota_example.c)
4. The `app_main` which calls the code is [here](https://github.com/chicken-coop-door-status/C2DS-esp32-coop-snooper/blob/main/main/main.c)

gamename, 2024-10-25:
Additional info. This is a manual download of a file from the same s3 directory and using the same ca cert:
```
curl --cacert ./AmazonRootCA1.pem https://charlies-farm-ota.s3.us-east-2.amazonaws.com/coop-snooper/test/v1.0.0-729f97d/firmware.bin --output my_test_firmware.bin
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1153k  100 1153k    0     0  1006k      0  0:00:01  0:00:01 --:--:-- 1006k
```

jthomp-chewie, 2024-10-25:
I've run into a similar issue with HTTP OTAs, specifically ones coming from S3. A few thoughts:

- Try enabling `DEBUG` level logs when you reproduce the problem, the `esp_https_ota` has some helpful logs here.
- My particular issue reproduced when I was using a cellular hotspot to connect by board to AWS. I found that my board could download files of a similar size as my firmware as long as those files came from outside of S3. One example I tried was downloading https://www.amazontrust.com/repository/cp-1.0.13.pdf (~1.5MB).
- You might try hosting your OTA images with a [CloudFront](https://docs.aws.amazon.com/AmazonCloudFront/latest/DeveloperGuide/GettingStarted.SimpleDistribution.html) distribution. A content distribution network is simply better suited for moving files over high latency connections than S3 can be.
- I'm looking into using OTA over MQTT as a "real" solution to this issue. If this is something you can get going, it's worth a try!

gamename, 2024-10-25:
> I've run into a similar issue with HTTP OTAs, specifically ones coming from S3. A few thoughts:
>
> * Try enabling `DEBUG` level logs when you reproduce the problem, the `esp_https_ota` has some [...]

Good info! Thanks for posting this.

gamename, 2024-10-27:
@jthomp-chewie Your suggestion worked!  Using a CloudFront distribution made the problem go away.  Thanks!

closed 2024-10-27 by gamename