espressif/esp-idf #14781: WiFi network latency is unstable, how to improve

luacerbb, 2024-10-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

idf_v4.4.4

### Espressif SoC revision.

ESP32-WROOM-32D

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom board

### Power Supply used.

USB

### What is the expected behavior?

I hope to improve the stability of communication under weak signals

### What is the actual behavior?

I test with four wifi dev,  801 and 819 dev used esp32 chip,we can see those two dev ping packet  lost  bigger than rachio3,and latency is unstable.
[image]

### Steps to reproduce.

1. Four devices are placed together and connected to the same wifi router
2. Test by sending ping requests to four devices on one computer

MaxwellAlan, 2024-10-28:
Hi @luacerbb 

Could you try disable WiFi modem sleep (calling esp_wifi_set_ps(0) after esp_wifi_start()) and re-test ?

luacerbb, 2024-10-30:
@MaxwellAlan 
thanks,I will arrange for my colleague to retest

AxelLin, 2024-11-07:
> Hi @luacerbb
> 
> Could you try disable WiFi modem sleep (calling esp_wifi_set_ps(0) after esp_wifi_start()) and re-test ?

@MaxwellAlan 
I don't get your point, do you mean it's normal behavior with WiFi modem sleep enabled?
If so, why the device rachio3 does not show the same symptom?

@luacerbb 
Can you make sure the setting is exactly the same for these 4 devices?

luacerbb, 2025-01-16:
@MaxwellAlan 
disable WiFi modem sleep,It will indeed improve the communication。thanks.

closed 2025-01-16 by luacerbb