espressif/esp-idf #14782: LVGL build fail

emesamwong, 2024-10-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am using LVGL v9.2.1 esp component for my development. I start with the esp-matter "light" example and add the lvgl/lvgl: "9.2.1" components. There is no code change for the 'light' project. However, contains the below error
[image]

What should be the value of CONFIG_LV_ATTRIBUTE_LARGE_CONST if I don't care about the flash region?

Please let me know if more info is needed

ESP-IDF: release/v5.1 (#632e0c2)
ESP-MATTER: #2f692ac
lvgl/lvgl component: 9.2.1

suda-morris, 2024-10-25:
https://github.com/lvgl/lvgl/issues/7154

emesamwong, 2024-10-28:
@suda-morris Thank you for your information. It builds successfully with LVGL 9.2.0

Another problem is the Task Watchdog timeout when the lv_task_handler() is executed.
[image]

lv_task_handler() disabled:
[image]

Here is the Config code of lvgl:
[image]
[image]
[image]

Code of lv_task_handler():
[image]

The example project is combined esp-matter "light" & esp-idf "i80_controller"
Here is the project code:
[light_lvgl.zip](https://github.com/user-attachments/files/17540580/light_lvgl.zip)

Is something wrong with my lvgl init code?  
Any successful example using LVGL v9.0.0 or above

LCD driver: ST7789