espressif/esp-idf #14783: OTA: Reuse HTTP connection when performing partial HTTP downloads

admarschoonen-joule, 2024-10-25:
### Is your feature request related to a problem?

If OTA is configured to do a partial HTTP download, I noticed that it closes the HTTP connection after every 4 KB that it has received, and opens a new connection for the following 4 KB. It would be nice if it could reuse this connection to save time on DNS requests and setting up / tearing down HTTP and TLS sessions.

nileshkale123, 2024-10-28:
Hello @admarschoonen-joule ,

Yes, we agree with you and have already identified this issue. We’ll be implementing a fix for it shortly.

admarschoonen-joule, 2024-10-28:
Hi @nileshkale123 ,

Thanks for you quick reply. Can you give an estimation of when this fix will be released and which ESP-IDF version number will have the fix?

Thanks in advance.

admarschoonen-joule, 2024-10-28:
Hi @nileshkale123 ,

Thanks a lot! Looking forward to the improvement!

nileshkale123, 2024-11-06:
Hello @admarschoonen-joule,

I’m attaching a patch that enables the session resumption feature in the advanced_ota example in IDF. This approach maintains the current flow, where each range request initiates a new connection, rather than persisting a single connection across all range requests.

Although session resumptionis quite slower than using a fully persistent connection, it provides significant advantages in low-bandwidth or unreliable network conditions and requires fewer resources.

Please review the patch and let us know your thoughts. If you have any specific requirements or preferences regarding the implementation, feel free to share them so we can work closely to meet your needs.

[partial_ota.txt](https://github.com/user-attachments/files/17643631/partial_ota.txt)

nileshkale123, 2024-11-29:
Thanks @admarschoonen-joule ,

The MR is under review. We will merge it soon and it will be backported on previous supported versions as well.

closed 2024-12-03 by espressif-bot