espressif/esp-idf #14787: `uart_set_pin()` with the same pin for both RX and TX doesn't work!

arduino12, 2024-10-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.0

### Espressif SoC revision.

ESP32-C3 (revision v0.4)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32 C3 Super Mini

### Power Supply used.

USB

### What is the expected behavior?

Full code demo:
```c
#include "freertos/FreeRTOS.h"
#include "driver/uart.h"
#include "soc/uart_periph.h"
#include "driver/gpio.h"
#include "esp_log.h"

#define BYPASS              (0) // don't use `uart_set_pin`
#define WORKAROUND          (1) // add workaround after calling `uart_set_pin`
#define UART_TX_RX_PIN      (0) // GPIO used for both TX and RX lines - can hook to logic analyzer / USB-UART
#define UART_PORT_NUM       (1)
#define UART_BAUD_RATE      (1000000)
#define UART_BUF_SIZE       (1024)

static const char *TAG = "UART TEST";

void app_main()
{
    uart_config_t uart_config = {
        .baud_rate = UART_BAUD_RATE,
        .data_bits = UART_DATA_8_BITS,
        .parity    = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
        .source_clk = UART_SCLK_DEFAULT,
    };

    ESP_ERROR_CHECK(uart_driver_install(UART_PORT_NUM, UART_BUF_SIZE, 0, 0, NULL, 0));
    ESP_ERROR_CHECK(uart_param_config(UART_PORT_NUM, &uart_config));

#if BYPASS
    ESP_ERROR_CHECK(gpio_set_pull_mode(UART_TX_RX_PIN, GPIO_PULLUP_ONLY));
    ESP_ERROR_CHECK(gpio_set_direction(UART_TX_RX_PIN, GPIO_MODE_INPUT_OUTPUT_OD));
    esp_rom_gpio_connect_out_signal(UART_TX_RX_PIN, UART_PERIPH_SIGNAL(UART_PORT_NUM, SOC_UART_TX_PIN_IDX), 0, 0);
    esp_rom_gpio_connect_in_signal(UART_TX_RX_PIN, UART_PERIPH_SIGNAL(UART_PORT_NUM, SOC_UART_RX_PIN_IDX), 0);
#else
    ESP_ERROR_CHECK(uart_set_pin(UART_PORT_NUM, UART_TX_RX_PIN, UART_TX_RX_PIN, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE));
#if WORKAROUND
    esp_rom_gpio_connect_out_signal(UART_TX_RX_PIN, UART_PERIPH_SIGNAL(UART_PORT_NUM, SOC_UART_TX_PIN_IDX), 0, 0);
#endif
#endif

    const char *TX_BUFF = "Test";
    char *rx_buff = (char *)malloc(UART_BUF_SIZE);

    while (1) {
        uart_write_bytes(UART_PORT_NUM, TX_BUFF, sizeof(TX_BUFF));
        int len = uart_read_bytes(UART_PORT_NUM, rx_buff, (UART_BUF_SIZE - 1), 20 / portTICK_PERIOD_MS);
        if (len) {
            rx_buff[len] = '\0';
            ESP_LOGI(TAG, "Echo: %s", rx_buff);
        }
        else {
            ESP_LOGW(TAG, "No echo");
        }
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}
```

I expected:
```c
uart_set_pin(UART_PORT_NUM, UART_TX_RX_PIN, UART_TX_RX_PIN, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE)
```
to work- but is seems to overwrite the TX output configuration and only the RX pin is configured correctly.

My workaround is to re-write the TX output configuration after the `uart_set_pin(..)` :
```c
esp_rom_gpio_connect_out_signal(UART_TX_RX_PIN, UART_PERIPH_SIGNAL(UART_PORT_NUM, SOC_UART_TX_PIN_IDX), 0, 0);
```
I think the `uart_set_pin(..)` needs to support this case of a single pin for both TX and RX-
(maybe test if `rx_pin == tx_pin` and if so- apply my workaround?)
I use it to control some single line smart servo motors with open-drain UART bus...

The ESP32 support this common UART configuration (like RS485),
it also support **[echo-cancellation](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/uart.html#overview-of-rs485-specific-communication-0ptions)** (may also be nice to have this functionality..)

### What is the actual behavior?

When:
```c
#define BYPASS              (0) // don't use `uart_set_pin`
#define WORKAROUND          (0) // add workaround after calling `uart_set_pin`
```
I get:
`W  UART TEST: No echo` - Bad!

When:
```c
#define BYPASS              (0) // don't use `uart_set_pin`
#define WORKAROUND          (1) // add workaround after calling `uart_set_pin`
```
or
```c
#define BYPASS              (1) // don't use `uart_set_pin`
```
I get:
`I  UART TEST: Echo: Test` - Good!

### Steps to reproduce.

Compile and run the demo code I made with and without the `WORKAROUND`..

arduino12, 2024-10-25:
I encountered this bug with MicroPython and opened an issue:
https://github.com/micropython/micropython/issues/16079
(Tested on ESP32-S2, S3, C3)

I finally got here realizing the bug is in `uart_set_pin(..)`

closed 2024-11-19 by espressif-bot

pixelgrb, 2025-09-30:
Hi @songruo,

I think this commit is breaking UART initialization for a bunch of people (include me) running this project: https://github.com/echavet/MitsubishiCN105ESPHome, which uses ESP32 devices to control Mitsubishi HVAC units.

AFAIK, it affects esp32-c6-devkitc-1 boards.  Some of us reporting are using the M5Stack NanoC6.

Examples:
- https://github.com/echavet/MitsubishiCN105ESPHome/issues/378#issuecomment-3348667904
- https://github.com/echavet/MitsubishiCN105ESPHome/issues/396
- https://github.com/esphome/esphome/issues/10338#issuecomment-3209070095
- and more I think.

I'm brand new to this project (and to ESP32), but I put in time to track this down.  I pinned it down to this one commit through git bisect.

**Important to understand**:
A lot of the information in those issues is incorrect, because the symptom **only shows after a cold boot** of the ESP32, yet resolves **without a cold boot** if going back to older firmware.

That is:
- If somebody has an ESP32 working with the Mitsubishi unit, and does an OTA update which includes this code, it stays working.
- When the ESP32 is powered off and back on, the UART connection to the Mitsubishi unit fails .. consistently.
- If they revert to an old ESPHome version (via OTA update) and it works again.  No cold reboot is required.
- But if they OTA update to a newer ESPHome version (which includes this change) - it stays working!  This is the confusing part.

I git bisected the ESPHome code, and then the ESP-IDF code.  Each time the device was working with the Mitsubishi, I had to unplug and replug the M5Stack NanoC6 to cold boot it after the next OTA update to test properly.

I was very meticulous when I was testing to make sure I made no mistakes, and I double-checked the results.  I summarized my findings in this comment:

https://github.com/echavet/MitsubishiCN105ESPHome/issues/378#issuecomment-3348667904

**What I need help with:**

I don't have enough experience with ESP32 or UART code to understand why this change causes the cold boot issue or how to fix it. I'm hoping someone more experienced can:
1. Review the commit to identify what might cause UART initialization to fail on cold boot
2. Suggest a fix or workaround
3. Let me know if there's any additional testing or information I can provide to help debug this

The UART configuration being used in the affected project is here: https://github.com/echavet/MitsubishiCN105ESPHome/blob/main/components/cn105/cn105.h

I'm happy to test any proposed fixes on my hardware if that would be helpful.

pixelgrb, 2025-10-19:
@songruo - thank you.  Sorry it took me a few days to get around to testing this.

**The patched code is still broken unfortunately.**

If there is anything else I can test, I'll be glad to.

I double checked everything.  I applied the patch to ESP-IDF v5.4.2, compiled with ESPHome v2025.9.2 (this version should not matter).  I made sure I was using your patched code.

This is how I get ESPHome to build against a local ESP-IDF:

```
esphome:
  platformio_options:
    platform_packages:
      - 'framework-espidf @ symlink:///Users/glen/Projects/minisplit-bisect/esp-idf'
```

In the esp-idf project folder:
```
% git tag --points-at HEAD; git status
v5.4.2
On branch v5.4.2-patched
Changes to be committed:
  (use "git restore --staged <file>..." to unstage)
	modified:   components/esp_driver_uart/src/uart.c

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	components/__pio_env/
```

Still the old error:

```
[16:48:31.487][I][app:185]: ESPHome version 2025.9.2 compiled on Oct 18 2025, 16:47:52
[16:48:40.993][E][CN105:032]: --> Heatpump did not reply: NOT CONNECTED <--
[16:48:40.993][I][CN105:033]: Trying to connect again...
[16:48:50.981][W][CN105:190]: Heatpump has not replied for 20 s
[16:48:50.982][I][CN105:191]: We think Heatpump is not connected anymore..
[16:48:50.983][I][CN105:008]: setupUART() with baudrate  2400
[16:48:52.989][W][CN105:190]: Heatpump has not replied for 22 s
...
```

gekkekoe, 2025-10-22:
https://github.com/esphome/esphome/issues/11038#issuecomment-3374955358
This seems to work for me. So basically reset gpio 17/18 and it seems to work. It still works after cold boot with the latest esphome 2025.10.2

It might point @songruo into the right direction.

pixelgrb, 2025-10-22:
Thanks @gekkekoe.  I also confirm the `gpio_reset_pin` trick works for me too.  In my case, I'm using an M5StackNanoC6 so are different pins, so my config is:

```
esphome:
  on_boot:
    priority: 1001
    then:
      - lambda: |-
          gpio_reset_pin((gpio_num_t)1);
          gpio_reset_pin((gpio_num_t)2);
```

gekkekoe, 2025-10-22:
Did you test with cold boot ? (so power completely off). I integrated the work around but it seems to work correctly. 17/18 are default tx/rx pints, so it might have something to do with that.

gekkekoe, 2025-10-22:
I am the maintainer of https://github.com/gekkekoe/esphome-ecodan-hp

Its for air to water source heatpumps. i use arom s3 lite.

songruo, 2025-10-23:
@gekkekoe @pixelgrb Maybe because I remove the RX pull up? Please try if calling `gpio_pullup_en` on the RX pin can help.

pixelgrb, 2025-10-23:
Yes - this works for me too.

GioMez, 2025-10-23:
> [@songruo](https://github.com/songruo) the `gpio_pullup_en` also works here, and unit can connect after power cycle. Looks like this is the culprit ? [@pixelgrb](https://github.com/pixelgrb) can you also verify with your unit since its a bit more picky than mines?
> 
> substitutions:
>   uart_rx_pin: 2
>   uart_tx_pin: 1
> 
> esphome:
>   on_boot:
>     - priority: 1001
>       then:
>         - lambda: |-
>             gpio_pullup_en(static_cast<gpio_num_t>(${uart_rx_pin}));

Have you tried using the YAML syntax suggested in [#9248](https://github.com/esphome/esphome/pull/9248) with explicit `pullup` and `input` booleans?

```
uart:
  - id: YOUR_UART_ID
    tx_pin: YOUR_TX_PIN
    rx_pin:
      number: YOUR_RX_PIN
      mode:
        pullup: true
        input: true

```
You’ll need to either reference that PR in your `external_components` source or use the upcoming ESPHome 2025.10.3 release, where it has been merged.
It might also depend on the specific board; on some setups the internal pull-up might not be needed.

gekkekoe, 2025-10-23:
@pixelgrb thanks for testing. I got user reports that the workaround is working for my users. So will test again when 10.3 is released (and remove the workaround)