espressif/esp-idf #14790: [ESP32S3] [5.3] Migration to 5.3 from 4.4 breaks custom BLE provisioning endpoint

KonssnoK, 2024-10-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

This is a bit embarassing.

We are migrating from 4.4 to 5.3,
the code of our provisioning app using security 1 is exactly the same.

what we see is that our custom endpoint in the BLE provisioning manager gets not sent to the provisioning phone.

specifically, we see that the endpoint is generated and added with the following 2 calls:

```
    /* An optional endpoint that applications can create if they expect to
     * get some additional custom data during provisioning workflow.
     * The endpoint name can be anything of your choice.
     * This call must be made before starting the provisioning.
     */
    wifi_prov_mgr_endpoint_create("prov-macs");

    ...

    /* The handler for the optional endpoint created above.
     * This call must be made after starting the provisioning, and only if the endpoint
     * has already been created above.
     */
    wifi_prov_mgr_endpoint_register("prov-macs", ble_prov_macs_data_handler, NULL);
```

```

W (01:00:09.641) wifi_prov_scheme_ble: set_config_endpoint prov-ctrl
W (01:00:09.642) wifi_prov_scheme_ble: set_config_endpoint prov-scan
W (01:00:09.645) wifi_prov_scheme_ble: set_config_endpoint prov-session
W (01:00:09.652) wifi_prov_scheme_ble: set_config_endpoint prov-config
W (01:00:09.659) wifi_prov_scheme_ble: set_config_endpoint proto-ver
I (01:00:09.666) prov_BLE: Starting BLE provisioning
W (01:00:09.672) wifi_prov_scheme_ble: set_config_endpoint prov-macs

```

but the app receives:

```
2024-10-26 11:09:54.584 14597-14609 BluetoothGatt           com.espressif.wifi_provisioning      D  discoverServices() - device: AC:17:54:00:0A:B2
2024-10-26 11:09:54.588 14597-14609 BluetoothGatt           com.espressif.wifi_provisioning      D  onSearchComplete() = Device=XX:XX:XX:XX:0A:B2 Status=0
2024-10-26 11:09:54.588 14597-14609 Espressif::BLETransport com.espressif.wifi_provisioning      D  On services discovered
2024-10-26 11:09:54.588 14597-14609 Espressif::BLETransport com.espressif.wifi_provisioning      D  Characteristic UUID : 021aff50-0382-4aea-bff4-6b3f1c5adfb4
2024-10-26 11:09:54.589 14597-14609 Espressif::BLETransport com.espressif.wifi_provisioning      D  Characteristic UUID : 021aff51-0382-4aea-bff4-6b3f1c5adfb4
2024-10-26 11:09:54.589 14597-14609 Espressif::BLETransport com.espressif.wifi_provisioning      D  Characteristic UUID : 021aff52-0382-4aea-bff4-6b3f1c5adfb4
2024-10-26 11:09:54.589 14597-14609 Espressif::BLETransport com.espressif.wifi_provisioning      D  Characteristic UUID : 021aff53-0382-4aea-bff4-6b3f1c5adfb4
2024-10-26 11:09:54.589 14597-14609 Espressif::BLETransport com.espressif.wifi_provisioning      D  Characteristic UUID : 021aff54-0382-4aea-bff4-6b3f1c5adfb4
```

Note: this is working under 4.4

rahult-github, 2024-10-29:
Hi @KonssnoK ,

Is the remote device iphone / iOS ?  If yes, then this is iOS side issue, where it caches the service information. The workaround for this is to make iOS side forget ble settings / device ( i.e. invalidate the cache ) and try again. This makes iOS to fetch the service information again and should show up the updated information.

KonssnoK, 2024-10-29:
@rahult-github i didn't even have the chance to try ios yet.
I am talking only about android for now.

What i was able to understand:
- the GATT DB is correctly created with all the entries
- the robust caching is disabled
- the BT_CLASSIC_ENABLED is disabled and doesn't seem to be a reqiurement looking at the example
- I was not able to understand which function is called on gatt.discoverServices().
- Nimble does not have the issue, Bluedroid is the only affected stack

KonssnoK, 2024-11-04:
@esp-zhp the device i was using was not paired to my mobile device.
The issue happens only with bluedroid, so i don't see how previous bonding information could be relevant.
We are not currently looking into this for now since Nimble works ok.
We will come back on bluedroid in case we find that some phones used by installers do not support Nimble.

esp-zhp, 2024-11-04:
ok

Alvin1Zhang, 2024-12-17:
Thanks for reporting and sharing the updates, feel free to reopen if the issue happens again.

closed 2024-12-17 by Alvin1Zhang

KonssnoK, 2024-12-17:
@Alvin1Zhang please note that bluedroid is still broken, we moved to nimble to resolve

esp-zhp, 2024-12-18:
@KonssnoK 
 if you use the Bluedroid host again, you can reopen this issue.