espressif/esp-idf #14797: v5.3 of ESP-IDF reports notably lower RSSI than v5.1

mbratch, 2024-10-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

Chip is ESP32-PICO-V3 (revision v3.0)

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32 Dev Kit 1

### Power Supply used.

USB

### What is the expected behavior?

After upgrading my ESP-IDF to v5.3.1 (from v5.1), I would expect to see no significant change in RSSI readings if the wifi environment hasn't changed.

### What is the actual behavior?

After upgrading my ESP-IDF to v5.3.1 (from v5.1), what I see is that the RSSI reported by ESP-IDF using the `esp_wifi_sta_get_ap_info` function is significantly lower after the upgrade. Specifically, the RSSI with v5.1 was between -70 and -60. After the upgrade to v5.3.1 it dropped to between -127 and -80.

### Steps to reproduce.

1. Using v5.1 ESP-IDF operate software and report RSSI
2. Observe RSSI between -70 and -60.
3. Upgrade to v5.3.1 ESP-IDF and operate software
4. Observe RSSI is now between -127 and -80.

zhangyanjiaoesp, 2024-11-07:
@mbratch 
What if you put the phy lib (idf/components/esp_phy/lib/esp32)under v5.1.4 on v5.3? Will the RSSI change to be consistent with 5.1.4?

mbratch, 2024-11-12:
@zhangyanjiaoesp thank you very much for the prompt reply and the information regarding internal testing.

I did not perform a deep sleep, and my test process was not very formal but just observational looking at an RSSI indicator that I defined and my units under test (I have four of them) are stationary in a test setup. Of course, with wifi, there are natural fluctuations, and I tried to take that into account. But based upon your comments, I suspect perhaps I didn't capture enough data over a period of time to firmly determine what is going on.

I will run more tests when I have the opportunity.

zhangyanjiaoesp, 2025-01-07:
@mbratch Do you still have this issue? Can we close it?

closed 2025-01-07 by mbratch